RalphBacon/205-Internet-Radio #1: ESP32 ECO V3

NonaSuomy, 2020-11-20:
Hi Ralph,

Was just watching your video and heard you say the latest version is V1 and that you were looking at using the PSRAM. Just wanted to show you this nfo from Expressif in case you hit these issues with the WROVER PSRAM on older V1 versions instead of the V3.

https://www.espressif.com/sites/default ... de__EN.pdf

Earlier versions had bugs in the silicon which they have fixed in this revision.

Design Change in ECO V3

Espressif has recently released one wafer-level change on ESP32 Series of products (ECO
V3). This document describes differences between V3 and previous ESP32 silicon wafer
revisions. Below are the main design changes in ECO V3 Series of chips:

1. PSRAM Cache Bug Fix: Fixed “When the CPU accesses the external SRAM in a certain
sequence, read & write errors can occur.”. Details of the issue can be found in item 3.9
in ESP32 ECO and Workarounds for Bugs.

2. Fixed “When each CPU reads certain different address spaces simultaneously, a read
error can occur.” Details of the issue can be found in item 3.10 in ESP32 ECO and
Workarounds for Bugs.

3. Optimized 32.768 KHz crystal oscillator stability, the issue was reported by client that
there is a low probability that under ECO V1 hardware, the 32.768 KHz crystal oscillator
couldn’t start properly.

4. Fixed Fault injection issues regarding secure boot and flash encryption are fixed.
Reference: Security Advisory concerning fault injection and eFuse protections
(CVE-2019-17391) & Espressif Security Advisory Concerning Fault Injection and Secure
Boot (CVE-2019-15894)

5. Improvement: Changed the minimum baud rate supported by the CAN module from 25
kHz to 12.5 kHz.

6. Allowed Download Boot mode to be permanently disabled by programming new eFuse
bit UART_DOWNLOAD_DIS. When this bit is programmed to 1, Download Boot mode
cannot be used and booting will fail if the strapping pins are set for this mode. Software
program this bit by writing to bit 27 of EFUSE_BLK0_WDATA0_REG, and read this bit by
reading bit 27 of EFUSE_BLK0_RDATA0_REG. Write disable for this bit is shared with
write disable for the flash_crypt_cnt eFuse field.

Great job with the webradio videos keep it up! Please keep deep-diving into how the code and ring buffers work!

I was playing with https://github.com/Edzelf/ESP32-Radio before and helped include iHeartRadio to it a long while ago. The thing I liked about this code base is you can use MQTT to control the radio for home automation.

RalphBacon, 2020-11-21:
Good morning Nona Suomy!

 

Thank you so much for the information.

 

Unfortunately I get a “forbidden” message when accessing that web page, but your summary serves well. I will read it in much more detail today.

 

I have implemented a software workaround for the PSRAM error in V1 chips (makes me wonder how much testing they ever did).

 

I looked at the Web Radio from “Edzelf” aka Ed Smallenburg and realised it was very well featured but I wanted to be able to design and build my own design, which I feel I have now done (hardware issues not withstanding). But I have quite a lot more work to do if I am going to make more useful so I won’t get bored.

 

I’ve been running my ESP32 Rev 1 WROVER board (delivered yesterday) all last night and overnight without a single disconnect. I’m going to be testing how well it (re) connects to the WiFi today and see whether my issues have disappeared. My other V1 chip ran great for 2 days then got the same errors. I have a nagging doubt that it was my config, the router or just my Internet connection that might have been to blame. Perhaps I’ll never find out. What I do know is the ESP8266 version ran without issue at all, ever.

 

Thanks for the info again, great to hear from you.

 

Best Regards

Ralph S Bacon

 <https://www.youtube.com/ralphbacon> https://www.youtube.com/ralphbacon

NonaSuomy, 2020-11-23:
Sorry I fixed the link on github but it already emailed you the bad link. Was basically just what I put below that. 

Here is the ECO V3 version https://www.aliexpress.com/item/4001095751970.html?spm=a2g0o.cart.0.0.c58e3c00UAaLaV&mp=1 

I wonder if using an antenna on the board as well would help with reception instead of the pcb ant. Some breakout boards put the ground planes under the pcb ant which is apparently a no-no.

Even with edzelf and karadio https://github.com/karawin/Ka-Radio32 I had that same stuttering issue. Will be great if you find a solution with your build!

I'm glad you are doing your own version as it's nice to see someone explain all the finer details instead of just dumping a bunch of code online. Just doesn't do justice for the extreme curious.

I one time contacted SpriteTM (ExpressIF staff) and asked what he thinks about using this 1053 codec board with the esp32 as a webradio. He told me in his colourful way that it's pointless as the esp32 can do all that without it :) more food for thought? haha...

Keep on hacking! Great content and thanks!

RalphBacon, 2020-11-23:
Thanks for the link to the up-to-date ESP32 V3 (ECO no less!) – I wish there was a suitable board to mount it on, rather than having to remove an existing ESP32 or creating a brand new board (with the voltage regulators, USB connectivity etc). I did find several like this (“Adapter board for welding”):

 <https://www.aliexpress.com/item/32980341113.html> https://www.aliexpress.com/item/32980341113.html

but the design is only suitable for DuPont cables or a PCB – in which case I could do a better job with a custom PCB myself! And I have vague memories of Andreas Spiess warning against them as they were missing a couple of essential resistors.

 

I have no stuttering at all on my WROVER version (you will hear my thoughts in Friday’s video, but not the radio). Rock solid. Working now for 4 days without a problem. I will have to post the updated code to my GitHub soon. Of course, stuttering depends on the Internet generally, but also the radio server to which you are connected. My buffering (10 seconds) will do nothing for problems in either of those areas! Apart from BBC Radio 4 (don’t get me started) my design work just fine. If I connect to BBC Radio 4 without requesting metadata then that works fine too. Good old BBC. Not.

 

Actually, it works “too well”. I’m very tempted to just throw it into a box and keep it running in there!!! But I want a PCB and an enclosure of some sort, bearing in mind the screen must be exposed for the touch functionality. I’ve been urged by other viewers to use a Nextion screen (just three wires, including the power!) so I will make the whole design screen agnostic for other to plug in what they want. Time permitting.

 

Regarding the antenna, my WROVER has a built in bit of flimsy metal (not a PCB trace) that seems to work well, but also has a connection point for an external aerial. I discuss this in Friday’s video.

 

I’m learning so much on this project about the ESP32 (and PlatformIO) it’s been a great topic and could go on and on!

 

Thanks for all the info, really useful, keep tuned!

 

Best Regards

Ralph S Bacon

 <https://www.youtube.com/ralphbacon> https://www.youtube.com/ralphbacon 

 

Here is the ECO V3 version https://www.aliexpress.com/item/4001095751970.html?spm=a2g0o.cart.0.0.c58e3c00UAaLaV <https://www.aliexpress.com/item/4001095751970.html?spm=a2g0o.cart.0.0.c58e3c00UAaLaV&mp=1> &mp=1

RalphBacon, 2021-01-24:
This will be progressed (using a V3 ECO chip) in the future.

closed 2021-01-24 by RalphBacon