RalphBacon/205-Internet-Radio #2: Need help to start, please

Aleks-Ale, 2021-01-06:
Please help me to run this project by any method.
ESP32 WROOM + 2.8"ili9341 w/touch + VS1053
TFT_eSPI-2.2.23\examples\320 x 240\TFT_Clock - work fine
ESP_VS1053_Library\examples\SimpleMp3Player - work fine

Arduino IDE: Compilation runs without errors. But after loading the program into ESP32, I see only a white screen and a cyclic restart in the serial port monitor.
.....................
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
abort() was called at PC 0x401578ef on core 0

Backtrace: 0x4008dc5c:0x3ffe3ab0 0x4008de8d:0x3ffe3ad0 0x401578ef:0x3ffe3af0 0x40157936:0x3ffe3b10 0x4013f85f:0x3ffe3b30 0x4013f94e:0x3ffe3b50 0x4013f905:0x3ffe3b70 0x400dd0b9:0x3ffe3b90 0x400d1ef4:0x3ffe3bb0 0x400e063b:0x3ffe3bd0 0x400820d9:0x3ffe3bf0 0x400822e8:0x3ffe3c20 0x40079053:0x3ffe3c40 0x400790b9:0x3ffe3c70 0x400790c4:0x3ffe3ca0 0x4007928d:0x3ffe3cc0 0x400806de:0x3ffe3df0 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20

Rebooting...`
.....................

PlatformIO: Unsuccessfully again.
I installed for the first time Visual Studio Code with PlatformIO Core: 5.0.4 in Python 3.9.1.

.....................
Library Manager: Installing git+https://github.com/lorol/LITTLEFS.git
FileNotFoundError: [WinError 2] 
  File "C:\users\alexander\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 170:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Alexander\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Alexander\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\users\alexander\.platformio\platforms\espressif32\builder\main.py", line 224:
    target_elf = env.BuildProgram()
  File "C:\Users\Alexander\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\users\alexander\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\Alexander\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\users\alexander\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Alexander\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\users\alexander\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1062:
    project.install_dependencies()
  File "C:\users\alexander\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 898:
    lm.install(spec)
  File "c:\users\alexander\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 49:
    spec, silent=silent, skip_dependencies=skip_dependencies, force=force
  File "c:\users\alexander\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 86:
    force=force,
  File "c:\users\alexander\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 97:
    pkg = self.install_from_url(spec.url, spec, silent=silent)
  File "c:\users\alexander\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 141:
    vcs = VCSClientFactory.new(tmp_dir, url)
  File "c:\users\alexander\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 55:
    src_dir, remote_url, tag, silent
  File "c:\users\alexander\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 137:
    self.configure()
  File "c:\users\alexander\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 146:
    result = proc.exec_command([cls.command, "--exec-path"])
  File "c:\users\alexander\.platformio\penv\lib\site-packages\platformio\proc.py", line 113:
    p = subprocess.Popen(*args, **kwargs)
  File "C:\Users\Alexander\.platformio\python3\lib\subprocess.py", line 800:
    restore_signals, start_new_session)
  File "C:\Users\Alexander\.platformio\python3\lib\subprocess.py", line 1207:
    startupinfo)
.....................

RalphBacon, 2021-01-06:
The Arduino version is somewhat out-of-date and I don't think it auto-detects whether you have a WROVER installed or not (ie additional PSRAM) so I suspect it's trying to allocate all your SRAM to the circular buffer. Go to the main.h and reduce the circular buffer size to 10,000 (ten thousand) which will be fine for WROOM-based ESP32s, although you only get about half a second of buffering. Let me know.

Aleks-Ale, 2021-01-07:
> reduce the circular buffer size to 10,000
Thank you! I tested different values. 80,000 is good.
Now I'm trying to understand why pressing the NEXT button on touch srceen does not work. I added calibration and getTouchRaw for the test. These tests work. But pressing the touch button does not work to switch station.
Any ideas?

RalphBacon, 2021-01-07:
Do any on screen controls work? Check that the T_IRQ from the screen is connected correctly and is being detected.

Aleks-Ale, 2021-01-08:
I understood and corrected all my stupid mistakes. 
This made it possible to prepare for your next project # 208. I really liked your work. 
This is a good example for understanding and learning. 
For example, your solution to change the screen brightness is very interesting. 
I will gladly study your project code # 208. 
Do you have a plan to integrate a spectrum analyzer? For example like this:
https://github.com/blotfi/ESP32-Radio-with-Spectrum-analyzer
It is also possible to create a screen with station names to scroll through the screen and select the desired station by touching.

RalphBacon, 2021-01-08:
I regret that I won't be implementing a spectrum analyser. 
But I have already implemented a station lister (it now reads from a text file in the SPIFFS partition) and selects from the screen; I have not uploaded to GitHub yet, as it is in PlatformIO project format only right now. I will have to port it to the Arduino format "soon".
[image]
[image]

The two photos show the new format of the screen. Pressing the "?" button will bring up pages of stations to choose from.

Aleks-Ale, 2021-01-08:
> implemented a station lister (it now reads from a text file in the SPIFFS partition) and selects from the screen

OMG!!! You rock!!! You're cool!! 
Short story. My first web player is a project from Edzelf. The next modification with a spectrum analyzer from blotfi. 
I made this screen design for myself. 

[image]

It works, but a lot of code, a lot of unnecessary functions, no touch. 
I dreamed of finding a short project with touch control. Web radio only. 
Because I am a beginner and cannot create such a solution. 
Your project is short and understandable.
If you can send me this project, as in your photo, as it is, I will be very grateful to you. Maybe I can port it for arduino ide.

Aleks-Ale, 2021-01-18:
> I have already implemented a station lister

Dear Mr. Ralph Bacon,
Sorry to bother you again. 
May I ask you to share this project with me?
This is very interesting and important to me.
Thank you!

RalphBacon, 2021-01-18:
There is now an Arduino ported version here:
https://github.com/RalphBacon/205-Internet-Radio

See the Arduino IDE Version folder.

Aleks-Ale, 2021-01-18:
Thank you SO MUCH! I will see right now!

Aleks-Ale, 2021-01-22:
Dear Ralf,
Could you check it out:

stationList.txt : Is necessary to have a comma at the end of each line except the last.
stationSelectHelper.h : swap port and path

```
typedef struct radioStationLayout
{
	std::string host;
        uint16_t port;
	std::string path;
	std::string friendlyName;
	uint8_t useMetaData;

```
Thank you.

RalphBacon, 2021-01-22:
1. You do not need a comma at the end of each line. This is an extract of my current list:
"stream.antenne1.de",80,"/a1stg/livestream1.aac","Antenne1.de","1"
"bbcmedia.ic.llnwd.net","80","/stream/bbcmedia_radio4fm_mf_q","BBC Radio 4",0
"stream.antenne1.de",80,"/a1stg/livestream2.mp3","Antenne1 128k",1

2. The only way to close the station list without changing station is to find the (highlighted) current station and choose that, as you have found out. I will probably change that behaviour in the future with a simple cancel button at the foot of the list (near page X of y). All a work in progress!

tadder, 2021-01-22:
Hi Ralph,
Your station lister addition is fortuitous as I have been attempting to add an extra station to the code as follows.
//8
  "internetradio.com",
  "/virgin",
  80,
  "Virgin Radio",
  1,
The lister renders this code redundant, so I will concentrate on adding the lister. Are there any guidelines I should follow when adding a new station?

RalphBacon, 2021-01-22:
The format of the station listing is pretty straightforward:

<host>,<port>,<path>,<friendly name>,<use MetaData>

Quotes are optional, it's the commas that matter (so ensure you don't have any, except to delimit the fields)

http is assumed (so don't add that as part of the host. 
Port is mostly 80 but might be something else. 
Friendly name is what you see on screen, so don't make it too long. 
Metadata (0=no, 1=yes) will cause the station to fail if it does not transmit Metadata but you have put a 1 here - that said, the station SHOULD still honour that setting (and just send 0 byte metadata) but some don't seem to follow the rules. Always try it out on a browser (or Winamp) if you have problems to see whether track/artist information is being sent out. 

Note that the absence of Track/Artist does not mean they don't send out metadata: I'm listening to a radio station right now that SHOULD send out metadata (and does, but only 0 byte markers, no actual track info) and when I queried it with them they said they were running off a backup server that doesn't send out that info but their main one (when repaired) will do so again!

Aleks-Ale, 2021-01-22:
> 1. You do not need a comma at the end of each line.
Unfortunately, I am not an expert in programming and I do not know why we have different results.
I see what I see :)  This is the log if you run the sketch without any changes.

```

[I][tftHelpers.h:148] setupDisplayModule(): TFT Initialised
[D][ESP32-WROVER_Web_Radio.ino:79] setup(): Starting player
[V][tftHelpers.h:484] displayTrackArtist(): No delimiter found - using default value
[D][ESP32-WROVER_Web_Radio.ino:84] setup(): Waiting for VS1053 initialisation to complete.
[D][ESP32-WROVER_Web_Radio.ino:91] setup(): Switch player to MP3 mode
[V][tftHelpers.h:484] displayTrackArtist(): No delimiter found - using default value
[V][tftHelpers.h:484] displayTrackArtist(): No delimiter found - using default value
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'stream.antenne1.de'
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'80'
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'/a1stg/livestream1.aac'
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'Antenne1.de'
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'1bbcmedia.ic.llnwd.net'
[V][stationSelectHelper.h:474] addStationToArray(): Vector size:1
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'80'
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'/stream/bbcmedia_radio4fm_mf_q'
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'BBC Radio 4'
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'0stream.antenne1.de'
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'80'
[V][stationSelectHelper.h:474] addStationToArray(): Vector size:2
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'/a1stg/livestream2.mp3'
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'Antenne1 128k'
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'1listen.181fm.com'
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'80'
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'/181-beatles_128k.mp3'

[I][stationSelectHelper.h:446] populateStationList(): Radio stations imported successfully.
[D][stationSelectHelper.h:456] populateStationList(): 0 - stream.antenne1.de /a1stg/livestream1.aac 80 (Antenne1.de) YES
[D][stationSelectHelper.h:456] populateStationList(): 1 - 80 BBC Radio 4 0 (0stream.antenne1.de) NO
[D][stationSelectHelper.h:456] populateStationList(): 2 - /a1stg/livestream2.mp3 1listen.181fm.com 0 (80) NO
[D][stationSelectHelper.h:456] populateStationList(): 3 - Beatles 128k 80 1 (/magicmellow.mp3) NO
[D][stationSelectHelper.h:456] populateStationList(): 4 - 1edge-bauermz-03-gos2.sharp-stream.com /net2national.mp3 80 (Greatest Hits 112k (National)) YES
[D][stationSelectHelper.h:456] populateStationList(): 5 - 8024 Mowtown Magic Oldies 0 (1live-bauer-mz.sharp-stream.com) NO
[D][stationSelectHelper.h:456] populateStationList(): 6 - /magicmellow.aac 1stream.btsstream.com 0 (8000) NO

```

As you can see, the sections are formed incorrectly. Now I put commas at the end of the line. 
Parsing completes correctly and creates a list of stations.

```
[V][tftHelpers.h:484] displayTrackArtist(): No delimiter found - using default value
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'stream.antenne1.de'
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'80'
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'/a1stg/livestream1.aac'
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'Antenne1.de'
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'1'
[V][stationSelectHelper.h:474] addStationToArray(): Vector size:1
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'bbcmedia.ic.llnwd.net'
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'80'
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'/stream/bbcmedia_radio4fm_mf_q'
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'BBC Radio 4'
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'0'
[V][stationSelectHelper.h:474] addStationToArray(): Vector size:2
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'stream.antenne1.de'
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'80'
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'/a1stg/livestream2.mp3'
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'Antenne1 128k'
[V][stationSelectHelper.h:399] populateStationList(): Parameter:'1'

[I][stationSelectHelper.h:446] populateStationList(): Radio stations imported successfully.
[D][stationSelectHelper.h:456] populateStationList(): 0 - stream.antenne1.de /a1stg/livestream1.aac 80 (Antenne1.de) YES
[D][stationSelectHelper.h:456] populateStationList(): 1 - bbcmedia.ic.llnwd.net /stream/bbcmedia_radio4fm_mf_q 80 (BBC Radio 4) NO
[D][stationSelectHelper.h:456] populateStationList(): 2 - stream.antenne1.de /a1stg/livestream2.mp3 80 (Antenne1 128k) YES
[D][stationSelectHelper.h:456] populateStationList(): 3 - listen.181fm.com /181-beatles_128k.mp3 80 (Beatles 128k) YES
[D][stationSelectHelper.h:456] populateStationList(): 4 - stream-mz.planetradio.co.uk /magicmellow.mp3 80 (Mellow Magic (Redirected)) YES
[D][stationSelectHelper.h:456] populateStationList(): 5 - edge-bauermz-03-gos2.sharp-stream.com /net2national.mp3 80 (Greatest Hits 112k (National)) YES
[D][stationSelectHelper.h:456] populateStationList(): 6 - airspectrum.cdnstream1.com /1302_192 8024 (Mowtown Magic Oldies) YES
```

RalphBacon, 2021-01-23:
Make sure you've taken the complete set of code in my GitHub - this looks like a previous codeset.

tadder, 2021-01-23:
I have uploaded the latest Arduino Version and followed the instruction fully. As you can see from the attached photo it is working up to a point. However, I suspect there are issues with reading the ‘stationList’ file. The first in the list ‘Antenne1.de’ plays with no problem but when changing station there appears to be a loss of sync (see 2 Ostream.antenne1 which should be radio 4. 3 is picking up the port number instead of the friendly name). Selecting some reports ‘Connecting’ and hangs others report ‘Could not connect to this station’ and allow further connection. Either way only the first station works.

As an aside, discovering radio station URL’s is difficult. I wonder if this is a ploy to get us to use ‘commercial’ WEB radios with inbuilt station lists?

As a separate exercise I also tried enabling PSRAM which results in ‘almost’ continuous reboots. I tried to get a copy of the output using the Arduino serial monitor, but was unsuccessful (though monitor appears to work when PSRAM is disabled).

[image]
[image]
[image]

[image]

[image]

Aleks-Ale, 2021-01-23:
> I have uploaded the latest Arduino Version

I wrote about this problem above. 
Just add a comma at the end of every line except the last line in the station list file.
I think that Ralph doesn't have such a error.

```
"live-bauer-mz.sharp-stream.com",80,"/net2lincoln.aac",Greatest Hits (Lincoln),1,
ais-edge09-live365-dal02.cdnstream.com,80,/a52107?listenerId=esAdblock0650048,Heckington Community Radio,1,
live-bauer-mz.sharp-stream.com,80,/net1lincoln.aac,Radio Lincs FM,1

```

RalphBacon, 2021-01-23:
I've just uploaded **a new version v1.19** for the Arduino IDE. 
Remember to also update/upload the **data partition** on your ESP32 to ensure the stationList is in the right format. 
The **circular buffer** will not work above about 10000 bytes without a change to the underlying ESP32 Circular Buffer file (which I can document, but it's a bit hacky). Anyone interested?

RalphBacon, 2021-01-23:
Compare your code to the picture in the readme.md file for the Arduino IDE.

Just to be clear, you should NOT need a trailing comma at the end of the stationList.txt file if you are using the most recent codebase.

Aleks-Ale, 2021-01-23:
> Anyone interested?

Yes, of course! Thank you!

RalphBacon, 2021-01-23:
Instructions on how to enable the circular buffer on ESP32-WROVER boards added:

https://github.com/RalphBacon/205-Internet-Radio/tree/main/Arduino%20IDE%20Version

tadder, 2021-01-23:
Thanks for that, I didn’t fully understand that you were describing the same problem. You are indeed right, I have now terminated each line with a comma and they now all work. I am using the Arduino code so I wonder if Ralph doesn’t get this issue because he is using PlatformIO? Just a thought.

I am now attempting to add an extra line at the end as follows:

internet-radio.com,80,/station/xrds/,XRDS.fm Clarksdale,0

This results in “Connected Reading stream…” and no sound. I couldn’t find the Port number so guessed at 80, I wonder if this might be the issue?

Aleks-Ale, 2021-01-23:
> ESP32-WROVER boards
I understood. I used ESP 32 WROOM without SPRAM. I expect delivery of WROVER soon.

RalphBacon, 2021-01-23:
You CAN use the SP32-WROOM quite successfully, just be aware that you might experience more "drop outs".

Use the new code I posted today and get that station list issue sorted out! Otherwise you are hacking, or working around, some sort of weird conflict (perhaps you did not upload the Data folder again?). Whatever, new version is there.

Aleks-Ale, 2021-01-23:
> new version is there

Thanks. Please tell me, for what reasons do you use LITTLEFS instead of the standard SPIFFS ? Is it possible to use SPIFFS in your project? I just replaced LITTLEFS -> SPIFFS in the program text . I use a single ESP32 chip for a couple of current projects that use SPIFFS.

tadder, 2021-01-23:
Software V1.19 now being used. I am using stationList as is (no extra comma’s).

- Libaries ARDUINO_LOG, ESP_vs1053_Library, TFT_eSPI-2.2.23 updated by copying direct to libaries folder in sketch folder (overwrite).
- Data folder content upgraded via ESP32 Sketch Data Upload.
- Code now stops at ‘Starting MP3 Decoder’
- 
- So I believe I have carried out the procedure correctly, but no doubt I am missing something.

MONITOR OUTPUT:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

[E][ESP32-WROVER_Web_Radio.ino:27] setup(): Version v1.210123_19 - 2021-01-23 14:29:43.815057
[image]

Aleks-Ale, 2021-01-23:
I see the same problem as Tudder.
Is it possible that we are making the same mistake?
The new version 1.19 is installed.
But you still need a comma at the end of the lines in the list of stations.
Without a comma, the list of stations is formed incorrectly. And I see the same offset of station names in the list.
It doesn't matter to me. It works anyway. 
This is a great tutorial on how to work with the touch screen and organize your code.

[image]

RalphBacon, 2021-01-23:
If it stops at "Starting MP3 Decoder" just it's the reboot on the device. This only happens to me (sometimes) after a new code upload, never at other times. You are powering board from separate +5v supply, not via USB port, right?

RalphBacon, 2021-01-23:
I will look at station list offset tomorrow, i know the problem but I have fixed it, perhaps I have not given you the new library. Only cosmetic!

Aleks-Ale, 2021-01-23:
> i know the problem

No problem, I just use

row[nthRow].setLabelDatum(-140,

tadder, 2021-01-23:
> 
> 
> If it stops at "Starting MP3 Decoder" just press the reset button on the device. This only happens to me (sometimes) after a new code upload, never at other times. You are powering board from separate +5v supply, not via USB port, right?

Yes, I confirm I am powering the board from a seperate 5V PSU. Unfortunately pressing the reset button just cycles back to the same "Starting MP3 Decoder message.

tadder, 2021-01-23:
Aleks_Ale - Take a look at Ralphs Video #203 it contains a great introduction to LittleFS.

Aleks-Ale, 2021-01-24:
> Video #203

Thanks! I found it now! :)

RalphBacon, 2021-01-24:
Regarding the **on-screen station list being offset,** please check the following:

**library**: TFT_eSPI-2.2.23
**file**: Button.cpp

Line 82/83 was: 
   `   _gfx->drawString(long_name, _x1 + (_w/2) + _xd, _y1 + (_h/2) - 4 + _yd);`

Now should be (with above line commented out): 
```
// RSB to correct centre alignment regardless of DATUM
_gfx->drawString(long_name, _x1 + _xd, _y1 + (_h/2) - 4 + _yd);

```
This is the correction that works on Arduino IDE and PlatformIO.

RalphBacon, 2021-01-24:
Regarding the **stationList.txt** and the strange comma at the end of the line, please confirm/check the following:

Open the file in an editor that can show you HEX characters (eg notepad++)

At the end of each station line, the last two characters should be 0D 0A (carriage return, line feed). If you don't use Windows it might be different. The order of those characters is important.

"stream.antenne1.de",80,"/a1stg/livestream1.aac","Antenne1.de","1"

22 73 74 72 65 61 6d 2e 61 6e 74 65 6e 6e 65 31 2e 64 65 22 2c 38 30 2c 22 2f 61 31 73 74 67 2f 6c 69 76 65 73 74 72 65 61 6d 31 2e 61 61 63 22 2c 22 41 6e 74 65 6e 6e 65 31 2e 64 65 22 2c 22 31 22 **0d 0a**

If the last two characters are not as shown (in bold) then we have found the issue. Re-edit the file with an editor that puts those characters in there.

tadder, 2021-01-24:
I have reloaded V1.19 but it still hangs at “Starting MP3 Decoder” as described yesterday.
Any thoughts?

RalphBacon, 2021-01-24:
This means the VS1053 is not initialising. 

Check the orientation of the cable going from the VS1053 to the motherboard. Specifically, ensure the 5v connections on the VS1053 go to the correct pins on the motherboard. If the cable points away from the VS1053 then it should do the same on the motherboard. If the cable runs across the VS1053 then it will connect to the motherboard with the cable running right next to the socket (a tight fit - see the pictures here: [https://github.com/RalphBacon/205-Internet-Radio/tree/main/ESP32-Web-Radio-Construction](url))

If you disconnect the cable from the VS1053 it will probably continue booting up but, obviously, you will hear no sound.

RalphBacon, 2021-01-24:
I've added a picture of the preferred way of connecting the VS1053:
[https://github.com/RalphBacon/205-Internet-Radio/tree/main/Arduino%20IDE%20Version/ESP32-WROVER_Web_Radio/images](url)
It's in the images folder for the Arduino IDE version.

tadder, 2021-01-24:
Thanks for your assistance Ralph. The board had been working prior to loading V1.19 so to double check I reloaded this prior version and it works. I have checked the wiring but I suspect you would agree in view of the old version working it’s unlikely to be a hardware issue. As you can see the VS1053 is mounted as per your suggestion.

[image]

RalphBacon, 2021-01-24:
I wasn't aware you had it working prior to the current version! 

Go to your sketch, find the setup() routine and find these lines:
[image]

Comment out those lines and add:
`player.begin();`

This is how it was previously (it doesn't check the result from the call). See how far you get.

Aleks-Ale, 2021-01-24:
> the last two characters should be 0D 0A

Yes, you right. This is your original stationList.txt
[image]

Aleks-Ale, 2021-01-24:
> _gfx->drawString(long_name, _x1 + _xd, _y1 + (_h/2) - 4 + _yd);

Thanks! Now correct.
[image]

RalphBacon, 2021-01-24:
@Aleks-Ale  I can only assume that the stationList.txt you show there must be an early version. This is the problem trying to keep two code bases in-line (not forgetting non-code files)!

At least the station list now lines up correctly on-screen, excellent.

Aleks-Ale, 2021-01-24:
> I can only assume that the stationList.txt you show there must be an early version.

Is this the correct download path?
https://github.com/RalphBacon/205-Internet-Radio
I just downloaded this archive again.
[image]

tadder, 2021-01-24:
OK, that worked, a couple of steps further, it now outputs the welcome message and reads the station list. However it will only connect to the first station in the list, so perhaps I am now experiencing similar issues to Aleks? I will re-visit the thread you have had with him a see if this helps. Thanks as always, looks as if I might actually have this working by the end of the day.

Aleks-Ale, 2021-01-24:
> I can only assume that the stationList.txt you show there must be an early version.

I found a solution to this problem. You just need to open the stations file in MS Word and save it again. 
After that, the correct codes appear at the end of the line.

[image]

Aleks-Ale, 2021-01-24:
I'm really looking forward to seeing how you solve the issue of returning from the list without changing the station.

RalphBacon, 2021-01-24:
Interesting about having to edit that stationList.txt file. My (local) GitHub shows the correct line endings. Perhaps it is the way GitHub processes this? Are you using Windows 10? something for me to bear in mind.

Returning from the station list (without click the current station again) is probably going to be done by touching the Page X of Y area (simple but not so obvious). I shall think about it some more. Perhaps a separate "Return Icon" button would be better.

RalphBacon, 2021-01-24:
@tadder  When you say it "only connects to the first station", does that mean when you select another one it tries to connect by produces no sound? 

Edit the stationList.txt in the way that @Aleks-Ale suggests above and retry.

Aleks-Ale, 2021-01-24:
> Are you using Windows 10?

Yes.

> Perhaps a separate "Return Icon" button would be better.

I think this is the best way.

tadder, 2021-01-24:
It just displays 'Connecting' with no sound of course. You have to wait for this process to time out (or press reset) in order to proceed. It then reports 'Could not connect to this station'
I note that their are 16 stations in the list but only 13 are read to the board. I believe I am downloading to LittleFS because its reading WiFiSecrets OK.
I will edit the list as you suggest.

RalphBacon, 2021-01-24:
**FIXED: stationList.txt line endings**

GitHub, in its infinite wisdom, only appends a LF at the end of the text file lines (and doesn't tell me it's now different to the one I have on my GitHub desktop). Apparently this is a *nix default.

I've now added a configuration file to my repository so that .txt files will always have CRLF at the end of every line.

I've tested it and now they text files look like this after downloading the GitHub zip and unzipping it:
[image]

Aleks-Ale, 2021-01-24:
Great! Thanks!

tadder, 2021-01-24:
Looking good. Opened in WordPad and this added CR. So each line now ends with CRLF. Other than lines 12 and 13 it now works well. Will now have a look at Ralphs revised stationlist.

RalphBacon, 2021-01-26:
As it all seems stable now I'll close this issue, thanks to everyone highlighting the issue which I would never have found as I don't download my own GitHub! (Maybe I should in the future).

closed 2021-01-26 by RalphBacon