RalphBacon/205-Internet-Radio #3: Bug during touch calibration data read

watts-on-root, 2021-01-09:
Whilst trying to figure out the format of the touch calibration data file I noticed a bug in file "tfthelpers.h" at line 40:

			uint16_t calData[5];
			if (f.readBytes((char *)calData, 14) == 14)

A 10 byte buffer (5 x 16-bit) is allocated, but then 14 bytes are being read into it. I'm not sure where the value 14 comes from as I assume 10 is the correct value.

RalphBacon, 2021-01-09:
Yes, this was weird, and I knew it was not logical but I exported the generated TFT calibration file and it had 14-bytes of info (if my memory serves)  If you run the same sketch from the TFT_eSPI-2.2.23 library you can then either export or just check  what it produces. I've attached the source file I hacked to get it to print the values. (I've add to rename it to a .txt type, it's really a .cpp)
[main.cpp.txt](https://github.com/RalphBacon/205-Internet-Radio/files/5791172/main.cpp.txt)

watts-on-root, 2021-01-13:
Yeah, it sure is odd. Maybe once upon a time, the cal data was stored as ascii data, which would be more bytes.
Anyway I've changed my code to only read the 10 bytes as otherwise it is for sure going the trample 4 other bytes of memory, which could hold some other unrelated variable(s).

RalphBacon, 2021-01-16:
I've done the same. I didn't realise I hadn't created the buffer large enough for 14 bytes (oops). But now I only read 10 bytes, the TFT keeps working as it always did, and there's no danger of memory corruption. In the fullness of time I will discover why the TFT calibration sketch stores 14 bytes.

RalphBacon, 2021-01-24:
I've tested this with just the (first) 10 bytes and all seems well, so no further action required.

closed 2021-01-24 by RalphBacon