RalphBacon/205-Internet-Radio #4: https

Aleks-Ale, 2021-01-10:
I used the following changes and these changes allow me to listen to https stations and http stations that have mirrored https.

```
main.h
#include <WiFiClientSecure.h>
WiFiClientSecure client;
```

Unfortunately this does not work if the station is http only.
Also I tried to make such improvements, but unfortunately it does not work.
WiFiClient client does not switch to WiFiClientSecure client
Any ideas?

**main.h**
```
#include <WiFiClientSecure.h>
WiFiClientSecure client;
bool http_s = true;
struct radioStationLayout
{
char host[64];
char path[128];
int port;
bool https;       // http=0 | https=1 <-----------!
char friendlyName[64];
uint8_t useMetaData;
};
```

**ESP32-WROVER_Web_Radio.ino**

```
void setup
{
http_s = preferences.getUInt("http_s", 0);
}

bool stationConnect(int stationNo)
{
if (http_s) {
WiFiClientSecure client;
int redirectedPort = 443;
}
else {
WiFiClient client;
int redirectedPort = 80; 
}

http_s = radioStation[stationNo].https;
...
preferences.putUInt("http_s", http_s);
...
}
```
*These are not complete changes just to demonstrate the idea.

RalphBacon, 2021-01-11:
That's a good start, but as you have discovered you have to end the current Http session and start a new one (either secure or not) to get https / http. This can be done via the StationConnect function, if we store the flag in the station information (ie whether that station requires https or http). 
It's something I can put on the improvements list but so far I've not found an https-only station - but they are bound to exist as http (not secure) is withdrawn by Google).
I wonder if they all transmit on port 443 (the http version have various ports, not always 80).

Aleks-Ale, 2021-01-11:
I use two stations for my tests:

```
//10
"radio-nice.ru",
"/stream",
8285,
0,
"http_S RADIO NICE LOFI",
0,

//11
"radio-nice.ru",
"/radio",
5110,
1,
"http RADIO NICE DEEP",
0,
```
it seems to me that this method has a better perspective of use.

main.h

```
WiFiClient* client = (http_s) ? new WiFiClient() : new WiFiClientSecure();
//WiFiClient client;
```

I tried different possibilities of using this method in the ESP32-WROVER_Web_Radio.ino. 
But unfortunately I don't understand very well what I was doing. Therefore, there is no success. 
Probably, when switching stations, the client type does not change.

```
void changeStation(int8_t btnValue)

  if (prevStnNo != nextStnNo)
  {
    prevStnNo = nextStnNo;
 client_stop ();
 WiFiClient* client = (http_s) ? new WiFiClient() : new WiFiClientSecure();
```

But my intuition tells me that this has a good perspective. 
I need help here please.

Aleks-Ale, 2021-01-11:
The first problem is how to restore http_s from EEPROM in main.h for the correct start of the last station.

main.h

```
Preferences preferences;
bool http_s = true;
  preferences.begin("WebRadio", false);
  http_s = preferences.getUInt("http_s", 0);
WiFiClient* client = (http_s) ? new WiFiClient() : new WiFiClientSecure();
```
main.h:45:3: error: 'preferences' does not name a type
   preferences.begin("WebRadio", false);
   ^
main.h:46:3: error: 'http_s' does not name a type
   http_s = preferences.getUInt("http_s", 0);

Aleks-Ale, 2021-01-11:
> This can be done via the StationConnect function

_Step 1:_

ESP32-WROVER_Web_Radio.ino - No any changes
**main.h**

```
bool http_s = 0;  //For manual switch to unsecure WiFiClient
const int8_t stationCnt = 1;
// Only one HTTP station
struct radioStationLayout radioStation[stationCnt] = {
"radio-nice.ru",
"/radio",
5110,
0,
"http RADIO NICE DEEP",
0,
};
WiFiClient* client = (http_s) ? new WiFiClientSecure() : new WiFiClient();
```
It starts and plays normally.

```
http_s: 0
Current station number: 0
--------------------------------------
        Connecting to station 0
--------------------------------------
http_s_stationConnect: 0
Free memory: 160768 bytes
METADATA ignored for this radio station.
Host: radio-nice.ru Port:5110
Connected to radio-nice.ru (http RADIO NICE DEEP)
Getting data from /radio (WITHOUT Metadata)
Waiting for header data
Bit rate:192
Restored screen brightness to 115
Free memory: 156296 bytes
--------------------------------------
```

_Step 2:_

**ESP32-WROVER_Web_Radio.ino**

```
bool stationConnect(int stationNo)
{
 client->stop (); // Del multiple definition for client 
 http_s = 0;  // For manual test

 WiFiClient* client = (http_s) ? new WiFiClientSecure() : new WiFiClient();
}
```
 Unfortunately I saw an error.

```
http_s: 0
Current station number: 0
--------------------------------------
        Connecting to station 0
--------------------------------------
http_s_stationConnect: 0
Free memory: 160712 bytes
METADATA ignored for this radio station.
Host: radio-nice.ru Port:5110
Connected to radio-nice.ru (http RADIO NICE DEEP)
Getting data from /radio (WITHOUT Metadata)
Waiting for header data
Bit rate:192
Restored screen brightness to 115
Free memory: 156252 bytes
Client not connected.      <-------------
Connecting to SSID: brg
Disconnecting from WiFi
Setting ESP32 to STA mode
Setting ESP32 to NOT store credentials in NVR
Setting ESP32 to NOT allow sleep
Setting ESP32 to NOT auto re-connect[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 8 - ASSOC_LEAVE
Initiating connection with WiFi.
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.115, MASK: 255.255.255.0, GW: 192.168.1.1
Waiting for WiFi connection...
WiFi connected with (local) IP address of: 192.168.1.115
--------------------------------------
        Connecting to station 0
--------------------------------------
http_s_stationConnect: 0
Free memory: 151348 bytes
METADATA ignored for this radio station.
.....

```
I do not know what to do.

RalphBacon, 2021-01-11:
There is no need to store this information in EEPROM. When a station change occurs, read the station info (eg host, path, port, metaData AND http format). Then retain the existing webclient if you can, or close it and open a secure one.
The error you highlight is what happens when a connection fails - it retries again, so it is probably retrying using standard http.

Aleks-Ale, 2021-01-11:
My brain is exploding. I give up!
My knowledge is not enough for this.
Are you interested in solving this problem?

RalphBacon, 2021-01-24:
I will look at this but it will be several weeks (if not months) before I get my new workshop setup and that is after moving home (happening any day now).

Keep tuned for an update around Easter (aspirational date).

closed 2021-01-24 by RalphBacon