RalphBacon/205-Internet-Radio #5: Editing the Station List

tadder, 2021-01-24:
I am attempting to edit the contents of stationList loaded in LittleFS. I have added an extra line at the end as follows:

internet-radio.com,80,/station/xrds,XRDS.fm Clarksdale,0

This is picked up on the screen but there is no sound only the message “Connected. Reading stream...”

The info comes from the internet-radio web page. I have guessed at the port number because so far I haven't found it specified anywhere.

Have others found a process for adding new stations that has better success?

`"stream.antenne1.de",80,"/a1stg/livestream1.aac","Antenne1.de","1"
"bbcmedia.ic.llnwd.net","80","/stream/bbcmedia_radio4fm_mf_q","BBC Radio 4",0
"stream.antenne1.de",80,"/a1stg/livestream2.mp3","Antenne1 128k",1
"listen.181fm.com",80,"/181-beatles_128k.mp3","Beatles 128k",1
"stream-mz.planetradio.co.uk",80,"/magicmellow.mp3","Mellow Magic (Redirected)",1
"edge-bauermz-03-gos2.sharp-stream.com",80,"/net2national.mp3","Greatest Hits 112k (National)",1
"airspectrum.cdnstream1.com",8024,"/1302_192","Mowtown Magic Oldies",1
"live-bauer-mz.sharp-stream.com",80,"/magicmellow.aac","Mellow Magic (48k AAC)",1
"stream.btsstream.com",8000,"/seaac","South East Radio (AAC 64k)",1
"uk2.internet-radio.com",8024,"/",UK Dance Radio,1
"us4.internet-radio.com",8197,"/stream?type=http&nocache=72","EZ Hits South FLA",0
s1.slotex.pl,7274,"/",Radio Kehlkopf,0
204.187.100.18,1485,"/",De Natchegaal,1
"live-bauer-mz.sharp-stream.com",80,"/net2lincoln.aac",Greatest Hits (Lincoln),1
ais-edge09-live365-dal02.cdnstream.com,80,/a52107?listenerId=esAdblock0650048,Heckington Community Radio,1
live-bauer-mz.sharp-stream.com,80,/net1lincoln.aac,Radio Lincs FM,1
internet-radio.com,80,/station/xrds,XRDS.fm Clarksdale,0`

RalphBacon, 2021-01-24:
Does the buffer show anything other than 0%? If it shows zero then nothing is coming down the line (possibly due to incorrect port). Try typing in the full (ie concatenated) url into your browser. If that doesn't start playing music (wihtout having to click anything) then that's the reason.

This web page has many stations:
https://www.internet-radio.com/

This is UK based:
http://www.radiofeeds.co.uk/mp3.asp

Sometimes you have to download the .pls file and look inside that to get the URL - but always test first on a browser (or WinAmp, even better).

tadder, 2021-01-24:
Thanks for that, internet-radio.com is in fact where I got the first station I tried from.
The latest line I have tried is as follows:
	www.internet-radio.com,80,/station/majesticjukebox/,Majestic Jukebox,1
With the 1 at the end I get ‘No MetaData interval found’ 0%
changing to 0 at the end I get ‘Connected Reading stream’ 1%
I tried with and without WWW.

I put internet-radio.com/station/majesticjukebox/ in my browser which changes it to:
https://www.internet-radio.com/station/majesticjukebox/ it then plays OK in the browser

RalphBacon, 2021-01-25:
That might the problem:

[image]

My ESP32 Web Radio does not support secure (https) addresses at this time; something I may look at in the future. You need to see if they mirror the stream with a standard http (port 80) url. Most commercial radio stations do, because they want you to listen to them (and the adverts).

RalphBacon, 2021-01-25:
@tadder 
I looked at this station and the WinAmp playlist (,pls) shows that the station translates to:
uk3.internet-radio.com,8405,/live,Majestic Jukebox,1
[image]

Plays OK. No https required.

tadder, 2021-01-25:
Ralph, That great, so my final issue was down to how I interpreted the URL. On Internet-radio.com I just click on the .pls button and read how it's translated. I don’t know if this works for all stations but I have added three so far and they all work. Really pleased.
Some stutter a little so I assume I need to get PSRAM up and running to see if that helps.
Thanks again, I couldn’t have done it without your support.

RalphBacon, 2021-01-26:
@tadder 
Yes, downloading the WinAmp .pls file will (should) always show one or more URLs that only require http. 

I'll close this issue (for now) but in the fullness of time I want to be able to parse the entire URL (splitting this way was the first step). 

In the even longer term I want to be able to upload the URLs without having to upload the entire "data" folder to LittleFS - but that's some way off.

Let me know if you can follow the instruction on the GitHub to getting the circular buffer running (open a different issue if you get stuck).

closed 2021-01-26 by RalphBacon