RalphBacon/205-Internet-Radio #6: Loop logging  TFT Calibration Data

rjshiffer, 2021-01-26:
Problem:  Infinite for loop while logging TFT Calibration Data

Environment: Arduino IDE
Hardware:  ESP-WROOM-32 + ILI9341  + VS1053

See file tftHelpers.h, near line 47 and 72:
```
// There are 10 bytes (5 integers @ 2 bytes each) for TFT calibration
#define totCalibrationBytes 10
.
.
.
// Buffer (integer 2 bytes * 5) to hold calibration data
uint16_t calData[5];

// On calibration, 14 bytes were written. TODO: why?
//if (f.readBytes((char *)calData, 14) == 14)
if (f.readBytes((char *)calData, totCalibrationBytes) == totCalibrationBytes)
	tft.setTouch(calData);

log_v("TFT Calibration data:");
for (auto cnt = 0; cnt < totCalibrationBytes; cnt++)
{
	log_v("%04X, ", calData[cnt]);
}

f.close();
log_i("Touch Calibration completed");
```
Notice that calData[5] array only has 5 elements, but the for loop will
end after "totCalibrationBytes" (10+1 for 0 start) loops.  The reference to calData[cnt]
refers to uint16_t (2 bytes), so the loop only needs to execute 5 times.
```
17:42:51.245 -> [E][ESP32-WROVER_Web_Radio.ino:34] setup(): Version v1.210123_19 - 2021-01-23 14:29:43.815057
17:42:51.292 -> [D][ESP32-WROVER_Web_Radio.ino:42] setup(): Total heap: 331196
17:42:51.292 -> [D][ESP32-WROVER_Web_Radio.ino:43] setup(): Free heap: 294856
17:42:51.292 -> [D][ESP32-WROVER_Web_Radio.ino:44] setup(): Total PSRAM: 0
17:42:51.292 -> [D][ESP32-WROVER_Web_Radio.ino:45] setup(): Free PSRAM: 0
17:42:51.292 -> [D][ESP32-WROVER_Web_Radio.ino:46] setup(): Used PSRAM: 0
17:42:51.292 -> [D][ESP32-WROVER_Web_Radio.ino:55] setup(): Starting SPI
17:42:51.292 -> [D][ESP32-WROVER_Web_Radio.ino:59] setup(): Free memory: 294732 bytes
17:42:51.292 -> [I][ESP32-WROVER_Web_Radio.ino:70] setup(): LITTLEFS system mounted SUCCESSFUL.
17:42:51.339 -> [V][tftHelpers.h:80] initDisplay(): TFT Calibration data:
17:42:51.339 -> [V][tftHelpers.h:84] initDisplay(): 0183, 
17:42:53.324 -> [V][tftHelpers.h:84] initDisplay(): 0D96, 
17:42:55.340 -> [V][tftHelpers.h:84] initDisplay(): 00FA, 
17:42:57.320 -> [V][tftHelpers.h:84] initDisplay(): 0D86, 
17:42:59.306 -> [V][tftHelpers.h:84] initDisplay(): 0007, 
17:43:01.307 -> [V][tftHelpers.h:84] initDisplay(): 7304, 
17:43:03.336 -> [V][tftHelpers.h:84] initDisplay(): 3F40, 
17:43:05.320 -> [V][tftHelpers.h:84] initDisplay(): 0000, 
17:43:07.337 -> [V][tftHelpers.h:84] initDisplay(): 0000, 
17:43:09.315 -> [V][tftHelpers.h:84] initDisplay(): 0000, 
17:43:11.298 -> [V][tftHelpers.h:84] initDisplay(): 0000, 
17:43:13.329 -> [V][tftHelpers.h:84] initDisplay(): 0000, 
17:43:15.313 -> [V][tftHelpers.h:84] initDisplay(): 0000, 
17:43:17.334 -> [V][tftHelpers.h:84] initDisplay(): 8CE4, 
17:43:19.305 -> [V][tftHelpers.h:84] initDisplay(): 3FFB, 
17:43:21.333 -> [V][tftHelpers.h:84] initDisplay(): 8C34, 
17:43:23.320 -> [V][tftHelpers.h:84] initDisplay(): 3FFB, 
17:43:25.343 -> [V][tftHelpers.h:84] initDisplay(): 1EE0, 
17:43:27.324 -> [V][tftHelpers.h:84] initDisplay(): 3FFB, 
17:43:29.339 -> [V][tftHelpers.h:84] initDisplay(): 1EC0, 
.
.
.
.
```
Changing the for loop code to:
```
for (int cnt = 0; cnt <= 4; cnt++)
{
   log_v("%04X, ", calData[cnt]);
}
```
results in:
```
17:46:27.370 -> [I][ESP32-WROVER_Web_Radio.ino:70] setup(): LITTLEFS system mounted SUCCESSFUL.
17:46:27.418 -> [V][tftHelpers.h:80] initDisplay(): TFT Calibration data:
17:46:27.418 -> [V][tftHelpers.h:91] initDisplay(): 0183, 
17:46:27.418 -> [V][tftHelpers.h:91] initDisplay(): 0D96, 
17:46:27.418 -> [V][tftHelpers.h:91] initDisplay(): 00FA, 
17:46:27.418 -> [V][tftHelpers.h:91] initDisplay(): 0D86, 
17:46:27.418 -> [V][tftHelpers.h:91] initDisplay(): 0007, 
17:46:27.418 -> [I][tftHelpers.h:95] initDisplay(): Touch Calibration completed
```
I have no idea why it loops, but certainly referring to elements beyond array
size is an error.  Also, can't help with the 10 bytes vs 14 bytes question.

Also, I have the same problem with this code found in the utility program that
writes the /TouchCalData1.txt file to the LITTLEFS.
```
Serial.println("Calibration data (14 integers of two bytes each):");
for (int cnt = 0; cnt < 14; cnt++)
{
	Serial.printf("%d. ", cnt);
	Serial.printf("%04X\n", calData[cnt]);
}
```
Ralph, I don't remember where I got the utility code to write the
TouchCalData1.txt file, but I think I may have gotten it from your comments
somewhere.  I couldn't find it on your Github (which would have helped with
may confusion as to how to create the file.)

The project is great.  I have your PCBs ready and I'm just waiting for my
TTGO T8 v1.7 to arrive.

RalphBacon, 2021-01-26:
@rjshiffer 
I had issues (very occasionally) when I tried to read 14 bytes into a 10-byte buffer (doh!) but I thought my change had fixed it.

However, you tracing shows that it must be reading 10 bytes (5 integers of 2 bytes). Reading more probably just corrupts the loop counter (as you found out).

I'll look at this again.

rjshiffer, 2021-01-26:
The change with Version V1.29 only prints 2 calData entries.

Version v1.210126_29 - 2021-01-26 10:39:16.425885

```
log_v("TFT Calibration data:");
for (auto cnt = 0; cnt < totCalibrationBytes / sizeof(int); cnt++)
{
    log_v("%04X, ", calData[cnt]);
}
```
The sizeof(int) is 4, but calData is type uint16_t which is 2 bytes. 10/4 = 2  (truncated)

This works:
```
for (auto cnt = 0; cnt < totCalibrationBytes / sizeof(uint16_t); cnt++)
```

``

RalphBacon, 2021-01-27:
@rjshiffer 
Oops. That's what happens when you spend years using 8-bit microcontrollers and just "know" that int is 2-bytes long! I've made this correction and I'll upload soon.

RalphBacon, 2021-01-27:
As this is a non-functional change (it just displays the calibration data, with the workaround described above) and will be included in the next release this bug is now closed.

closed 2021-01-27 by RalphBacon