RalphBacon/205-Internet-Radio #7: ESP32 log messages do not appear in Serial Monitor

rjshiffer, 2021-01-26:
Problem: ESP32 log messages from the log_n, log_d, etc., functions do not appear in the Serial Monitor
Environment: Arduino IDE
Hardware: ESP-WROOM-32 + ILI9341 + VS1053

I may be doing something wrong, but I could not get the logging messages from all the logging functions 
(log_e, log_d, etc.) to appear in the Serial Monitor.

Is there something in the Arduino IDE to set to enable this logging by default?  Perhaps the Platform.io environment sets something during the build process to enable logging.

I had to add the following to the very beginning of ESP32-WROVER_Web_Radio.ino file:
```
#ifdef CORE_DEBUG_LEVEL
#undef CORE_DEBUG_LEVEL
#endif

#define CORE_DEBUG_LEVEL 5
#define LOG_LOCAL_LEVEL ESP_LOG_DEBUG
```
After this, all the logging messages appeared on the Serial Monitor.

RalphBacon, 2021-01-26:
@rjshiffer  Yup, this is a recent regression bug! 

When I now compile the code for the Arduino (using PlatformIO initially, so that it increments the build number you see in the title bar and in the serial monitor output) I have set the debugging level to _Informational_ - meaning that messages for debug (_d) and verbose (_v) won't be shown. I do this because my "stable" version (as used in the Arduino) is the version I listen to all day and don't need so much debugging; I have another unit for that.

Your workaround was perfect and something I need to think about how I might incorporate that into the Arduino version. I need to so this otherwise when you get the next version (has a cancel button in the on-screen station list, amongst other refinements in the code) your version will get reverted.

Of course, you could switch to PlatformIO... :)

RalphBacon, 2021-01-27:
@rjshiffer 
Alsoi, ensure you have set the Core Debug Level correctly:
[image]

My Arduino version "just works" by doing this.

rjshiffer, 2021-01-27:
@RalphBacon 
Perfect!  That's exactly what I needed.  No code changes either.  I learn something new from you every day. And yes,  PlatformIO is next!  :)  Thanks.

RalphBacon, 2021-01-27:
Good luck with PlatformIO - it's definitely worth the initial head-against-brick-wall moments!

closed 2021-01-27 by RalphBacon