RalphBacon/205-Internet-Radio #8: Incorporating PSRAM

tadder, 2021-01-26:
I downloaded the latest version and straight off the bat it works fine. However when I attempt to incorporate PSRAM with the new cbuf.cpp code the following two results are manifest.

       With PSRAM enabled and the new version of cbuf.cpp
       Serial monitor output:

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[E][ESP32-WROVER_Web_Radio.ino:27] setup(): Version v1.210126_29 - 2021-01-26 10:39:16.425885
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40165225  PS      : 0x00060030  A0      : 0x80167e40  A1      : 0x3ffaf330  
A2      : 0xfffffffd  A3      : 0x0000001f  A4      : 0x00000034  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x3f401670  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00060420  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xfffffffd  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x40165225:0x3ffaf330 0x40167e3d:0x3ffaf350 0x40092047:0x3ffaf370 0x40093508:0x3ffaf3a0 0x400912ab:0x3ffaf3d0 0x4009331d:0x3ffaf470 0x4008c905:0x3ffaf4b0

           With PSRAM disabled and the new version of cbuf.cpp:

Arduino: 1.8.13 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

C:\Users\Tadder\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\cbuf.cpp: In member function 'size_t cbuf::resize(size_t)':

C:\Users\Tadder\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\cbuf.cpp:59:6: error: 'BOARD_HAS_PSRAM' was not declared in this scope

  if (BOARD_HAS_PSRAM)

      ^

Multiple libraries were found for "User_Setup.h"

 Used: C:\Users\Tadder\Documents\Arduino\libraries\TFTTFT_eSPI-2.2.23

 Not used: C:\Users\Tadder\Documents\Arduino\libraries\TFT_eSPI-2.2.23

Multiple libraries were found for "ArduinoLog.h"

 Used: C:\Users\Tadder\Documents\Arduino\libraries\ArduinoArduinoLog

 Not used: C:\Users\Tadder\Documents\Arduino\libraries\Arduino-Log

Multiple libraries were found for "VS1053.h"

 Used: C:\Users\Tadder\Documents\Arduino\libraries\ESP_VS1053_Library

 Not used: C:\Users\Tadder\Documents\Arduino\libraries\ESPESP_VS1053_Library

exit status 1

Error compiling for board ESP32 Dev Module.

RalphBacon, 2021-01-27:
When you  selected the board, did you ensure it was the ES32-WROVER board? If you selected ESP32 Dev Module you need an additional parameter:

[image]
[image]

Please check and confirm. The first error you got looks very much as though PSRAM was NOT being used so you used all the SRAM up. Keep the circular buffer value (in main.h) to 10000 (ten thousand) until we have this working for you.

The serial output will show immediately  (unless it panics) whether PSRAM is detected and how much was used.

tadder, 2021-01-27:
You were spot on, I did indeed have Dev Module selected. I have changed this to Wrover Module and this does compile. I note that Wrover Board does not provide the option in Tools of enabling PSRAM.

Confirmed that CIRCULARBUFFERSIZE is 10000

The board works for a few seconds and then reboots with the message:
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
What does ‘panic’ mean in this context?

Serial OP is as follows:

[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[D][ESP32-WROVER_Web_Radio.ino:86] setup(): Waiting for VS1053 initialisation to complete.
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[D][ESP32-WROVER_Web_Radio.ino:97] setup(): Switch player to MP3 mode
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[V][ESP32-WROVER_Web_Radio.ino:119] setup(): Previous volume was 99
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'stream.antenne1.de'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'80'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'/a1stg/livestream1.aac'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'Antenne1.de'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'1'
[V][stationSelectHelper.h:496] addStationToArray(): Vector size:1
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'bbcmedia.ic.llnwd.net'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'80'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'/stream/bbcmedia_radio4fm_mf_q'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'BBC Radio 4'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'0'
[V][stationSelectHelper.h:496] addStationToArray(): Vector size:2
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'stream.antenne1.de'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'80'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'/a1stg/livestream2.mp3'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'Antenne1 128k'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'1'
[V][stationSelectHelper.h:496] addStationToArray(): Vector size:3
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'listen.181fm.com'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'80'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'/181-beatles_128k.mp3'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'Beatles 128k'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'1'
[V][stationSelectHelper.h:496] addStationToArray(): Vector size:4
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'stream-mz.planetradio.co.uk'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'80'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'/magicmellow.mp3'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'Mellow Magic (Redirected)'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'1'
[V][stationSelectHelper.h:496] addStationToArray(): Vector size:5
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'edge-bauermz-03-gos2.sharp-stream.com'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'80'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'/net2national.mp3'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'Greatest Hits 112k (National)'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'1'
[V][stationSelectHelper.h:496] addStationToArray(): Vector size:6
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'airspectrum.cdnstream1.com'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'8024'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'/1302_192'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'Mowtown Magic Oldies'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'1'
[V][stationSelectHelper.h:496] addStationToArray(): Vector size:7
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'live-bauer-mz.sharp-stream.com'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'80'
[... 226 lines cut ...]
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
[E][ESP32-WROVER_Web_Radio.ino:27] setup(): Version v1.210126_29 - 2021-01-26 10:39:16.425885
[... cut ...]
[I][tftHelpers.h:87] initDisplay(): Touch Calibration completed
[V][bitmapHelper.h:74] drawBmp(): Loaded in 4l ms
[V][bitmapHelper.h:74] drawBmp(): Loaded in 6l ms
[V][bitmapHelper.h:74] drawBmp(): Loaded in 5l ms
[I][tftHelpers.h:206] setupDisplayModule(): TFT Initialised
[D][ESP32-WROVER_Web_Radio.ino:70] setup(): Starting player
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value

RalphBacon, 2021-01-27:
@tadder 
The key point to note here is that PSRAM is enabled (by default on the WROVER board) but is NOT being used:

[image]

You need to add the "patch" (it's only a one line change) by following the instructions in the Circular Buffer folder (part of the Arduino IDE Version folder).

The reboot seems to indicate that you're running out of memory but even if it does allocate the 10K circular buffer in SRAM it should not cause a reboot. I'm suspicious of the 10000 (10K) buffer you say you have allocated, compare the output from my startup:

[image]
According to this I have an extra 100K of SRAM for the program to run in!

Also, check that the SPI frequency set to 40MHz (not 80Mhz which can cause issues)?

Try the circular buffer "hack" as described and let me see the debug output from a clean reboot (I need to see the stuff right at the start of the output, as shown in my snip above).

tadder, 2021-01-27:
Just to confirm I copied cbuf.cpp direct (the entire file) from Arduino IDE Version/Circular Buffer on github. However to double check here is the code:

// RSB Use PSRAM here if required
	char *newbuf;
	if (BOARD_HAS_PSRAM)
	{
		newbuf = (char *)ps_malloc(newSize);
	}
	else
	{
		newbuf = new char[newSize];
	}
	char *oldbuf = _buf;

To confirm Circular Buffer Size here is the relevant code in main.h. I assume I have the correct file?

// Circular "Read Buffer" to stop stuttering on some stations
#ifdef BOARD_HAS_PSRAM
#define CIRCULARBUFFERSIZE 10000 // Divide by 32 to see how many 2mS samples this can store
#else
#define CIRCULARBUFFERSIZE 10000
#endif
cbuf circBuffer(10);
#define streamingCharsMax 32

I have changed the flash frequency to 40MHz as you suggest:

When you ask for Debug Output I assume you mean as shown on the Serial Monitor. I'm not aware of any other way to do it but I might be missing something.

So output is as follows but unfortunately still reports 'Used PSRAM: 0'

[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
⸮[E][ESP32-WROVER_Web_Radio.ino:27] setup(): Version v1.210126_29 - 2021-01-26 10:39:16.425885
[D][ESP32-WROVER_Web_Radio.ino:35] setup(): Total heap: 321148
[D][ESP32-WROVER_Web_Radio.ino:36] setup(): Free heap: 285120
[D][ESP32-WROVER_Web_Radio.ino:37] setup(): Total PSRAM: 4194252
[D][ESP32-WROVER_Web_Radio.ino:38] setup(): Free PSRAM: 4194252
[D][ESP32-WROVER_Web_Radio.ino:39] setup(): Used PSRAM: 0
[D][ESP32-WROVER_Web_Radio.ino:47] setup(): Starting SPI
[D][ESP32-WROVER_Web_Radio.ino:51] setup(): Free memory: 284996 bytes
[I][ESP32-WROVER_Web_Radio.ino:62] setup(): LITTLEFS system mounted SUCCESSFUL.
[V][tftHelpers.h:80] initDisplay(): TFT Calibration data:
[V][tftHelpers.h:83] initDisplay(): 0179, 
[V][tftHelpers.h:83] initDisplay(): 0DD2, 
[I][tftHelpers.h:87] initDisplay(): Touch Calibration completed
[V][bitmapHelper.h:74] drawBmp(): Loaded in 4l ms
[V][bitmapHelper.h:74] drawBmp(): Loaded in 6l ms
[V][bitmapHelper.h:74] drawBmp(): Loaded in 5l ms
[I][tftHelpers.h:206] setupDisplayModule(): TFT Initialised
[D][ESP32-WROVER_Web_Radio.ino:70] setup(): Starting player
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[D][ESP32-WROVER_Web_Radio.ino:86] setup(): Waiting for VS1053 initialisation to complete.
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[D][ESP32-WROVER_Web_Radio.ino:97] setup(): Switch player to MP3 mode
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[V][ESP32-WROVER_Web_Radio.ino:119] setup(): Previous volume was 99
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'stream.antenne1.de'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'80'
[V][stationSelectHelper.h:420] populateStationList(): Parameter:'/a1stg/livestream1.aac'
[... 222 lines cut ...]
[V][tftHelpers.h:232] drawPercentageLine(): Volume 84, volume line length 75
[V][tftHelpers.h:232] drawPercentageLine(): Volume 85, volume line length 77
[V][tftHelpers.h:232] drawPercentageLine(): Volume 86, volume line length 79
[V][tftHelpers.h:232] drawPercentageLine(): Volume 87, volume line length 81
[V][tftHelpers.h:232] drawPercentageLine(): Volume 88, volume line length 83
[V][tftHelpers.h:232] drawPercentageLine(): Volume 89, volume line length 85
[V][tftHelpers.h:232] drawPercentageLine(): Volume 90, volume line length 87
[V][tftHelpers.h:232] drawPercentageLine(): Volume 91, volume line length 89
[V][tftHelpers.h:232] drawPercentageLine(): Volume 92, volume line length 91
[V][tftHelpers.h:232] drawPercentageLine(): Volume 93, volume line length 93
[V][tftHelpers.h:232] drawPercentageLine(): Volume 94, volume line length 95
[V][tftHelpers.h:232] drawPercentageLine(): Volume 95, volume line length 97
[V][tftHelpers.h:232] drawPercentageLine(): Volume 96, volume line length 99
[V][tftHelpers.h:232] drawPercentageLine(): Volume 97, volume line length 101
[V][tftHelpers.h:232] drawPercentageLine(): Volume 98, volume line length 103
[V][tftHelpers.h:232] drawPercentageLine(): Volume 99, volume line length 105
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400d80d0: 40002a78 3ffc13b0 3ffc1234
Core 1 register dump:
PC      : 0x400d80d4  PS      : 0x00060130  A0      : 0x800d82b8  A1      : 0x3ffd3df0  
A2      : 0x3ffc157c  A3      : 0x3ffc1650  A4      : 0x00000020  A5      : 0x3ffc1670  
A6      : 0x134a7f6f  A7      : 0xe36d11ed  A8      : 0x800d81f9  A9      : 0x7f6f245a  
A10     : 0x3ffc157c  A11     : 0x00000002  A12     : 0x00000050  A13     : 0x3ffb1da0  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000

RalphBacon, 2021-01-27:
I suspect your compiler does not pick up the amended library.

My amended ESP32 core file is here (where all the ESP32 core files are, for the Arduino environment, anyway):
C:\Users\Ralph\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32

If you open the same folder on your machine (obviously it won't have Ralph in the name, but your own) then find the **cbuf.cpp** file and look at  the **resize** function. It _should_ look like this:

[image]

If it doesn't look like that then we know what the issue is. Replace that file with the one in the Github that you have already downloaded, recompile and see what happens.

Let me know!

tadder, 2021-01-27:
Unfortunately I had allready checked and the new file is in the correct position.
i think this proves the point:
[image]

RalphBacon, 2021-01-27:
Let's just do this. Change 
if (BOARD_HAS_PSRAM) 
to
if (1)

Recompile and see what happens.

tadder, 2021-01-27:
Ralph, Thanks for your patience and the workaround
The change you suggest has resulted in a longer period before reboot (presumably depends on how long it takes for the buffer to empty?). Unfortunately still resetting approximately every 20 seconds. Used PSRAM: still 0 despite PSRAM being enabled.

22:45:51.734 -> [D][ESP32-WROVER_Web_Radio.ino:86] setup(): Waiting for VS1053 initialisation to complete.
22:45:51.734 -> [V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
22:45:52.249 -> [D][ESP32-WROVER_Web_Radio.ino:97] setup(): Switch player to MP3 mode
22:45:52.249 -> [V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
22:45:52.624 -> [V][ESP32-WROVER_Web_Radio.ino:119] setup(): Previous volume was 99
22:45:52.624 -> [V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
22:45:55.202 -> [V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
22:45:55.249 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'stream.antenne1.de'
22:45:55.249 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'80'
22:45:55.249 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'/a1stg/livestream1.aac'
22:45:55.249 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'Antenne1.de'
22:45:55.249 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'1'
22:45:55.249 -> [V][stationSelectHelper.h:496] addStationToArray(): Vector size:1
22:45:55.249 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'bbcmedia.ic.llnwd.net'
22:45:55.296 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'80'
22:45:55.296 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'/stream/bbcmedia_radio4fm_mf_q'
22:45:55.296 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'BBC Radio 4'
22:45:55.296 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'0'
22:45:55.296 -> [V][stationSelectHelper.h:496] addStationToArray(): Vector size:2
22:45:55.296 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'stream.antenne1.de'
22:45:55.296 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'80'
22:45:55.343 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'/a1stg/livestream2.mp3'
22:45:55.343 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'Antenne1 128k'
22:45:55.343 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'1'
22:45:55.343 -> [V][stationSelectHelper.h:496] addStationToArray(): Vector size:3
22:45:55.343 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'listen.181fm.com'
22:45:55.343 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'80'
22:45:55.390 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'/181-beatles_128k.mp3'
22:45:55.390 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'Beatles 128k'
22:45:55.390 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'1'
22:45:55.390 -> [V][stationSelectHelper.h:496] addStationToArray(): Vector size:4
22:45:55.390 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'stream-mz.planetradio.co.uk'
22:45:55.390 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'80'
22:45:55.390 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'/magicmellow.mp3'
22:45:55.437 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'Mellow Magic (Redirected)'
22:45:55.437 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'1'
22:45:55.437 -> [V][stationSelectHelper.h:496] addStationToArray(): Vector size:5
22:45:55.437 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'edge-bauermz-03-gos2.sharp-stream.com'
22:45:55.437 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'80'
22:45:55.437 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'/net2national.mp3'
22:45:55.484 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'Greatest Hits 112k (National)'
22:45:55.484 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'1'
22:45:55.484 -> [V][stationSelectHelper.h:496] addStationToArray(): Vector size:6
22:45:55.484 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'airspectrum.cdnstream1.com'
22:45:55.484 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'8024'
22:45:55.484 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'/1302_192'
22:45:55.484 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'Mowtown Magic Oldies'
22:45:55.484 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'1'
22:45:55.484 -> [V][stationSelectHelper.h:496] addStationToArray(): Vector size:7
22:45:55.484 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'live-bauer-mz.sharp-stream.com'
22:45:55.484 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'80'
22:45:55.484 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'/magicmellow.aac'
22:45:55.484 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'Mellow Magic (48k AAC)'
22:45:55.531 -> [V][stationSelectHelper.h:420] populateStationList(): Parameter:'1'
22:45:55.531 -> [V][stationSelectHelper.h:496] addStationToArray(): Vector size:8
[... cut ...]
22:46:51.265 -> [V][bitmapHelper.h:74] drawBmp(): Loaded in 5l ms
22:46:51.265 -> [I][tftHelpers.h:206] setupDisplayModule(): TFT Initialised
22:46:51.265 -> [D][ESP32-WROVER_Web_Radio.ino:70] setup(): Starting player
22:46:51.265 -> [V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
22:46:53.375 -> [V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
22:46:55.531 -> [V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value

RalphBacon, 2021-01-28:
I see the amount of heap has now increased to a sensible figure, so that is a positive.
`setup(): Free heap: 285120`

What is not clear is why the PSRAM is not being allocated by the change I asked you to make.
`Used PSRAM: 0`
This seems to imply that either that line of code is not being executed or that 'ps_malloc' is not using PSRAM.

Please send me the output of the compilation, from the very start to end.

Whilst you wait for my reply to that info, add in the following line to cbuf.cpp so it looks like this:
[image]

Then send me the output of a clean reboot serial monitor output (debug output) _where it shows the free memory_. To do this (so I don't get all the output) start up the device, start the serial monitor, press the reboot button and then just copy the lines that I need here.
Also a snapshot of the change you made in cbuf.cpp too.

tadder, 2021-01-28:
Sorry for the delay in my response, I am having extensive building work carried out here, and when builders ask for a decision they expect ‘instant’ answers. Nothing I suspect to the disruption your move is causing you, but I do look forward to the distraction of an ESP32 project.

Compilation output before your latest suggested change:

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Tadder\AppData\Local\Arduino15\packages -hardware C:\Users\Tadder\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Tadder\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Tadder\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32wrover:PartitionScheme=no_ota,FlashMode=qio,FlashFreq=40,UploadSpeed=921600,DebugLevel=verbose -ide-version=10813 -build-path C:\Users\Tadder\AppData\Local\Temp\arduino_build_144348 -warnings=none -build-cache C:\Users\Tadder\AppData\Local\Temp\arduino_cache_798536 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\Tadder\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Tadder\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\Tadder\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\Tadder\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Tadder\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\Tadder\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -verbose C:\Users\Tadder\Documents\Arduino\RalphBacon\ESP32-WROVER_Web_Radio\ESP32-WROVER_Web_Radio.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Tadder\AppData\Local\Arduino15\packages -hardware C:\Users\Tadder\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Tadder\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Tadder\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32wrover:PartitionScheme=no_ota,FlashMode=qio,FlashFreq=40,UploadSpeed=921600,DebugLevel=verbose -ide-version=10813 -build-path C:\Users\Tadder\AppData\Local\Temp\arduino_build_144348 -warnings=none -build-cache C:\Users\Tadder\AppData\Local\Temp\arduino_cache_798536 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\Tadder\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Tadder\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\Tadder\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\Tadder\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Tadder\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\Tadder\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -verbose C:\Users\Tadder\Documents\Arduino\RalphBacon\ESP32-WROVER_Web_Radio\ESP32-WROVER_Web_Radio.ino
Using board 'esp32wrover' from platform in folder: C:\Users\Tadder\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4
Using core 'esp32' from platform in folder: C:\Users\Tadder\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4
Detecting libraries used...
"C:\\Users\\Tadder\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\Tadder\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/config" "-IC:\\Users\\Tadder\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_trace" "-IC:\\Users\\Tadder\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_update" "-IC:\\Users\\Tadder\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/asio" "-IC:\\Users\\Tadder\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bootloader_support" "-IC:\\Users\\Tadder\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bt" "-IC:\\Users\\Tadder\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/coap" "-IC:\\Users\\Tadder\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/console" "-IC:\\Users\\Tadder\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/driver" "-IC:\\Users\\Tadder\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/s [... cut ...]
Using library ArduinoArduinoLog at version 1.0.3 in folder: C:\Users\Tadder\Documents\Arduino\libraries\ArduinoArduinoLog 
"C:\\Users\\Tadder\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-size" -A "C:\\Users\\Tadder\\AppData\\Local\\Temp\\arduino_build_144348/ESP32-WROVER_Web_Radio.ino.elf"
Sketch uses 1037693 bytes (49%) of program storage space. Maximum is 2097152 bytes.
Global variables use 46432 bytes (14%) of dynamic memory, leaving 281248 bytes for local variables. Maximum is 327680 bytes.

tadder, 2021-01-28:
Snapshot of modified cbuf.cpp:

// RSB Use PSRAM here if required
	char *newbuf;
	//if (BOARD_HAS_PSRAM)
	if (1)
	{
		newbuf = (char *)ps_malloc(newSize);
		Serial.println("Allocated PSRAM");
	}
	else
	{
		newbuf = new char[newSize];
	}
	char *oldbuf = _buf;

tadder, 2021-01-28:
Interestingly after the mod it wont compile because of the Serial command.
Perhaps I need to “Serial.begin(9600)” first?

Relevant part of the compile log is as follows:

C:\Users\Tadder\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\cbuf.cpp: In member function 'size_t cbuf::resize(size_t)':
C:\Users\Tadder\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\cbuf.cpp:63:3: error: 'Serial' was not declared in this scope
   Serial.println("Allocated PSRAM");
   ^
Multiple libraries were found for "User_Setup.h"
 Used: C:\Users\Tadder\Documents\Arduino\libraries\TFTTFT_eSPI-2.2.23
 Not used: C:\Users\Tadder\Documents\Arduino\libraries\TFT_eSPI-2.2.23
Multiple libraries were found for "ArduinoLog.h"
 Used: C:\Users\Tadder\Documents\Arduino\libraries\ArduinoArduinoLog
 Not used: C:\Users\Tadder\Documents\Arduino\libraries\Arduino-Log
Multiple libraries were found for "VS1053.h"
 Used: C:\Users\Tadder\Documents\Arduino\libraries\ESP_VS1053_Library
 Not used: C:\Users\Tadder\Documents\Arduino\libraries\ESPESP_VS1053_Library

RalphBacon, 2021-01-28:
I wondered if that Serial output would give us problems; just remove that Serial.println line.

The compilation output indicates that the correct board has been selected and has the required "fix" built in for the PSRAM use (by Espressif).

I did notice that your libraries contain folders such as:

- TFTTFT_eSPI-2.2.23
- ArduinoArduinoLog
- ESPESP_VS1053_Library
- and possibly others

Please **remove all these odd libraries** that clutter up the compilation process to somewhere else (eg C:\ArduinoArchive), certainly nowhere that the compiler can find them.

Recompile (and check that you have no strange libraries being picked up) and as it uploads please capture the output just as starts the process and check that the Revision of your chip is Revision 1

[image]

That all said, I'm now scratching my head a bit. The compilation, whilst a bit messy, seems to work OK, at run time the memory (now) looks normal.

So something else to try.

Ensure the circular buffer remains at 10000 (ten thousand) in both places in the main.h file.
Change the board type to ESP32 Dev Module and these parameters:
[image]

If your partition is not as shown above (or you change it) you will have to upload the data folder again via LittleFS.
The reason for doing this (grasping at straws) is that your ESP32 seems to panic at the slightest opportunity so I want to make sure it is stable just running as a WROOM board (no PSRAM) before trying anything else.

tadder, 2021-01-28:
Interestingly these extra Libraries were created when I originally followed the instructions for Arduino in the README.md file. Maybe I didn’t follow the instructions as I should have done? Does this gives us a clue?

I have archived the duplicate libraries as you suggest and confirm the compilation output now looks a lot cleaner.
Will send upload output next.

tadder, 2021-01-28:
Not sure how the Arduino IDE locates libaries but it finds old libary files in Archive if ileave this in the libraries folder. Moving it to my desktop puts it out of reach.
I confirm that it is a Rev1 board, and buffer length is set to 10000. PSRAM still at 0
I havent changed the board type yet (still WROVER MODULE. But its a lot more stable. At the moment it hasn't reset in 20 minutes. I stops a lot due to a low buffer, but recovers on its own accord,
I will now change the board type.

tadder, 2021-01-28:
Board Changed to "ESP 32 Dev Module" tried with PSRAM both Enabled and Disabled. The good news is that it appears to be working well. However I cannot sernd the serial output as for some reason evewry time I reboot I get the following 14 lines only. So no idea if PSRAM is being used.

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[E][ESP32-WROVER_Web_Radio.ino:27] setup(): Version v1.210126_29 - 2021-01-26 10:39:16.425885

RalphBacon, 2021-01-28:
You need to change the Core debug to Verbose to get more output. (A lot more). But good news on the stability. You are powering the device using the +5v socket not the USB (the switch on the side should be set to OFF)?

RalphBacon, 2021-01-28:
If the reset is power-related in any way, you should see "Brownout detected" on the serial monitor output as it resets.

tadder, 2021-01-28:
I haven't changed the settings since the change of board.
Good news is that the project is still going with no glitch (over one hour ago)
[image]

RalphBacon, 2021-01-28:
This setting:
[image]

tadder, 2021-01-28:
Doh. Stupid me!
Sadly Used PSRAM: is still 0.
However board is still working so that’s very good.
I will leave it on ‘Soak Test’ overnight (any excuse to listen to Beatles radio).

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
[E][ESP32-WROVER_Web_Radio.ino:27] setup(): Version v1.210126_29 - 2021-01-26 10:39:16.425885
[D][ESP32-WROVER_Web_Radio.ino:35] setup(): Total heap: 321160
[D][ESP32-WROVER_Web_Radio.ino:36] setup(): Free heap: 285088
[D][ESP32-WROVER_Web_Radio.ino:37] setup(): Total PSRAM: 4194252
[D][ESP32-WROVER_Web_Radio.ino:38] setup(): Free PSRAM: 4194252
[D][ESP32-WROVER_Web_Radio.ino:39] setup(): Used PSRAM: 0
[D][ESP32-WROVER_Web_Radio.ino:47] setup(): Starting SPI
[D][ESP32-WROVER_Web_Radio.ino:51] setup(): Free memory: 284964 bytes
[I][ESP32-WROVER_Web_Radio.ino:62] setup(): LITTLEFS system mounted SUCCESSFUL.
[V][tftHelpers.h:80] initDisplay(): TFT Calibration data:
[V][tftHelpers.h:83] initDisplay(): 0179, 
[V][tftHelpers.h:83] initDisplay(): 0DD2,

tadder, 2021-01-29:
I switched the radio back on again this evening and would you believe, it was dead. LED’s illuminated and screen but no data. So I deduced I needed to download both the DATA folder to LittleFS and the Sketch again. Result every time was the message:

“A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header”

Re booted the PC (Win 10) and the board, same result. I really can’t believe this after the success yesterday. Successfully downloaded a blink sketch to a UNO using same USB connection, so am assuming an issue with the ESP32, but no idea as to why.

I appreciate this cannot be anything to do with the WEB-RADIO software, but thought you might like to know just for interest.

RalphBacon, 2021-01-30:
Sad news. 
If you find out the COM port the ESP32 is connected to (if it is) see if anything appears on the serial monitor. You've definitely selected the correct board in Tools?

I had one ESP32 go bad; the COM port was identified but the ESP32 was non-responsive. But I probably shorted something underneath. Yours sounds like a random failure.

tadder, 2021-01-31:
Yes, I did check that the COM port was correct. Researching online it would appear that ESP32’s failing in this way is not unknown. However remedies suggested range from pressing the BOOT button or (as you eluded to in a previous video) looking to the east, spinning round clockwise, and throw salt over your left shoulder. Some suggest a capacitor from GPIO0 to ground which I tried with no success. My standard flash process with this board has always been hold down the boot button then momentarily press RST, so the ‘cures’ are just the same.
I think I have to accept that this is a fault, but of three TTGO T8 purchased two have failed, a high attrition rate I think.

Any way the good news is that I have utilised my final spare board and the radio is now working well (no problem programming it).

I have seen a suggestion that utilising a USB to serial port (TTL) converter might help by bypassing the USB. I have such converter so perhaps I will give it a try on the faulty boards.

It still reports Used PSRAM 0 so presumably the ring buffer is not being utilised. Just a thought, am I correct in thinking that the only difference between your set up and mine is that you use Platform IO and I use Arduino IDE? Is this a clue I wonder?

Aleks-Ale, 2021-02-01:
Hi Ralph, hi tadder. 
I'm with you again, because I have the same problem. By coincidence, the same board was delivered to me yesterday.  
ESP32-WROVER T8 V1.8, 8 MB, with SPRAM. The problem is the same as above: 

```
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
[E][ESP32-WROVER_Web_Radio.ino:27] setup(): Version v1.210126_29 - 2021-01-26 10:39:16.425885
[D][ESP32-WROVER_Web_Radio.ino:35] setup(): Total heap: 321172
[D][ESP32-WROVER_Web_Radio.ino:36] setup(): Free heap: 285144
[D][ESP32-WROVER_Web_Radio.ino:37] setup(): Total PSRAM: 4194252
[D][ESP32-WROVER_Web_Radio.ino:38] setup(): Free PSRAM: 4194252
[D][ESP32-WROVER_Web_Radio.ino:39] setup(): Used PSRAM: 0

```

cbuf.cpp checked:

```
cbuf::resize(size_t newSize)
{
	// RSB Use PSRAM here if required
	char *newbuf;
	if (1)
//		if (BOARD_HAS_PSRAM)
	{
		newbuf = (char *)ps_malloc(newSize);
//		Serial.println (">>>>>>>>> ALLOCATED PSRAM <<<<<<<<<<<<");
	}

```
[image]

Revision 1.

[image]

This works with the PSRAM on and off if the buffer is small.

```
#ifdef BOARD_HAS_PSRAM
#define CIRCULARBUFFERSIZE 10000
#else
#define CIRCULARBUFFERSIZE 10000
```
Dear Doctor, how to treat a sick patient? :)

Aleks-Ale, 2021-02-01:
My board:
[image]

RalphBacon, 2021-02-01:
@tadder  @Aleks-Ale 
The fact that both of you report zero bytes used in the circular buffer indicates that the new Espressif cbuf.cpp code is not being compiled - perhaps there is another pre-compiled (cached) version being used or a previously compiled version.

You could try uninstalling/reinstalling the Espressif framework. If you remove the current Espressif environment does this file get removed?

[image]

Once removed, you should scour your systems for remnants of this environment (eg in C:\Users\XXXXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32)

Remove any stragglers and you can then reinstall and do the change again. As it would be a new install, I would expect the compiler to actually compile the cbuf.cpp code at least once (so make the change first before your first compile).

I'm hesitant to do that here as I'm in the middle of making some code changes so I don't to cause myself issues in the Arduino environment (which works) - you, on the other hand, have nothing to lose. Incidentally, my **PlatformIO** code is now different to this Arduino change. 

`bool IRAM_ATTR psramFound(){
    return spiramDetected;
}

void IRAM_ATTR *ps_malloc(size_t size){
    if(!spiramDetected){
        return NULL;
    }
    return heap_caps_malloc(size, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT);
}`

As you can see, it automatically detects whether **SPI RAM** (PSRAM) is installed and then uses that in the **ps_malloc** call. This should eventually filter down to the Arduino version of the Espressif code.

Aleks-Ale, 2021-02-01:
> You could try uninstalling/reinstalling the Espressif framework.

Unfortunately, the miracle did not happen.
- I deleted the esp32 board,
- cleared all repositories,
- cleared the temp folder,
- made a new installation of the board
- Changed the cbuf.cpp
Nothing.

[image]

tadder, 2021-02-01:
Thanks for that, it certainly makes sense to reset like you suggest. I am thinking that I could install the Arduino IDE on a laptop that I have, that will certainly ensure that there is no residual cbuf.cpp. And leave me with my main installation as a fall back if it all goes pear shape.

Your enthusiasm for PlatformIO has not been lost on me and indeed I have spent some time attempting to get up to speed with it. A simple sketch is one thing, but I have yet to succeed when importing the entire project.

Aleks-Ale, 2021-02-02:
Dear Ralph, I will tell you about my tests. I have a HDD with an old Windows 7 and I installed an arduino IDE on it. But at first, PSRAM did not appear. Then, I replaced the file cpbuf.cpp and I saw a working PSRAM!!!
[image]

Then, I connected TTGO to my office computer with Windows 10. No cpbuf.cpp changes were required! 
The only difference between the office and home installation of the Arduino ide is that the arduino ide is installed from the ms store on the home computer!!!
I think it would be a good idea to remove the arduino ide on my home computer and install it from the installation file. I think it would be a good idea to remove the arduino ide on my home computer and install it from the installation file.
As a result, I have the following results:
- Windows 10 on my home computer and arduino ide from ms store: nothing helps to run PSRAM.
- Windows 7 on my home computer and arduino ide from the installation file: cpbuf.cpp replacement helps.
- Windows 10 on my office computer and arduino ide from the installation file: works immediately. No changes are needed.
This Is A Paradox!

RalphBacon, 2021-02-02:
@Aleks-Ale That is good news about the circular buffer. You can increase that now to 150000 to get the benefits. It will be interesting to see whether a reinstallation of the Arduino IDE also sorts out the problem with your home computer. Let us all know how you get on!
I'm wondering if the updated Espressif framework that auto-allocates memory into PSRAM (when available) is in the newest environment (as per my screenshots above).

Aleks-Ale, 2021-02-02:
> I'm wondering if the updated Espressif framework that auto-allocates memory into PSRAM (when available) is in the newest environment (as per my screenshots above).

I'm sorry, but here I have difficulties translating or understanding. 
Can I ask you to explain what I should check and how? Some of your screenshots you have in mind. Thanks.

RalphBacon, 2021-02-02:
@Aleks-Ale 
What I meant was that I think your "new" installation has an updated Espressif ESP32 code base (framework). Thus, no cbuf.cpp code changes are required.  Your code base perhaps looks like the PlatformIO code snippets I showed above (where it checks for PSRAM and if present executes the ps_malloc to use it).

I am only guessing though, but it might explain why you have success in one environment but not another.

Aleks-Ale, 2021-02-02:
After reinstalling the arduino ide home, everything works fine. PSRAM is present! 
I would not be able to assume that it is necessary to reinstall the software. Thank you for your help!
[image]

RalphBacon, 2021-02-03:
@Aleks-Ale 
I'm very pleased it all works now for you. I will bear in mind that an Arduino IDE reinstallation may be necessary for others, too, who experience this problem.

I will wait for @tadder to confirm that he has success too with a new installation before I close this thread.

tadder, 2021-02-03:
It took me a lot longer to install a new system on my laptop than I initially expected as I had an issue with the location of mkspiffs.exe. In the end guided by others with the same issue, I placed it in  ...Arduino15/packages/esp32/tools. Placing in the sketch folder tools/ESP32FS/tool wasn't seen and I got “could not find mkspiffs” error. This wasnt an issue with my original PC install.

So with a completely new system up and running it’s time to get the ring buffer working.

Well it looks as if I might have to admit failure with this project. You may recall that about a week ago I had my second board fail with the “Failed to connect to ESP32:” error. This has now happened again. I think that makes it the third board to fail this way. I appreciate this must be down to something I am doing, but I am beginning to think they have a built in limit on how many downloads they will accept. In the process it appears as if the previous file has been deleted so I no-longer have a working radio. This happens with both Arduino IDE setups I now have. The problem always manifests itself on a board that was previously working fine

I have a spare WROOM-32 board that continues to accept download's in the “ESP 32 Dev Module” environment (albeit with PSRAM init failed) so I deduce there is nothing wrong with my basic setup.

RalphBacon, 2021-02-04:
@tadder 
That's a shame. If you have the same TTGO T7 board as me, I can confirm I must have uploaded the code to them both here about 200 times each without an issue. However, I did have one board fail, unexpectedly a few weeks ago. The USB port is fine but there is nothing at the other end. One day I will investigate to see whether the ESP32 chip got fried, but that will have to wait until I complete my house move.
Where did you get your boards from? Just in case it was a "bad batch".

tadder, 2021-02-04:
Yes it had occurred to me that it might be a bad batch, but the supply so far has been from two locations:

    1. 19th December		Amazon
    2. 14th Jan		eBay st-8078
    3. 20th Jan		eBay st-8078
Not that this precludes them originating in the same batch, but unlikely.
st-8078 is your affiliate, so suspect your boards have come from the same supplier as mine.

I have now ordered another (final?) two replacements from Sensor-out (Germany) and treelight2016 (Shenzhen).

I hope your move continues with minimal delay. No doubt you would love to have it all done and dusted by the time lockdown finishes.

RalphBacon, 2021-02-08:
At this point I'd suspect that you are zapping the devices with static electricity. If they are all connected up via the USB socket (but with the switch set to OFF, so the main unit is not powered via it) then they are grounded. If you then switch them on and off via a wall wart (mains switch, don't keep unplugging them) they should last. It's too easy to zap devices with static (thankfully rare, though) so try the above of keeping them connected to a GND point. And don't physically touch them!

Yes, I'd love to move before Xmas 2021. It's like wading through cold treacle. Solicitors have refined the delaying technique to a fine art. That's why they charge so much.

This thread is now too long to manage so I'm closing it. If anyone has issues with PSRAM please open a new thread.

closed 2021-02-08 by RalphBacon