RalphBacon/205-Internet-Radio #10: PSRAM

tadder, 2021-02-09:
Circular (Ring) Buffer

As suggested before my ESP32 went to the PCB bin in the sky, I have carried out a new implementation of the Arduino IDE (1.8.13) on my laptop. I have downloaded the alternative version of “cbuf.cpp” from the Circular Buffer section in github (205-Internet-Radio) and this I have placed in:
C:\Users\Tadder\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32

I confirm that the CIRCULARBUFFERSIZE is set at 10,000

As can be seen from the following output PSRAM is enabled, but Used PSRAM still reports 0.
Any further thoughts?

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
[E][ESP32-WROVER_Web_Radio.ino:27] setup(): Version v1.210126_29 - 2021-01-26 10:39:16.425885
[D][ESP32-WROVER_Web_Radio.ino:35] setup(): Total heap: 321172
[D][ESP32-WROVER_Web_Radio.ino:36] setup(): Free heap: 285232
[D][ESP32-WROVER_Web_Radio.ino:37] setup(): Total PSRAM: 4194252
[D][ESP32-WROVER_Web_Radio.ino:38] setup(): Free PSRAM: 4194252
[D][ESP32-WROVER_Web_Radio.ino:39] setup(): Used PSRAM: 0
[D][ESP32-WROVER_Web_Radio.ino:47] setup(): Starting SPI
[D][ESP32-WROVER_Web_Radio.ino:51] setup(): Free memory: 285108 bytes
[I][ESP32-WROVER_Web_Radio.ino:62] setup(): LITTLEFS system mounted SUCCESSFUL.
[V][tftHelpers.h:80] initDisplay(): TFT Calibration data:
[V][tftHelpers.h:83] initDisplay(): 0179, 
[V][tftHelpers.h:83] initDisplay(): 0DD2, 
[I][tftHelpers.h:87] initDisplay(): Touch Calibration completed
[V][bitmapHelper.h:74] drawBmp(): Loaded in 6l ms
[V][bitmapHelper.h:74] drawBmp(): Loaded in 9l ms
[V][bitmapHelper.h:74] drawBmp(): Loaded in 9l ms
[I][tftHelpers.h:206] setupDisplayModule(): TFT Initialised
[D][ESP32-WROVER_Web_Radio.ino:70] setup(): Starting player
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[D][ESP32-WROVER_Web_Radio.ino:86] setup(): Waiting for VS1053 initialisation to complete.
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value
[D][ESP32-WROVER_Web_Radio.ino:97] setup(): Switch player to MP3 mode
[V][tftHelpers.h:809] displayTrackArtist(): No delimiter found - using default value

RalphBacon, 2021-02-09:
Another maker (see the previous thread on this topic) reinstalled everything including the Arduino IDE, ESP32 boards and it all "just worked" without any hacks required. It also seemed to depend on where he took the Arduino IDE code from and what version of Windows he ran (7 or 10).

However, back to your issue, that is a clear sign that PSRAM is not being used. Try removing just the ESP32 board(s) so that everything in this folder (and below) should disappear:
`C:\Users\Tadder\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\`
Then reinstall the ESP32 boards (and hence environment) and check the cbuf.cpp to see whether it is the updated version that requires no hack. If it still has the edited hack then it is not being replaced by doing the above. You might have to manually deep clean it all and try again.

When you recompile, look very closely at the output from the compiler. It is is using a cached, previously compiled version you will have to delete that to force a new compilation. But I don't fully understand why the Arduino behaviour is so different from that of PlatformIO. Talk about confusing us all.

tadder, 2021-02-11:
Yes, I remember the other maker and the ‘cure’ that worked for him. I am using Windows 10 Pro and downloaded the Arduino IDE (1.8.13) from www.arduino.cc/en/Main/Software&.
As you suggested I removed the ESP32 core and confirm that everything in
C:\Users\Tadder\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\ 
did disappear. Once the core was reloaded (via the Expressif link in additional board manager) cbuf.cpp was indeed a new version.
Unfortunately PSRAM usage is still 0. I cannot see any reference in the compiler output to a previously cached version being used.

The fact that I appear to be the only one of your Arduino IDE users having an issue is worrying and I am very aware of taking up a significant amount of you time. I therefore think the best way forward is for me to attempt to utilise PlatformIO as this clearly works for you. The process of working out how to import your code into my copy of PlatformIO seems to be my biggest challenge, but once I have conquered that hurdle I will hopefully be on the road to a solution. The project is currently working well with the odd ‘hiccup’ due to buffering issues, so although not urgent I would rather not be beaten by this. Challenges like this are what being an “Arduinoite” is all about (he said with his positive hat on).

RalphBacon, 2021-07-21:
New version v1.35 for the Arduino IDE now available.

closed 2021-07-21 by RalphBacon