RalphBacon/205-Internet-Radio #9: Static Issues

tadder, 2021-02-09:
"At this point I'd suspect that you are zapping the devices with static electricity. If they are all connected up via the USB socket (but with the switch set to OFF, so the main unit is not powered via it) then they are grounded. If you then switch them on and off via a wall wart (mains switch, don't keep unplugging them) they should last. It's too easy to zap devices with static (thankfully rare, though) so try the above of keeping them connected to a GND point. And don't physically touch them!"

The first of my replacement ESP32 has arrived and has been flashed OK.
Exactly what switch are you referring to above?

In the future I shall treat the board with a little more respect and be less cavalier with my static precautions.

But, your suggestion has got me thinking (always dangerous). In my setup, power reaches the WebRadio by two routes:
    1. Via USB connected to my PC, usually only when flashing, but sometimes concurrent with -
    2. Bench PSU (Tenma 72-2690) to a barrel jack connected to the main PCB +5V.

However, I don’t currently strap the PSU negative to GND, so I wonder is it possible that allowing this to float combined with the USB derived power could result in excessive voltage?
I’m not very o fay with this subject, but my limited research would suggest that this is at least theoretically possible?

RalphBacon, 2021-02-09:
There is a slide switch near the battery connector (assuming you have a TTGO T8-style unit) on the LHS when viewed from the battery connector end. Push that _away_ from the USB socket end. This now means that the USB cable will only power the serial interface but not the ESP32 and other circuitry. Using two power sources simultaneously is often fraught with danger as power can flow in the wrong direction (eg into a laptop's USB port with potentially disastrous consequences). You'll know when you have done this correctly because the PC will identify a COM port but the ESP32 (and screen etc) will not be running.

My unit is powered by a wall wart-type phone charger, 5v (about 2A but I take that with a pinch of salt). It's not connected to ground. The USB cable is, of course, connected to ground via my PC and I have noticed that if I (dis)connect the cable it can cause lockups. Hence I leave it connected all the time on one of my units so I can monitor the serial output. The other is not connected via the USB at all (except when I infrequently reflash an updated program).

I'd want to check the Tenma power supply on a scope to see what happens when you switch it on (or off). Is there a spike, however brief? It's the same PSU as I have and I've never experienced any issues whilst prototyping with it but you never know. You might want to consider a dedicated wall wart charger (a better quality one, not a cheap Chinese one that delivers 6v @ 100mA). When I built my mum's Home Alone (movement alert) monitor, I bought a medical grade plug-in PSU just to be safe. Yeah, cost about 5 times what a cheap Asian brand would sell at.

Now, I don't think you will get excessive voltage with your setup, but unless you actually check it you won't know. I'd suspect the static zapping by you touching the device rather than that. You can mitigate that risk by earthing yourself via a wrist band (connected to mains earth via a built-in 1MΩ resistor - some don't have the resistor which are death traps!) Alternatively, don't touch the device, especially not the pins, just the edge of the board. And touch a GND connector first to discharge any static you are carrying.

I'll leave you to setting this up and getting it running - good luck and no more failed devices! My two units are running all day (no longer overnight, my soak testing is now complete) without an issue.

tadder, 2021-02-11:
I have checked the output of my PSU with a scope and as far as I can see it’s very clean. There is no visible sign of any spikes although I have to accept if it was very fast It wouldn't show. But on balance I would say this is not my problem.

So I will continue to implement the suggestions you have made re static precautions. Thanks as always for sharing your knowledge.

I will now close this issue.

tadder, 2021-08-08:
The following may be of interest just for the record:
After replacing the ESP32 WROVER board in February everything has worked fine. However, the other day I went to switch the project on and my finger slipped off the ‘POWER’ switch (on my Tenma PSU). I surmise that this may have sent a spike to the board, because afterward it was dead. Did some fault finding which with my competence level is pretty much restricted to ensuring VCC is present (it was). Luckily after the problems earlier in the year I had a spare in the draw and after replacing the dead unit everything jumped back to life.
I now have three ESP 32 WROVER boards that are RIP. They do appear to be a little sensitive to power glitches. I have now moved to using a WallWart PSU, lets hope this stops me consuming the boards which are this year costing in excess of £10 a piece.

RalphBacon, 2021-08-08:
Condolences on your loss. 

However, I have to admit I would never leave an item connected to my Tenma PSU then switch the PSU on. At switch on there could easily be a surge, spike or some other nasty. I  always switch on the PSU, _then_ connect any equipment.

Maybe you were unlucky, but a lesson learned for us all, for sure.

Regarding sensitivity of the ESP32 modules, I noticed that if I even _touch_ the GND connection (eg via the USB cable) with another cable (ie the metal outer sheath of another cable to a spare USB socket) my ESP32 will crash/hang. A power on reset sorts it out. Drives me nuts, but luckily the ESP32 is pretty much a floating ground device as long as the USB cable is not attached (I power mine from a wall-wart and the audio output is isolated via those tiny transformers).

tadder, 2021-08-09:
I have the project power connected via a standard barrel jack. The irony is that I switched on my Tenma after connecting for the very reason (I thought) that this would ensure a clean connection i.e. no potential fiddling around with the jack trying to make a connection whilst the power alternatively made and unmade its connection due to my fumbling hands. I suspect from what you say, that my thinking is flawed as far as this goes.
At the time I was experimenting with adding extra stations to the list. This involved much trial and error and subsequent multiple downloads to the data folder. It therefore seemed to make sense to retain the USB connection at least while developing.

closed 2021-08-13 by tadder