RalphBacon/205-Internet-Radio #13: Very choppy audio (without PSRAM) and some more questions

bepaald, 2021-06-10:
Hi! First of all thank you very much for this project and the corresponding youtube videos. I had a TFT display lying around, and always a couple of esp32's so I ordered the vs1053 since this seemed like it would be fun to make.

I think I have a working radio on a breadboard now, though I had a lot of trouble with the wiring. I should note I am not a hardware-guy, and it is all a bit new to me. But after going through the source code, the board schematics and some youtube comments, and realizing 3 of the four SPI connections can be shared (and that the touch devices' mosi and miso are labelled with other names) I think I have the wiring correct. The only one I'm not sure about is the vs1053's 'XRST' pin. I have it hooked up to the esp32's 'EN' pin (which I believe is the reset), but I read in a youtube comment that it should not be connected to the esp32 at all? I think I tested it unconnected, but then I get no audio. Could you help me with that?

Now, I only had WROOM modules around, so I was expecting a bit of a glitch in the audio once in a while, however, it is much worse. I think there is about as much glitching as there is playing even though I already increased the ring buffer to 50000 bytes. I also sometimes hear the audio 'catch up' (playing sped up) after having been cut out for a bit. I made a little video of it here: [https://sendvid.com/nloezqcf](https://sendvid.com/nloezqcf). Is this expected? Is it just a matter of swapping in the WROVER (it is already underway) and enabling the PSRAM, or do you suspect something else is wrong. Maybe my wiring is bad, or am I missing some passive components? At the moment the project is just the esp32, tft, vs1053, a breadboard and dupont wires. 

Lastly, I notice that the `player.begin()` loop almost always (but strangely enough not always) runs the full countdown without ever breaking early because `player.begin()` returns true. I notice no change in performance between the 'failed' initializing and the rare few times were true is returned and the countdown doesn't reach zero. Should `player.begin()` usually return true quickly? I notice, in the current upstream version, `VS1053::begin()` is a void function and the example code provided does no checking for success. Also, the next loop again has the structure for a countdown, but `mp3StartCount` is never actually decremented, not sure but that looks unintended.

Thanks again, sorry for the long post.

RalphBacon, 2021-06-12:
Hi bepaald!

 

Sorry to hear you have issues with choppy sound.

 

Firstly, even when not using a circular buffer of much substance (eg 10K on a non-WROVER device) the sound should be fine if your internet stream Is stable.

 

If you are using my code, without any (substantial) changes then it should play just fine (as mine did before I went the WROVER route). Of course, there will be interruptions as the internet stream is never perfect but it should not be choppy as per your example video.

 

Occasionally, I have had a similar experience if I am outputting too much on the serial port (for debugging/monitoring) as it takes too long and interrupts the process of delivering the 32 byte chunks to the VS1053. If you have lots of serial output (it MUST be at 115200 anyway, anything else is too slow) try reducing the debugging level to WARNING  just to see if that helps.

 

If the above does not help then I do suspect a wiring issue, but let me know how it goes before we try and investigate further.

 

One last point: you seem to using an older version (1v29) Arduino ported code; I’ve made some changes to the PlatformIO version 1v32 but the fundamental music streaming should always work.

 

Best Regards

Ralph S Bacon

 <https://www.youtube.com/ralphbacon> https://www.youtube.com/ralphbacon

rjshiffer, 2021-06-12:
Hi Ralph,

In your reply to @bepaald you mentioned that he is using an older version 1v29.  I'm also using that version.  Now that I'm on Platform IO, is the 1v32 Platform IO version available somewhere?  I can't seem to locate it.

Thanks, 
Richard (a daily radio listener!)

RalphBacon, 2021-06-12:
A newer version (with a separate task for the music streaming, but it’s not 1v32 as far as I can recall, it was months ago that I did this) for PlatformIO can be found here:  <https://github.com/RalphBacon/208-Arduino-IDE-Web-Radio-using-a-Task> https://github.com/RalphBacon/208-Arduino-IDE-Web-Radio-using-a-Task

 

If you can install that project into PlatformIO then I can put up a zip of the absolute latest version and it should be simple to update. Yes, I know all this should be done via GitHub but it all takes time.

 

Remember to update the “data” folder with your Wifisecrets.txt and do calibrate the screen (a bit of a pain) to get the string of numbers out. If you don’t do this you might find all the “buttons” don’t quite seem to be where they should be when you “press” them.

 

Let me know how it goes.

 

Best Regards

Ralph S Bacon

 <https://www.youtube.com/ralphbacon> https://www.youtube.com/ralphbacon

rjshiffer, 2021-06-12:
Hi Ralph,

The version  <https://github.com/RalphBacon/208-Arduino-IDE-Web-Radio-using-a-Task> is an older version than <https://github.com/RalphBacon/205-Internet-Radio> (version 1v29).  Version1v29 does have a separate task for streaming music.

Let me clarify my setup:  I took  the Arduino IDE version 205-Internet-Radio (1v29) and converted it to PlatformIO. (I used  "208-Arduino-IDE-Web-Radio-using-a-Task" as a PlatformIO model).  So I already have an operating PlatformIO version of the radio.   If you could put up a zip file with the latest version (if not private--1v32 shows in your videos), I could simply replace my PlatformIO version, with the noted exceptions for Wifisecrets.txt and calibration, of course.

Regards,
Richard

RalphBacon, 2021-06-13:
Hi Richard

 

Well you certainly made yourself a lot of work by re-converting the Arduino code back to PlatformIO code!

 

Yes, I can zip up my entire project (there will be some external libraries still) and put it in Dropbox, hopefully later today.

 

Do you get choppy audio too or was that just bepaald?

 

Best Regards

Ralph S Bacon

 <https://www.youtube.com/ralphbacon> https://www.youtube.com/ralphbacon

bepaald, 2021-06-13:
Hi Ralph, thanks for your quick reply. After some fiddling around with the wires yesterday and then applying just the right pressure under just the right angle to some of the connections, I actually had perfect audio! So I definitely have some dodgy dupont wires in my collection. I should be able to work around them (or replace them).

Still, after reading the exchange with @rjshiffer, I too would be very interested in the latest code. No need to port it to arduino on my account if you haven't done so already, I think I can manage. It would be _great_ if you would upload the code to one single git repo sometime in the future, which is then updated from time to time. I also initially thought the code at 208 was newer, but looking at the dates that doesn't seem to be the case. But I do realize it'll be a bit of a time investment to set that up.

Thanks again! Regards,
Selwin

RalphBacon, 2021-07-21:
New version v1.35 for the Arduino IDE now available.

closed 2021-07-21 by RalphBacon