RalphBacon/205-Internet-Radio #14: Touch screen problems

pafa17, 2021-06-16:
Hello Ralph
I have just built your excellent radio and most items are working well except for the touch screen. It is the same screen that you used only slightly smaller at 2.4".
The display is perfect but the touch positions don't match. The +,-, Bulb icons can be changed by touching a corresponding position at the top of the screen.
I tried rotating the screen which worked for the background but the button icons stayed in their original positions.
I reverted to original then I ran the touch screen calibration and got " uint16_t calData[5] = [374, 3538, 318, 3518, 3]".
I then substituted this in the folder "TouchCalData1.txt" but still no change.
I am now at a loss as to what to try next. Maybe you could advise

RalphBacon, 2021-06-16:
Hello pafa17

 

I’m glad you’re getting good (audio) results from this project. Now to your problem.

 

When you do the screen calibration it must be in the same rotational state as the ESP32 Web Radio sketch.

 

As the ESP32 Web Radio sketch executes (around line 151 in tftHelpers.h)

 

    // Rotation as required, the touch calibration MUST match this

    // 0 & 2 Portrait. 1 & 3 landscape

    tft.setRotation(1);

 

You must also include that line in the screen calibration sketch. If “1” doesn’t work (it’s mirrored/flipped) try “3”. A bit suck it and see. Not very scientific, I know.

 

It’s been so long since I got this bit working (I too had some issues but I can’t remember if they were the same as yours) that’s the best I can suggest.

 

When the calibration sketch writes "Touch screen to test!" I’m thinking that should be in the orientation that you are expecting to run the screen. I’m trying to remember whether the display orientation is any different to the touch calibration but I’m pretty sure that as long as the screen was in the correct orientation at the time of calibrating it, it should work fine.

 

You can always test this theory by rotating the screen using a number from 1 - 4 (0 & 2 Portrait, 1 & 3 landscape) and see where buttons (touch, not displayed) end up!

 

Let me know how you get on.

 

Best Regards

Ralph S Bacon

 <https://www.youtube.com/ralphbacon> https://www.youtube.com/ralphbacon

pafa17, 2021-06-16:
Hello Ralph,

Thanks for speedy reply.
I have tried the 4 orientations in touch screen calibration with tfthelper line 141 set to (1) and no change whatsoever.

Then I tried changing line 141 to (3) the only thing that flipped was the "ESP32 WEB RADIO" logo ,the v1.29,tune,bulb,+,-,% icons all stayed in their original position and the touch positions never changed position either.

I do not understand
 yÒ
Ò
 ”?@? 
code in the  TouchCalData1.txt file.
 I tried to substitute " uint16_t calData[5] = [374, 3538, 318, 3518, 3]". and
" uint16_t calData[5] = [448, 3432, 343, 3490, 5]" opposite orientation
which I got from touch screen calibrate but it made no difference.
Basically the touch positions stay in their original positions.
Maybe it's my display although I have a second one doing exactly the same thing.

Regards
pafa17

RalphBacon, 2021-06-17:
You may have missed the other orientation commands; see the screen shot where I have highlighted the lines:

 

The first and last of these commands should be the same, the middle one (rotation 0) is just for the v1.32 so it fits into the banner.

 

Check you’ve done the first and last rotation setting.

 

If you’re still having problems, you will need to load another sketch, eg the included examples in lib > TFT_eSPI-2.23 > examples. The Test_Touch_Controller.ino might give you an insight as to how the screen rotation works. I seem to remember I did this before I could understand how it all worked.

 

FYI the default orientation (rotation 0) of the screen is portrait, which confused me, I was expecting landscape as default.

 

 

Best Regards

Ralph S Bacon

 <https://www.youtube.com/ralphbacon> https://www.youtube.com/ralphbacon

pafa17, 2021-06-17:
Hello Ralph,
Thankyou for taking the time to help a 72 year old hardware man attempting his first arduino project with very little experience of software.
First of all I can't see a screengrab so I followed your instruction and  changed the rotation  on line 
141 tft.setRotation(3); and
166 version no. also to setRotation(3)
I do not see any more rotation instructions
So partial success now I seem to have a mirror image in that the + and minus icons are reversed as are the the bulb and speaker icons.
I could try to figure out the coordinates of these and see if I could swap them.
Thanks again for your help

Best Regards
pafa17

RalphBacon, 2021-06-17:
As you have an earlier version of the code the line numbers here might well be different. Here's the screenshot that you could not see.

[image]

I'd like to know what screen exactly you have. You should not have to change anything regarding the positioning of the buttons - doing so indicates that the image is mirrored/flipped so we need to resolve that rather than change code (otherwise on my next code update you will lose your updates).

Can you post a photo of your screen, front and back plus anything you know about it please?

pafa17, 2021-06-17:
Hello Ralph,

Those are the very lines that I changed. 
I now appreciate the amount of work you put into this project with the slight messing about that I am trying.
The display seems to have the correct touch ic  XPT2046
The label on the display says 
2.4" SPI TFT Module 
Driver ILI9341
Dots 240x320 touch yes
SKU MSP2402
Thanking You
pafa17

[image]
[image]
[image]

RalphBacon, 2021-06-20:
I've checked my TFT lib selections and your screen is identical to mine (except for the actual size). So why you are experiencing a flipped touch screen when using a non-modified version of my code baffles me.

Rather than trying to get it to work by changing the code, I suggest loading up the sketch:
I:\Documents\PlatformIO\Projects\ESP32 Better Internet Radio\lib\TFT_eSPI-2.2.23\examples\Test and diagnostics\Test_Touch_Controller\Test_Touch_Controller.ino

And then check that the coordinates are as expected. 

Next try sketch:
I:\Documents\PlatformIO\Projects\ESP32 Better Internet Radio\lib\TFT_eSPI-2.2.23\examples\320 x 240\Keypad_240x320\Keypad_240x320.ino
And see whether the calculator pad behaves correctly.

I remember (vaguely) doing this to get it all working on my setup.

If both these tests work then it is not your screen but something in my code that is either done or hasn't been done but is required for your particular screen.

Have a go with the first two sketches and see what happens before we investigate further.

pafa17, 2021-06-20:
Thanks for the reply Ralph but excuse my ignorance, where do I find both of those files. I am using the arduino ide

RalphBacon, 2021-06-20:
Oh, OK, they are in the library for the screen, in the folder, under examples. Slightly different path but you should find them there.

pafa17, 2021-06-20:
Found them; the test touch controller seems to run forever and unfortunately I cannot interpret it.
The keypad240x320.ino tells me SPIFFS was not declared in this scope.
Sorry for all the hastle but I am a complete noob. I think I might have included a part copy of the test touch controller
[test touch controller.txt](https://github.com/RalphBacon/205-Internet-Radio/files/6682664/test.touch.controller.txt)

pafa17, 2021-06-23:
Hello Ralph
Sorry to be such a pain,I know that you can't lead people by the hand through your projects but finally after watching your video about littlefs & spiffs I finally got the  \TFT_eSPI-2.2.23\examples\320 x 240\Keypad_240x320\Keypad_240x320.ino to work.
I now have the display of the keyboard and when I touch the numbers they all display correctly in the display window so the touch and display sections must be working properly, Any advice where to go from here ?
If you are too busy I will understand and live with the buttons and touch being in reverse

RalphBacon, 2021-06-24:
Hi Ken (I just made that bit up as you are _still_ anonymous!)

Now that you have a working portrait screen, try adding in a line to the sketch to rotate the display using
**tft.setRotation(1);**
as landscape for the parameter - that line is in setup(). 

Some of the keypad keys may disappear off-screen but we’re not worried about that right now.

Do the keys that you CAN see still correspond to the numbers being displayed? In other words, is the touch orientation the correct one for the screen orientation?

Note: You may have to repeat the calibration part of that sketch again first (i.e. change line **#define REPEAT_CAL false** to **true**).

pafa17, 2021-06-24:
AH! Sorry about that Ralph,I didn't realise. It's Pat from a wet and cool Dublin.
I did as you suggested and the display rotated and the buttons I could see 1-6 when touched displayed perfectly in the display window.
Just to reiterate again as to my situation with the radio display,it seems that only the y axis is reversed ie
V1.29 brings up the station pages.
When the + button is pushed the - button lights up and the volume decreases and vice versa
The bulb icon switches on the mute button
On the station pages the < & > are reversed but the station buttons operate correctly.
Thanks again for your assistance
Pat

RalphBacon, 2021-06-24:
Hi Pat! Thanks for the clarification. Quick question: are you using the Arduino IDE version or PlatformIO?

pafa17, 2021-06-24:
Hello Ralph
I'm using the Arduino ide I think it will be a while before I progress to platformio as I am struggling to understand the code in the sketch

Regards

Pat

RalphBacon, 2021-07-21:
New version v1.35 of the Arduino IDE code available.

closed 2021-07-21 by RalphBacon

pafa17, 2021-07-26:
Hi Ralph,

I have used the new version 1.35 and no change in my display issue.
Have you changed anything which may have helped me, if not I will live with
it or possibly buy a third display and see will it work.

Regards

Pat

RalphBacon, 2021-07-28:
Sorry, I have not found any issues with the screen-related code that would affect you.

That said, you could try loading the calibration sketch and extracting the values to be loaded each time (placed into a file in the data folder).

Alternatively, if you buy the _exact same screen_ as me then you will no issues, as everyone else who has done that testifies. Unfortunately, the price has increased a bit to $10.99 not least because Banggood include VAT now in the price (a 20% increase for UK purchasers and whatever the VAT rate is where you live).
Link: https://www.banggood.com/custlink/3GvdgmuHt3

pafa17, 2021-07-28:
Thanks for the reply Ralph.I just did what you said and purchased the
display from Banggood yesterday. I notice the pcb tracks to the xpt2046 is
different on mine compared to your one
Just as a matter of interest how did you get the data in the file
TouchCaldata 1.txt.
Again thanks for all your help the radio is working well otherwise.

Regards

Pat

RalphBacon, 2021-07-30:
@pafa17 
I’m sure the new screen will perform flawlessly, Pat! Drat the VAT!

I got the data into the text file simply by copying what was on screen (sent via the Serial monitor output) and typing it into a text file! I was too lazy to get the program to write to a file, then me having to extract the file to prevent it being overwritten every time. You should only need to calibrate it once.

Let me know how you get on.

rjshiffer, 2021-08-04:
@pafa17
Hi Pat,

I see that your question about the contents of the "TouchCalData1.txt" file
was not answered.  I too had trouble understanding and creating
the "TouchCalData1.txt" file.   I had to create mine manually. I don't
know a shorter way. Maybe this will help:

Short Answer: The TouchCalData1.txt file contains the 5 2-byte integer
values shown in the serial output of the touch screen calibrate sketch
**EXACTLY** in the format stored in the ESP32 memory (i.e, uint16_t).

Long Answer:  Let's use your data from your 1st post:

>uint16_t calData[5] = [374, 3538, 318, 3518, 3]

1. These are in decimal, so first convert to hex:

>374  = 0x'0176'
>3538 = 0x'0dd2'
>318   = 0x'013e'
>3518 = 0x'0dbe'
>3       = 0x'0003'

2. The ESP32 is what's called a "Little endian machine". That is, it stores 
data little-end first. When looking at multiple bytes, the first byte is 
smallest. For example, 0x'0176' is actually stored in memory as 0x'7601'.
Don't ask me why.

3. Flip the bytes to match EXACTLY the ESP32 format:

>374  = 0x'0176'  -> 0x'7601'
>3538 = 0x'0dd2'  -> 0x'd20d'
>318   = 0x'013e'  -> 0x'3e01'
>3518 = 0x'0dbe'  -> 0x'be0d'
>3       = 0x'0003'  -> 0x'0300'

4. So the TouchCalDate1.txt file should be (in binary):
 
>7601d20d3e01be0d0300

   @RalphBacon's file is:

>7901d20d0401d20d0700.......... 

   @RalphBacon's file has more bytes but they are not used.

5.  How do you enter this binary data in the file?  I use the VIM editor
that permits binary editing. If it's easier, I created a TouchCalData1.txt
file using your data and is attached.  Save the original file in
the data folder, replace it with the attached, upload the LITTLEFS file system
using the Arduino IDE. 
[TouchCalData1.txt](https://github.com/RalphBacon/205-Internet-Radio/files/6934469/TouchCalData1.txt)

6. Monitor the serial output from ESP32 Radio to verify your touch calibrate numbers are used.
   
7. Good Luck.

P.S. The extra 2 bytes in the file are CR-LF and are ignored.

pafa17, 2021-08-05:
Hello rjshiffer
Thank you very much for taking the time to explain the data in CalTouchData1.txt.
That is very interesting and quite convoluted.
I did as you suggested and unfortunately the display now works diagonally ie touching bottom left affects top right.
Thinking about it, it occured to me that when I did the original touch cal sketch I didn't set the rotation to 3.
I have rerun the touch cal again and got these values
( 436, 3448, 235, 3458, 5 ) I converted these to hex and got
0x01B4. 0X0D78, 0X0EB, 0X0DB2, 0X0005  which I then changed them to
0XB41. 0X780D, 0XBE0, 0XB20D, 0X0500
Which gives me 
B41780DBE0B20D00500
I cannot figure how to get this to the data used in the CalTouchData1.txt file
Any help would be much appreciated.

Regards

Pat

rjshiffer, 2021-08-05:
@pafa17 

Hi Pat,

You have a few mistakes.  3458 is 0x'0d82' and you're missing some zeros in the final string of digits.

I made another file:

>b401 780d eb00 820d 0500  <-- digits

>b401780deb00820d0500      <--- file contents

[TouchCalData1.txt](https://github.com/RalphBacon/205-Internet-Radio/files/6940210/TouchCalData1.txt)

I didn't give rotation any thought when I generated my numbers.  I don't know how that will effect the display.  I think we're in "trial and error" mode.

Richard

RalphBacon, 2021-08-05:
Guys, remember you must set the rotation correctly (for the orientation you intend to use the screen) when running the calibration sketch, otherwise the universe will (probably) implode. 

As the ESP32 Web Radio is setting the screen to landscape (default is portrait) set the rotation to:
`tft.setRotation(1);`
in your calibration sketch to get the right numbers.

Good luck and if the universe _does_ implode I'll know why.

pafa17, 2021-08-06:
Hi Richard,
Thanks again for your help and sorry for my mistakes.
I ran the new data that you provided and looking at the serial out,the data shows cal data
01B4, 0D78, 20EB, 0D82, 2005.
Does it look as if the 3rd data is wrong (20EB)
No Touch working at the moment

Pat

PVsolar, 2021-08-06:
Hello Richard, Ralph,
I am following your discussion with Pat with more than normal interest; I have simmilar problems (sorry, challenge it should be called). In V1.29 all positioning was ok (but a few other issues)
In V1.35 everything else is solved but the buttons at the lower line (SET, bulb, +, - , and speaker) can only be touched in the area bolow the button. It looks the sensitive area has shifted down exactly the button size. The Station button is right at the correct location. At the station pages the touch position of the stations is ok but also the lower line buttons have to be touched below the button. So I am not sure there is a calibration issue because it not linear from top correct to bottom shifted but 80% top is ok and 20% bottom is shifted. I have modified the position of the bulb and speaker touch area but for the other buttons I have not found the location of the position setting yet. @Ralph, can you give me an indication? But as said before, I do not like to modify in the program as newer releases have to be modified as well. I rather understand the touchcall procedure.
 I have understand following:
Result from the calibration.  uint16_t calData[5] = { 342, 3477, 640, 3110, 7 };
my calculations:
Hex: 		0176, 0D95, 0280, 0C26, 0007
LSBfirst:	        7601, 950D, 8002, 260C, 0700
To VIM editor:	7601950D8002260C0700
But what should the VIM editor do? When I open the touchcall data txt file with a notepad I see some symbols and weird signs but unreadable and less characters as in the list above. It looks simmilar to the result as opening a mp3 file in the notepad. So to which program or code should it be converted?
I am very happy with the result so far and could live with this small deviation but I am also curious and want to learn.
Kind Regards, Hans

pafa17, 2021-08-07:
Hi Richard,

Again apologies,when I had first built the radio I had changed the rotation in tfthelpers file to rotate the display and the data that you worked on originally was pre rotation.
So after thinking about it for a while I realised this and when I reverted the rotation to the original settings the display now functions perfectly,
Thank you for your concise explanation and new data. I would never have been able to fix it myself.
It is good to know that there are people out there that will help novices like myself

Best Regards

Pat

rjshiffer, 2021-08-09:
@pafa17

Hi Pat,

Good news on getting your display working.  It looks like my bad eyes
and bi-focals made errors too on that 2nd file.  Good that the 1st file
worked. Glad I could be of help.

@PVsolar

Hi Hans,

Here is an updated TouchCalData1.txt file for you.

For:

> uint16_t calData[5] = { 342, 3477, 640, 3110, 7 };

I get:

>Hex: 0156, 0d95, 0280, 0c26, 0007
>LSBfirst: 5601, 950d, 8002, 260c, 0700
>File: 5601950d8002260c0700

Notice that for 342 I get 0x'0156'

Here's the file.

[TouchCalData1.txt](https://github.com/RalphBacon/205-Internet-Radio/files/6951987/TouchCalData1.txt)

Save the original file in the data folder, replace it with the attached,
upload the LITTLEFS file system using the Arduino IDE. Monitor the serial 
output from ESP32 Radio to verify your touch calibrate numbers are used.

Although I use the VIM editor, you may find that the Notepad++ editor
with the HEX-Editor plugin installed an easier way to create these
binary files.

PVsolar, 2021-08-10:
Hello Richard,
Thank you for the translation (and correction of my typo) of the calibration data.
Unfortunatly it did not work. As you can see in the monitoring data the program could not find the caldata. Using the old file did work as intended.
I installed the HEX converter in notepad++ and played with the data and formats. I still does not understand why in the standard notepad all those weird characters are showed, but notepad++ seems to understand it and converts it back to readable numbers. I noticed two things: in the converted file the last figure 0007 was converted into 2007. That file contains two lines while the original file has three. Maybe not of interest but it shows I do not understand what I am doing. The result of using the new file is that the touch point of the lower line buttons lies on the lower edge of the title block and the station button cannot be used. I did do some calibration repetitions and the outcome gives different but close to each other numbers. Can you explain step by step what I have to do to the conversion via notepad++?

Kind Regards, Hans

[image]
[image]
[Calibration history.txt](https://github.com/RalphBacon/205-Internet-Radio/files/6962055/Calibration.history.txt)

PVsolar, 2021-08-11:
Hello Richard,
It works perfectly! Thank you for spending so much time for us. I don't know what is different now compared to yesterday. The only thing I can think of is that I got several times error messages from littlefs when uploading Data. Sometimes the third or forth time it worked without changing anything. (monitor was off all the times) Today the first attempt was right and all buttons are touchable perfectly in the centre. The data is identical to the CalData except for the 0007 which is now 2007 but it seems not of influence.
Kind Regards, Hans
[image]

rjshiffer, 2021-08-12:
Hello Hans,

Good news that the button selection is working fine.  Although I'm curious about the last value showing '2007' rather than '0007'.  That means the last part of the file contains x'0720'.  How can that be? 

I ran the file I sent you and I get:
>[V][tftHelpers.h:87] initDisplay(): TFT Calibration data:
>[D][tftHelpers.h:90] initDisplay(): 0156, 
>[D][tftHelpers.h:90] initDisplay(): 0D95, 
>[D][tftHelpers.h:90] initDisplay(): 0280, 
>[D][tftHelpers.h:90] initDisplay(): 0C26, 
>[D][tftHelpers.h:90] initDisplay(): 0007, 
>[I][tftHelpers.h:94] initDisplay(): Touch Calibration completed

Can you verify that the last bytes in the TouchCalData1.txt file you are using ends in x'0700'?  I don't know the implications if some future update to the code will cause a problem with the wrong value.

Two solutions:

1.  If you are adventuresome use notepad++ and the HEXeditor to change  x'0720' to x'0700' in the file and upload the data/ folder.

or

2. Simply re-download my file from a previous message to ensure you are using an un-modified file, verify x'0700' at the end (via notepad++, etc)  and then upload the data/ folder.

Richard

PVsolar, 2021-08-12:
Hello Richard,
I noticed the change from 0007 into 2007 already in your first converted file. As there are many different apearances fro the same data (which I do not understand all) I made an overview of all my actions.
First, the difference in line number from tftHelpers (h:90 in yours and h:92 in mine) comes from two modifications I made for the location of the bulb and speaker to try to shift the touchpoint to the icon. I left the original location but commented out so two additional lines. I deleted my modifications and uncommented the original lines because the caldata works fine now. In the monitor output the lines are now h:90 also. (but still 2007)
I made a file "Calibration history.txt" but to show what I see in the various programs and settings I made a screenshot called caldat_history12-8.PNG. In this file you see the different actions I took from calData[5] output until the translation I got from you (first time) but showed in the "simple" notepad. (Kladblok means Notepad)
When I open the TouchCalData1.txt in Notepad++ I get the lines as shown in the lowest line in screenshot caldata12-8. (VSOH.....etc.) I copy/pste these lines in a file called TouchCalData11.txt (just to have another name) and activated the HEX in plugins. Only the 0156 is recognizable but the rest... I don't know. BUT, when I in the original TouchCalData1 (lowest line) switch to Hex via plugins I get the output on the top. Exactly as in the monitor output. At the end of the HEX files some characters are showing up under Dump but I think that is not of interest.
I hope you can make something up from what happens here. Espescially the difference between copy/past> HEX and direct showing in HEX I do not understand at all.

Kind Regards, Hans

[image]
[image]

rjshiffer, 2021-08-12:
Hi Hans,

None of the files you show me match the file I sent you.  So, last try:

For:

uint16_t calData[5] = { 342, 3477, 640, 3110, 7 };

Download this file (my original):

[TouchCalData1.txt](https://github.com/RalphBacon/205-Internet-Radio/files/6976624/TouchCalData1.txt)

Don't do anything but place it in data/ folder, upload to ESP32.

Good Luck.

Richard

PVsolar, 2021-08-13:
Hello Richard,
Mistery solved???????
There is nothing wrong with your files. I have played around with all versions in the "simple" notepad, notepad++ in standard view (whatever that is) and in HEX view. Finaly I found a strange behaviour. When I open your file(s) it opens in simple notepad. From there I stored it in the data folder and the result is 2007 in the monitor. When I open that file from data with notepad++ I also see the 2007. When I force to open your file directly in notepad++ and from there store it in data I get 0007 in the monitor. Viewing that file in HEX I see the 0007 also. So the difference is going through the simple notepad or not. In the HEX view there is, beside of the 2007 and 0007 a difference in the dump partition. The 0007 ends with & . . while the 2007 ends with & _ .  So one dot becomes a space. Maybe this dump information is missing in the simple notepad. Do you have an explanation for this behaviour? Opening either file in simple notepad afterwards gives identical result in characters.

Kind Regards, Hans

rjshiffer, 2021-08-13:
Hello Hans,

Yes, you have discovered the mystery of the corrupted file.  Quite simply, if you save the file by "simple notepad", x'00' is changed to x'20' an ASCII space character.  Lesson learned: never edit a binary file with "simple notepad", a text editor. 

@pafa17, Pat, this may explain the changes you saw in your post https://github.com/RalphBacon/205-Internet-Radio/issues/14#issuecomment-894175828
where x'00eb' was changed to x'20eb' and x'0005' was changed to x'2005'

Hans,  if you are interested in a little self-study/homework this weekend here are two web sites that will explain the "weird characters" in notepad you didn't understand:

This first site uses a different hex editor in its explanation but its display is similar enough to Notepad++/Hex-Editor:  [A little diddy about binary file formats](https://betterexplained.com/articles/a-little-diddy-about-binary-file-formats/)

This web page explains why the bytes had to be swapped after we converted from decimal to hex: [Understanding Big and Little Endian Byte Order](https://betterexplained.com/articles/understanding-big-and-little-endian-byte-order/)

RIchard

PVsolar, 2021-08-14:
Hello Richard,
I never realized the TouchCalData1 file is a binary file because it is a .txt file. Unless I use "open with..." such files are automatically opened in the text editor, at least in the windows 10 settings I use.
I am glad that the confusion is cleared now.
I like your suggestions to study those sites. It has been a long time ago I learned about one's and two's complement etc. and I never used it anymore. So time to refresh and extend.
Tanks again for spending your time,
Have a nice weekend, Hans