RalphBacon/205-Internet-Radio #15: Simple Schematic Question

geoffw123, 2021-07-06:
I would like to have a crack at building this web radio, but after having downloaded the latest repo I cant find the schematic diagram to work to. Have I missed it, or is it missing ? A link to it here would be great please. Thanks for your help (Ralph or anyone that has a copy of the schematic)

RalphBacon, 2021-07-06:
Just before I dig out that info, Geoff, can you tell me please:

1) Are you in the UK?
2) Can you use PlatformIO as your IDE of choice or are you an Arduino IDE person?

Whilst you answer this I'll dig out that information :)

geoffw123, 2021-07-06:
Hiya Ralph 
Thanks for the speedy reply, a PlatformIO Brummie here :)

If I can ask another simple question please, are you selling any excess supplies of  your WebRadio PCB, or maybe your repo provides your  Gerber files for sending to JLPCB?
Thanks

RalphBacon, 2021-07-06:
Hence my reason for asking (although commiserations for having to live in Birmingham, but I guess someone has to 🤣)

Ok, enough foolishness (not to mention rudeness 😁)

Yes, I have a spare PCB. In fact, two, one for the motherboard and one for the screen.

If you are going to use these PCBs there are some other, specific, parts you will need.

- 1 x TFT **Touch** Screen link: https://www.banggood.com/custlink/DmDyHMw4SR 
- 1x 10-way (2 rows of 5 pins) IDC female cable (the same as used in programming the Arduino with an ISP)
  Like this one: https://www.hobbytronics.co.uk/idc-ribbon-cable-12-10?keyword=IDC 
- 1 x 20-way (2 rows of 10 pins) IDC female cable to connect the screen to the motherboard
  Like this one: https://www.ebay.co.uk/itm/220mm-IDC20-IDC-Female-Female-Flat-Ribbon-Cable-20-Pin-Gray/272723050970
  Or this one: https://www.ebay.co.uk/itm/114032991586?ViewItem=&item=114032991586 
- 1 x 470μF capacitor (at least 10v but don’t go mad as they get bigger the higher the voltage) or similar (whatever fits nicely on the board, you can go as low as 100μF)
- 1 x 10kΩ resistor
- 1 x two-pin pushbutton (optional just for testing but mandatory if you don’t have the screen). 
- 1 x 100nF SMD capacitor (optional, a case of belt-and-braces)
- 2.5mm (0.1”) pin headers just like on the Arduino, but for the IDC cables to plug into. Total pins approx. 80 pins, very cheap to buy!
- 2 x 15-way female sockets for the ESP32 to plug into: https://amzn.to/3axmvsf

You can also make your own IDC cables of course, something I did too. Just be very careful of the cable orientation or the whole thing will be reversed which will only end badly when you plug it in.

None of the above is particularly pricey, although the Chinese sites now seem to add VAT on as per the EU directive (hang on, didn't we Brexit a while back?). Since I did this project they increased the price of the screen to $11.99 (was $6.99) such is my influence! At least shipping is free to the UK.

Anyway, if you think you can source those components, I'll be more than happy to send you the PCBs FOC, and as you're a PlatformIO developer it makes things _very_ much simpler with the up-to-date code (I'll send the entire project).

Let me know your thoughts.

RalphBacon, 2021-07-06:
Most importantly are the schematics, which you can find here:
https://github.com/RalphBacon/208-Arduino-IDE-Web-Radio-using-a-Task/tree/main/Schematics%20(subject%20to%20change)
Note that this GitHub, although a later video than #205, doesn't contain the latest code (but as you're PlatformIO you get a newer version anyway).

I didn't mention items such as the VS1053 decoder or the ESP32 WROOM TTGO v1.7.1 as I take these requirements as read but if you need to know where I got them from (The ESP32 from a UK supplier) let me know. 

The components really need to be an exact replica of what I have if the PCBs are going to work.

geoffw123, 2021-07-06:
Hi Ralph

Many thanks for the detailed reply. I have already got most of the parts for this, I am going to build it on a breadboard to begin with and see what problems I hit. Just a matter of finding the free time and the brain power to do it !  I will give you a shout in the future if I have success and would like to put it on a pcb. Thanks for the offer of your spare board.

RalphBacon, 2021-07-07:
Good luck with your build; it will be much more difficult on the breadboard than a PCB, of course; as you saw in my video there was a huge number of DuPont wires I had to use! The PCB construction is pretty much like Lego, plug and play. No, that's _play_ not _pray_. 😁

When you need the newest code just let me know; I may have organised myself enough to get the PlatformIO repository up-to-date by then :)

If you get stuck let me know how far you've got, the stumbling block and I'll try and help.

RalphBacon, 2021-07-21:
New version v1.35 of the Arduino IDE code now available.

closed 2021-07-21 by RalphBacon