RalphBacon/205-Internet-Radio #16: Some stations give no sound

PVsolar, 2021-07-15:
Hello Ralph,

Thank you for this great project. I already build an ESP32 radio but without touch screen and without buffer. I learned a lot from that project (originally I am a hardware guy) but your design is next level. It gave me a lot of learning moments the last weeks and now the radio is working and I am very happy. To learn even more I will modify things, like more than one AP to be selected automatically or on screen and truncating the music title because it now sometimes is printed over the buttons. 
But one thing is very important and I don't know what to do: Some radio stations can be contacted, deliver meta data and I can see the bytes are streaming in. But no sound is heard and the buffer remains empty. Selecting the same station on my "old" radio produces music so the station is sending. The difference with working stations is that the logging of the stream is continuing endless and in portions of 2 to max 28 b, while the working stations after some time produce 32b continiously and stop logging because the buffer is full. Can you give me some direction how to get these stations working? The one I like most for the moment is: 208.92.53.98,3690,"/SRGSTR25AAC_SC","Classic FM Chill Out",1
Thank you in advance, Hans.

RalphBacon, 2021-07-15:
@PVsolar 
Hi Hans! 
Last things first. I added that station to my list and it plays normally. With metadata.
[image]

However, you did not tell me whether you are using the very, very, very old Arduino code or the slightly less old PlatformIO project?

The code has moved on a bit (now V1.35) but not yet uploaded to GitHub. It now deals with long song titles/artists and reduces the font title until it fits. for example. Plus a few dozen other (mostly behind-the-scenes) fixes. Like working with chunked data on the BBC and HTTP/1v1.

If you are using Arduino this is a problem as I have promised others to upload newer code but this takes time (to convert from PlatformIO to Arduino IDE).

Let me know what IDE you use first please and we will take it from there.

PVsolar, 2021-07-16:
Hello Ralph,
Thank you for the quick response. I am glad to see my issue has been solved already. I also see a new button in the lower left corner so I have to update. I am using version 1.29 from your 205 video on Arduino and Arduino IDE. As said, from education I am a hardware guy and could up till my retirement leave the software to the specialists in my department. I started two years ago learning the Arduino IDE and have found my way more or less but still learning every day. Because of your enthousiasme I want to try to dive in platformIO but I am afraid to be confused and mix both platforms. If it's necessary to follow your updates is a motivator. Originally I used the same ESP32 version as in your earlier videos and there I had the same problem where ti connect the reset. Connecting to enable (pin 1) did not work. So now I have the same T8 version as in your 205 video.
Hans
[image]

RalphBacon, 2021-07-16:
I think it best that I update the Arduino version, not just for you but for many others too. PlatformIO is not something you can pick up in an hour (I'm still learning after several months - yes, I'm obviously a slow learner). 

I will endeavour to update the code in video #205 to V1.35 very soon; if you do not hear anything in one week please nudge me again.

BTW that "SET" button in the bottom left of my screenshot is being developed still, to allow adjustments to the bass, treble and anything else I can think of. But it is not finished yet so don't get too excited about that!

I like your slimline case for the radio - did you 3D print that yourself?

PVsolar, 2021-07-16:
Hello Ralph,

You are fast!
So no need to dive into platformIO yet. I’ll wait for your update to V1.35. I tried to subscribe your channel but it seems I have to have a google account first. Is that true?
The housing is from Otto, Germany but as far as I could find not in production anymore. There are comparable housings from e.g. Hammond. The “secret” in this compact setup is the use of special Ampmodu connectors for the display. The connector body is on the component side but the contact entrance on the solder side. Also I soldered the pins on the ESP at the component side so the components are “hanging” between the headers. The location of the ESP32 and the VS1053 are such that the usb and audio socket are just protruding the side walls. A power inlet fits just beside the ESP and some C’s found a place inbetween. The interconnection board is so simple that I could make it with discrete wires but as always, I made a layout first. Mechanical work at the housing took most of the time. If you are interested I can make a 3D printable housing. That’s another direction I started learning last two years.

Have a nice weekend, Hans
[image]
[image]
[image]

RalphBacon, 2021-07-21:
@PVsolar 
Good afternoon, Hans!

The new v1.35 is available from this GitHub for the Arduino IDE. I've downloaded it, compiled and uploaded to my test rig and all seems fine.

If you fancy a go please let me know how you get on by opening a new Issue!

closed 2021-07-21 by RalphBacon