RalphBacon/205-Internet-Radio #17: Programm too big; continiously rebooting

PVsolar, 2021-07-21:
Hello Ralph,
Thank you for your swift action to convert version 1.35. Unfortunately it did not succeed here up till now.
Because I was working on some older programs I had to replace some files before the compilation id work (cbuf, VS1053)
I also changed the playlist and wifi secrets for my own data.
Then the compilation succeeded but a failure notice poped up the the program was too big for memory.
I put the debug mode to none and that helped somewhat but I had to remove the ToDo.h to get it small enough.
After the upload the processor continiously rebooted.
I have to prepair for tomorrow now and tomorrow I have no time at all so I will continue looking for a solution on Friday and the weekend. If you have a suggestion in the mean time that will be welcome.
Kind Regards, Hans

[reboot data_V1.35- 20210721.txt](https://github.com/RalphBacon/205-Internet-Radio/files/6856755/reboot.data_V1.35-.20210721.txt)
[image]

rjshiffer, 2021-07-22:
@PVsolar, I  also had continuous reboots after installing v1.35 until I realized that I did not change the size of the Circular Buffer to 10000 if the changes to cbuf.cpp not installed.  The boot error messages seem identical to yours.

I changed the #define found in main.h near line 100:

    #ifdef BOARD_HAS_PSRAM
    //#define CIRCULARBUFFERSIZE 150000 // Divide by 32 to see how many 2mS samples this can store
    #define CIRCULARBUFFERSIZE 10000
    #else
    #define CIRCULARBUFFERSIZE 10000
    #endif

and the radio worked fine without other changes.  Maybe that is also your problem.

@RalphBacon, I have since changed cbuf.cpp and the Circular Buffer size to 150000 and it works without errors.  Even the station that I could not hear the audio now works!  Thanks much.

RalphBacon, 2021-07-23:
@rjshiffer, that's good news! I've since discovered that I need to change the cbuf.cpp to include some compiler directives (rather than a C++ "if" statement) otherwise if you compile a different sketch for the ESP32 you can get compilation errors. I will update in due course on my GitHub.

My cbuf.cpp file resides in this folder:
C:\Users\Ralph\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32

If you wish to update your cbuf.cpp file now, this is the change required:
```
//char *newbuf = new char[newSize];
// RSB Use PSRAM here if required
char *newbuf;
#ifdef BOARD_HAS_PSRAM
    newbuf = (char *)ps_malloc(newSize);
#else
#warning "PSRAM not defined or used"
    newbuf = new char[newSize];
#endif

You can remove the warning if you're not bothered about being told about no PSRAM.
	char *oldbuf = _buf;
```

PVsolar, 2021-07-23:
@rjshiffer; Thank you for your solution, it worked!!!! But that made me think: why does this work? I had installed the V1.29 cbuf.cpp so why going back to the smaller size? Just to verify, I checked the version of cbuf.cpp and that was right (V1.29). So without psram the original cbuf.cpp should work also. No way, again rebooting. Going back I changed to the cbuf.cpp from the V1.35 version. That did not work either. But with all this changes I noticed something:
@Ralph: I do have three versions of cbuf.cpp now: the original with on line 21 "#include "cbuf.h" and further no includes. The version that came with V1.29 with on line 21 - 24
"#include "cbuf.h"
"#include "esp_system.h"
"#include "esp_himem.h"
"#include "esp_heap_caps.h"
and finally the version that came with V1.35 with on line 21 and 22
"#include "cbuf.h"
"#include "esp32-hal-psram.h"
At this moment the version V1.29 is installed and the modification of @rjshiffer is in main.h The radio is working except for the "problem station", that still does not work. The difference with before is that the buffer regurarly shows some content but immediately drops to zero again and no sound is heard. (on the other radio still playing)
So for me is the question now: How can I make use of the buffer and which or maybe another version of cbuf.cpp should I use?
Regards, Hans

RalphBacon, 2021-07-23:
I've updated the instructions for updating the cbuf.cpp file for v1.35 which will ensure you use PSRAM.
https://github.com/RalphBacon/205-Internet-Radio/blob/main/Arduino%20IDE%20Version/Circular%20Buffer/readme.md

Ensure you take the entire Arduino code set for v1.35 otherwise we'll all be very confused!

PVsolar, 2021-07-24:
Hello Ralph,
I tried the 1.29 version of cbuf.cpp because the 1.35 version did not work.
I now replaced it by the 1.35 version. In combination with a smaller buffer size the program loads (up to 100k) but larger gives the reboot again. In all cases the used PSRAM is 0. The 1.35 version cbuf.cpp is not completely identical to the modifications in the "read me"  so I copied these modification in it. No difference.
During al these experiments I saw Littlefs failures and the program failed. Clearing all PSRAM and reloading and quitting and starting the arduino IDE again (to not use cash items) it succeeded to get the program working again but still no PSRAM in use. I hope you have a suggestion.
Have a nice weekend, Hans.
[image]
[image]

RalphBacon, 2021-07-24:
@PVsolar  Hans, I believe your Arduino IDE is not picking up the cbuf.cpp ESP32 framework as expected. I've had this before and nothing I did would make it recompile the ESP32 framework.

Can you please confirm the following for me:

1. The cbuf.cpp file you edited is found in:

RalphBacon, 2021-07-24:
@PVsolar  While you're at it, please just confirm your Arduino IDE Boards Manager entry for the ESP32 looks like this:

[image]

PVsolar, 2021-07-24:
Hello Ralph,
Both questions affirmative. The board manager is identical and as you can see the cbuf.cpp is in the same location. ("gebruiker" is Dutch for user)
BY the way, I had great trouble when I started this project to get littlefs running. In contrast with your instructions in the video I had to locate the mklittlefs.exe in the tools of 1.0.6. Could that be of influence?
Hans
[image]

RalphBacon, 2021-07-24:
@PVsolar 

Yes, having the LittleFS tools in the tools folder of the core is correct:
[image]
(I have updated the information in my LittleFS video as so many users were having issues).

There are a couple of things you can now try to get the Arduino IDE to recompile the ESP32 framework and pick up your edits.

1. Open the Arduino IDE with your sketch loaded.
2. Change the board to another ESP32 board.
3. Recompile. You should see the following lines:

PVsolar, 2021-07-24:
Hello Ralph,
Regrettable. All lines come as you described. I did use ESP 32 Dev board all the time and now changed to ESP 32 Wrover. First attemt did not work so I changed back to DEV board (whiched worked but without PSRAM) and back again to Wrover. Now it works but still without PSRAM.
Does using another board help?

Hans

RalphBacon, 2021-07-24:
@PVsolar 
Well, Hans this will be a little more work for you (but not much) to find out why the edited core is not being used.

1. Follow the **new** instructions to update the `cbuf.cpp` file, from here:
https://github.com/RalphBacon/205-Internet-Radio/tree/main/Arduino%20IDE%20Version/Circular%20Buffer
(This now uses compilation `#ifdef` conditions rather than C++ `if` statements)

2.  Compile a simple Blink sketch (so it is quick) but choose a random ESP32 board (_not_ the WROVER).

If the edit has worked you will see a compilation warning like this:
```
C:\Users\Ralph\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\cbuf.cpp:57:2: warning: #warning "PSRAM not defined or used" [-Wcpp]
 #warning "PSRAM not defined or used"
```
Note: the warning will be amongst many, many lines so you may miss it if you do not pay attention! You can scroll back when the compilation has finished to double check.

If there is _no warning_ (and you followed the instruction in step 1 above) then it cannot be compiling the correct core.

Let me know you get on!

PVsolar, 2021-07-24:
Hi Ralph,
I checked the cbuf.cpp again and corrected some tab settings but all lines are as in your instructions.
I used ESP 32 pico kit and compiled several programs from examples. Blink did not work because "ledpin" was missing. None of the programs generated the warning. I also searched with ctr+F but everything seems ok.
I will try some other boards tomorrow and let you know the results.
Thank so far for the help, Hans

PVsolar, 2021-07-25:
Hello Ralph,
Partial success!!! PSRAM is available now and used. Don't ask me why but I did do following:
Used various boards and example programs but never got a cbuf warning. I noticed that with each board change I had to go through all settings in tools. (didn't do that in some ocasions before) Then I went back to wrover module and noticed there is no line to enable/disable PSRAM. However, I did try it and from now on also renewed data update also after renewing all tool settings. Back to DEV board and all over again. Then I went back to the original cbuf.cpp. No problem to load the program but still no result. Back again to the 1.35 version of cbuf.cpp AND IT WORKED. The only thing I can think of is the consequent checking the tool settings and data upload. Now I switched over to wrover module, all data and settings renewed except for PSRAM enabled because that is not there. And that also works.
The only minor thing is that my classic fm still gives no sound. In the monitor (verbose) I see 10 times low buffer and restart. The time in between the attempts is much shorter now compared to 1.29. Did you changed anything in that part? Most of the time buffer content is 0 but always the second attempt is appr. 25% and the next 0 again. Counting starts again until 10 or meta data. The content type is audio/aacp.
Also the TFT calibration seems different because I have to touch below the buttons for reaction but that I should be able to solve myself. Also the text in the station menu is shifted half the sceen as was in version 1.29. There I solved that in the program so could be done here the same I asume.
So if you have some hints to get the classic fm working I will be very happy.

Hans
[image]

RalphBacon, 2021-07-25:
@PVsolar 
Well done on getting it working, Hans, even if it was a bit "hit and miss"!

The station listing offset is a bug in the library I am using; if you cannot solve it quickly let me know and I will dig out the simple, one-line solution.

Classic FM is playing here perfectly OK (I've been listening to it this morning). **NB the station.txt entry is this one:**
21293.live.streamtheworld.com,80,/SRGSTR25.mp3,"Classic FM Chill Out",1

I've attached what the (verbose) log looks like when I select that station
[Classic FM Chill Out - log.txt](https://github.com/RalphBacon/205-Internet-Radio/files/6873711/Classic.FM.Chill.Out.-.log.txt)

Ensure you are using the above entry in your stations.txt file and check your output against mine.

PVsolar, 2021-07-25:
Hurray,

Thank you Ralph. Your weblink dit work immediately. I do not onderstand why my web address works on the other radio and not here but there is a good saying:
If you don't understand, admire it.
I see your output is coming from main.cpp and mine from web_radio.ino. I have the setting in tools "verbose". Is there an explanation for this difference?
Thanks again. Hans
[monitor output Hans.txt](https://github.com/RalphBacon/205-Internet-Radio/files/6873838/monitor.output.Hans.txt)

PVsolar, 2021-07-25:
I see the the content type is audio/mpeg. Could that be the reason?

RalphBacon, 2021-07-25:
My output is from my VS Code (PlatformIO) "main.cpp" program which in Arduino IDE speak is "web_radio.ino". But the code is (at the moment, at least) identical!

Glad you got it working, Hans. Nothing else to do but to listen to the music? All done?

RalphBacon, 2021-07-25:
> I see the the content type is audio/mpeg. Could that be the reason?

I can listen to both URLs for that station at the moment but sometimes the non-MP3 one stops in the way you describe. So best to use the one I gave you and see how that goes.

PVsolar, 2021-07-25:
I'll check the TFT calibration for better allignment of the buttons and the position of the text in the station list so you can see the whole line. I understood from Pat that his alignment problem is not solved with version 1.35. So we will discuss that in direct e-mail.
I use the radio all day when behind the computer so now with display. I am happy.
After that, up to the next project.

RalphBacon, 2021-07-25:
The station text title is fixed by a simple edit in this file (search for RSB):
**\TFT_eSPI-2.2.23\Extensions\Button.cpp**

Just replace your TFT_eSPI-2.2.23 library with the one in my GitHub and the station text list will correct itself.

PVsolar, 2021-07-25:
You already solved all the issues. I have to think what the next project can be.

Hans

closed 2021-12-30 by RalphBacon