RalphBacon/205-Internet-Radio #18: Arduino Version fails to compile

tadder, 2021-07-30:
Hi Ralph, my building work has finally been finished, it seems like a long time since I last worked on the radio, good to see you back to your regular slot on YouTube after your move.

Thanks for compiling an Arduino Version of the Internet Radio. I updated the esp32 core to 1.0.6 and used your modified cbuf.cpp file, Version 1.5 now compiles fine until it hangs at line 126 with the message:

	'class VS1053' has no member named 'setAudio'
	Could not find boards.txt in C:\Users\Tadder\AppData\Local\Arduino15\packages\esp32\	hardware\esp32\1.0.2. Is it pre-1.5?

There are two versions 1.0.2 and 1.0.6; 1.0.2 does not contain ‘boards.txt’. so I tried removing 1.0.2 to force the use of 1.0.6 but no joy.

The ‘Adafruit VS1053 Library’ is at Version 1.2.0.

I suspect I am missing something basic, but wonder if you can point me in the right direction please?

RalphBacon, 2021-07-30:
This means that you have not replaced the "official" version of the VS1053 library with the one in my GitHub, to which I've added a function (or two). You can find the library you need here, just put it in your usual "libraries" folder (which is in the sketches folder, usually).
https://github.com/RalphBacon/205-Internet-Radio/tree/main/Arduino%20IDE%20Version/ESP32-WROVER_Web_Radio/libraries

I know I shouldn't really modify other people's libraries like this but it's a work in progress!

RalphBacon, 2021-07-30:
@tadder I've just noticed you had TWO queries. 

For the query regarding why "boards.txt" is missing, well, my boards.txt can be found here:
C:\Users\Ralph\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6

I don't think it's changed since the last (1.0.2) release so I've attached it here for you but you really should not have had v1.0.2 _and_ v1.0.6 installed at the same time, sounds like a recipe for disaster (or, at the very least, confusion). Did you _update_ to 1.0.6 or _install_ that version (whilst keeping the previous one)?

You could try uninstalling all versions of the ESP32 framework (via the Boards Manager) and then installing only the latest.

[boards.txt](https://github.com/RalphBacon/205-Internet-Radio/files/6908172/boards.txt)

tadder, 2021-07-31:
I can confirm that the libraries folder is contained in the same folder as the ESP32-WROVER_Web_Radio.ino the date on this folder and all the files therein is 26/7/21. So I assume this is correctly using your modified libraries?

I deleted ESP32 framework and then installed 1.0.6 and this seems fine now (no reference to missing boards.text)

Unfortunately I seem to have swapped the original problem for a new one:

the compilation now terminates at line 21 of 'main.h' (#include <Preferences.h>)
with error:
_In file included from C:\Users\Tadder\Documents\Arduino\RalphBacon\July21\205-Internet-Radio-main\Arduino IDE Version\ESP32-WROVER_Web_Radio\ESP32-WROVER_Web_Radio.ino:2:0:
main.h:21:10: fatal error: Preferences.h: No such file or directory
 #include <Preferences.h>_

So I wonder where 'Prefereces' is/should be located?

I think I am almost there (or a mad optimist).

RalphBacon, 2021-08-02:
@tadder 
The "Preferences.h" file is located within the ESP32 framework and allows writing to (pseudo) EEPROM for storing the last used volume, the current station and so forth.

On my Arduino IDE system (and yours too, I expect) it should be found here:
C:\Users\Ralph\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\Preferences\src

If you don't have that file then your 1.0.6 environment has not been correctly set up. Remove it from the Tools>Boards, shut down the Arduino IDE, restart it and reinstall the ESP32 boards again.

If you do have it, that's a different concern as the compiler is not finding it (but should do). Which version of the Arduino IDE are you running, 1.8.15 (the latest at time of writing)? If not, please do the upgrade as that might fix missing paths etc.

As you say, nearly there, but this is more environmental now.

tadder, 2021-08-03:
Hi Ralph,

Significant progress (I’m listening to Majestic Jukebox as I type). You were correct in suggesting that my version of the IDE was not the latest, I have now updated this to 1.8.15 and it now locates Preferences just fine.

The case of the libraries and in particular ESP_VS1053_Libary was interesting. I had read the Arduino documentation as to how the compiler determines which library to use. As you know, it starts with whatever it can find in the libraries folder in the SKETCH folder. I had deduced (incorrectly as it turns out) that libraries held in the same folder as the sketch, being further down the hierarchy would take preference, however it seems that these are ignored.

You advise that any libraries that need to be replaced (e.g. with your older proven versions) should be moved to another location, so that the replacement is picked up by the compiler. I duly renamed the library I didn’t want as ESP_VS1053_LibaryHOLD, thinking this would have the desired effect, but no, this original library was still used.

In desperation I moved the offending file to my desktop, and bingo it found the correct file. I fail to see how logically this is any different from my renaming exercise, but hey ho it works. I’m sure you are nodding your head sagely, (after all you did say) but a lesson learnt for me.

I have now to see if PSRAM is being used. This entails, as you know, monitoring the output with a serial monitor. This is something I have done in the past, but have since forgotten how to do it. It will come back to me, I’m sure, thanks for your support.

RalphBacon, 2021-08-04:
@tadder 
Significant progress indeed!

The reason the "old" library was still found, even though you had renamed the folder, is because the folder name is pretty much irrelevant; the compiler digs down deep and can still find the ".h" file in a subordinate folder. Moving the old library to your desktop was a good idea: firstly, you have the old library intact, and secondly, the compiler does not consider your desktop folder a candidate for libraries so doesn't look there.

To see the serial output, load the sketch (it should have the correct port, as you have previously uploaded the code to it) and click on the magnifying glass. That will load a new window with the serial output (set the speed to 115200 if not already set).

Image shows the details:
[image]

tadder, 2021-08-04:
Hi Ralph,
Thanks for the explanation, helps a lot.
I have no problem with monitoring the Serial output, using either the IDE or Cool Term; my problem is getting some useful data.

If I reset I just get the following 12 lines, nothing more:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
[E][ESP32-WROVER_Web_Radio.ino:46] setup(): Version v1.210701_35 - 2021-07-01 14:37:42.854906

I wonder (pure guess work on my behalf) but could this lack of output be tied up with what we might call the Diagnostic Level? (I’m sure there is a proper name) As specified from line 12 of the project? I seem to remember you were able to monitor operation to quite a low level when conducting a soak test. Maybe this is all tied in together?

Lines of code in ESP-WROVER_Web_Radio is as follows:

// Level is set (see platformio.ini to set this)
/*
	LEVEL_NONE       (0) // Stumm. Nada. On your own.
	LEVEL_ERROR      (1) // Usually fatal
	LEVEL_WARN       (2) // Only when things go wrong
	LEVEL_INFO       (3) // Useful just to see it working
	LEVEL_DEBUG      (4) // Debugging programming
	LEVEL_VERBOSE    (5) // Every message
*/

Thanks as always.

RalphBacon, 2021-08-04:
You are indeed correct; it is the debug level that controls the output.

On the Arduino IDE make sure your parameters for the ESP32 look like this:

[image]

Pay special attention to the **Core Debug Level** which should be set to **Verbose** initially; once it's all running (including the circular buffer in PSRAM) we can start decreasing it.

tadder, 2021-08-04:
What a dumbo I am, I was convinced that debug level implementation would be within the code, so I didn’t look to a solution within the IDE. As you can see, I now have confirmation that PSRAM is implemented and used. Hurrah!

I have returned CIRCULARBUFFERSIZE to 150000, but the capacity indication subsequently showed between 46% and 100% accompanied with some stuttering. Returning this value to 10000 shows a 100% figure and no stuttering. Seems counter-intuitive to me, but at least the solution is obvious?

Am I correct in believing that the ‘NEXT’ function implementation via the break out pins on the PCB is yet to be implemented?

[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
[E][ESP32-WROVER_Web_Radio.ino:46] setup(): Version v1.210701_35 - 2021-07-01 14:37:42.854906
[D][ESP32-WROVER_Web_Radio.ino:53] setup(): This board has PSRAM installed.
[D][ESP32-WROVER_Web_Radio.ino:60] setup(): Total heap: 329600
[D][ESP32-WROVER_Web_Radio.ino:61] setup(): Free heap: 303208
[D][ESP32-WROVER_Web_Radio.ino:62] setup(): Total PSRAM: 4194236
[D][ESP32-WROVER_Web_Radio.ino:63] setup(): Free PSRAM: 4044232
[D][ESP32-WROVER_Web_Radio.ino:64] setup(): Used PSRAM: 150004
[D][ESP32-WROVER_Web_Radio.ino:72] setup(): Starting SPI
[I][ESP32-WROVER_Web_Radio.ino:84] setup(): LITTLEFS system mounted SUCCESSFUL.

Thanks Ralph, now to make an enclosure.

RalphBacon, 2021-08-04:
@tadder Not a dumbo, because had you been using the PlatformIO version that is _exactly_ how I control the debugging level (hence all the comments in the code on that topic); but the Arduino IDE works differently!

You will usually get 100% buffering when using a tiny 10K buffer because the buffer is so small (therefore easy to fill with just a half-second of audio). A larger buffer may _never_ get to 100% on some stations (mine is hovering around 37% right now). But you should definitely not get stuttering with a larger buffer, quite the reverse. 

The only time I get stuttering is if I start sending out too much information on the serial (debugging) port, even at 115200 baud, but even on the Verbose setting here it's been running perfectly for some months now. Whilst a song is playing you don't get much info except the name of the track/artist and how much memory is remaining. 

Let me know the station URL on which you get stuttering with a large buffer (or it all of them?) and I will check it out here.

tadder, 2021-08-05:
So the percentage report indicates the amount of buffer actually used? So the lower the figure the better?
I take back what I said yesterday, there is in fact only the occasional stutter on ‘UK Dance Radio’ and two stations that don’t work at all. Very happy with that.

If you are interested here is a report on all the stations in my list and the amount of buffer reported (at 115k):

Antenne1.de				52%
BBC Radio 2 (new)			67%
BBC Radio 4 (new)			44%
Antenne1 128k			100%
Beatles 128k				100%
Mellow Magic (Redirected)		88%
Greatest Hits 112k (National)	88%
Mowtown Magic Oldies		100%
Mellow Magic (48k AAC)		37%
South East Radio (AAC 64k)		42%
UK Dance Radio			100%	occasional stutter
EZ Hits South FLA			100%
Radio Kehlkopf			connects but no sound
De Natchegaal			does not connects
Greatest Hits (Lincoln)		38%
Heckington Community Radio	100%
Radio Lincs FM			39%
Majestic Jukebox			41%
Classic FM				100%
XRDS.fm Clarksdale Mississi	100%

Thanks Ralph, I shall stop bugging you now, your support has been over and above.

RalphBacon, 2021-08-06:
@tadder Just to clarify, the higher the buffer level, the better. So 100% means that the buffer is stuffed full of data and gives the full 8 seconds of buffering. Lower than that means a correspondingly lower level of buffering but doesn't mean that you will get audio artefacts or stuttering; it all depends on the radio station.

Why are some below 100% you ask (I anticipated)? Basically, I read the streaming data upon connection to a station and mostly they return lots of data _initially_ (for buffering purposes) but then restrict it to a more "real-time" stream. Now, I could wait and continue to buffer but we would end up with a long silence before the buffer was full (Amazon Echo seems to do this and it's very annoying, that initial, long silence).

Anyway, after a few seconds, and as long as the buffer has got a "reasonable" amount (25-30% of buffer) of data in it, we start playing the audio. Some of my stations start at about 37% but over time (several hours) they creep up to that 100% mark.

Your station list percentages seem fine. You have used my "test station" list, which contains some very dodgy ones (poor connectivity, so I could test the buffering) so I'm sure that a commercial radio station will ensure trouble free streaming, in the same way I get it here.

As you're now happy I'll close this thread - but feel free to re-open if you hit any issues (or open a new thread for new issues). Happy listening!

closed 2021-08-06 by RalphBacon

PVsolar, 2021-08-06:
Hello Ralph, tadder,
Very interesting discussion between you and also here I learn a lot. I struggeled with the issue of which library is used in other projects as well. Also I noticed that only a simple name change is not sufficient to hide a library version from the IDE. My solution is that I created a special folder where I place all different library versions with a relation to the program where they were used. So before updating such program I have to look which librries I have to replace in the library folder. Very annoying because I sometimes forget it or miss one. I hoped that the library folder in the folder where the ino file is (like the data folder) was the preferred library but no, I had to copy them to the official library folder. But is this not a good idea to ask the Arduino team to implement? Then you can keep the library variations with the program were they are used and when there is no library in the library folder in the program folder than the IDE looks in the general library.

By the way, as I am Dutch, I saw a typo in a station name in the list: De Natchegaal must be: De Nachtegaal, meaning The Nightingale. 
Hans

tadder, 2021-08-07:
Hi Hans,

I agree completely, if a library located in the same folder as the 
sketch were to take priority over libraries located anywhere else this 
would bring some clarity. But I suspect (complete guesswork on my part) 
that the reason the Arduino team does not implement this is that it is 
not really consistent with how libraries are expected to be used. 
Libraries are about standing on the shoulders of those who have been 
before us and ensuring we do not re-invent the wheel. Any library that 
is not available to other programs (sketches) doesn’t really adhere to 
this base philosophy. As a hobbyist my libraries are used by a limited 
number of programs and it would be no problem to duplicate them when 
required, but presumably this is not the case if you are a developer 
with multiple projects on the go.

One things for certain, if it wasn't for libraries, my projects would 
have to be considerably simpler and hence not so rewarding.

Brian

RalphBacon, 2021-08-08:
This is one reason I switched to, and remain on, PlatformIO.

In PlatformIO, one option is to add libraries for a project and they are cast in concrete, never to change again. A snapshot of what the library was, and will be for ever more. No "surprises" when you fire up the project in 12 months' time and find the library has been updated and you get compilation errors!

The other option is to tell PlatformIO which libraries you want to import (just like the Arduino) _but you can specify the versions you're willing to take_ . So you might say, I'll take all minor versions of library X at version 2 but I don't want a version 3 library because I know that will not be backward compatible.

Worth considering? Start with getting it all installed and get the Blink sketch working. Then follow the (excellent) documentation for PlatformIO on how to do the above. Yes, it's a learning curve and requires some effort but stops all the above issues with libraries mentioned for the Arduino IDE, plus a huge host of programming aids (auto completion, hints, finding the location of library functions... the list goes on).

PVsolar, 2021-08-08:
Hello Ralph, Brian,
I agree with Brian that libraries should be compatible but in practise you can not hold to that without a "library manager" who will check all updates and release a next version. Not for a volunteers platform. So implementing some system of handling various versions in parallel is not a sin in my opinion. As PlatformIO has this possibility it demonstrates that there is a need for that. Ralph, you smashes me left and right to start with IO. I was enthousiast, than you advised against it and now you make me enthousiast again. I do not blame you but you made it clear that IO has advantages but is not something you do in an afternoon. Certainly not for a beginner like me. I follow your advise and start with blink. Then I can better judge to continue or not. Thank you both for your time and help.
Hans