RalphBacon/205-Internet-Radio #20: Future Development

tadder, 2021-08-13:
Hi Ralph,
I can see that you currently have a lot of ‘irons in the fire’ when it comes to subject matter for your channel, so I am wondering what plans (if any) you have for the WebRadio? Thanks to a lot of support from yourself I now have a working radio. I know that you have place holders within the project for possible tone controls and have mooted that maybe you would put some effort into handling http 1.1 and playlists (.pls). I can understand that you may have other priorities and if so would you consider publishing a few ‘notes’ that might allow those of us new to the code to pick up the mantle?

RalphBacon, 2021-08-13:
Yes, many irons and not enough time. Such is life. 🤷‍♂️

Regarding the ESP32 Web Radio, I never realised that so many viewers would run with it. Some, more experienced developers just take the code and get on with it, others require more hand-holding. 

Now, before I develop further features I really, really want to create a new, structured GitHub that is a central repository for all the code (both Arduino IDE and PlatformIO) with examples, hints, tips & traps. This means I can point my existing GitHub repositories that cover the radio to the centralised one, with no duplication or confusion over what is the latest code.

I've fixed the http 1.1 issue now (it only affects the BBC, bless them) but dealing with playlists (.pls files) would be useful to many. Even WinAmp from 10 years ago could do this and is what I use to test out URLs.

My unit plays the radio every day, all day here in my workshop and it "just works" (except when a farmer digs up the nearby broadband line). So bugs are probably few, but I could do a "wish list" of features or usability changes and see what viewers think? Maybe someone would then "pick up the mantle" as you put it, and do that?

Of course, if I create a new GitHub, there's nothing to stop someone forking the code and doing their own thing with the option of new code being merged back into the main branch at some point - in which case I will be a consumer of other people's changes to my own code. Quite bizarre! But that's the Open Source way.

Anything specific you had in mind? Would the above approach work?

tadder, 2021-08-13:
Of course it’s this very ‘hand-holding’ that makes your projects so attractive to those of us who do require this extra support. Long term I’m sure this does enable us to develop our skills. I first discovered your Channel when you introduced me to the MAX7219 (back in 2016 would you believe). I went on to build a few projects based on this and believe I have developed general electronic knowledge since.

Back to the WebRadio, my personal view is that the tone controls may potentially be very useful. Also the ability to store a volume offset against a particular station could enable compensation for the likes of the BBC that consistently broadcast at lower levels (or is that me just being plain lazy?)

Perhaps any developments could be linked to tuition in PlatformIO? A suggestion that makes me slightly nervous, but if this is the future way forward, then those of us still hanging on to the Arduino IDE need to adapt and the more ‘hand-holding’ offered during the transition the better. 

No doubt all achievable if you sacrifice any spare time you might still be hanging on to. ;-)

minolmal, 2021-12-01:
Hi Ralph,
I just finished this project and got it to work. works seamlessly with LOLIN32. I wish to adopt a WROVER and properly make it on a PCB and all. Just had to config for my setup. I've searched for a working Internet Radio project for so long. Thank you for all time and sweat you put on this. I think this project is the most reliable, almost ready-made like. 💯 

I'm working on a DIY Audio setup (just the audio input methods for now.) with FM, Internet Radio, AUX, BLE. Prototype is in the making. Of course, it'll be open source. When I post it publicly, I'll sure mention you sir. :)

I'm need of a method to upload a station link, after compiling. I have few idea I'll try in future. If you'd like this development, I can sort of collaborate. 

Cheers.

RalphBacon, 2021-12-01:
> Hi Ralph, I just finished this project and got it to work 
<snip>
> I'm need of a method to upload a station link, after compiling. I have few ideas I want to try in future. If you'd like this development, I can sort of collaborate.
> 
> Cheers.

Hi Minol, very glad you got it to work. Many others have also reported success. Using a WROVER gives you the Circular Buffer that I do find useful inasmuch that short (<8 seconds) interruptions of the data stream are hidden.

Mine is running 24 x7 here on my workbench in a lovely 3D case given to me by another constructor. I also have my original version indoors so I can listen to the radio in the shower!

If you use PlatformIO (VS Code) then the updates are much easier to do! Are you using the Arduino IDE?

UPDATE: You may find that the VS1053B hangs sometimes on switch on. It requires a small change to the hardware and code. Basically the reset pin of the VS1053B must not be connected to the reset pin of the ESP32 but to another pin, I've used pin 25 in my code. If you don't do this hardware change the software continues to work as normal (but your VS1053b may hang sometimes on startup).

I do have some PCBs available still. One PCB for the main unit, and one PCB for the TFT Touch Screen, connected via an IDC cable (rather than the unreliable DuPont cables). I also use two audio isolating transformers connected to the VS1053b output; that stops any noise/buzzing and conforms to the datasheet for that device which specifies that the COM output pin should NOT be connected to GND! Noise free and works so well. The PCB caters for these tiny (and cheap) transformers. Of course, to use my PCBs the components will have to be exactly the same as mine (I use a TTGO T8 v1.71 ESP32). Perhaps you wish to design your own PCBs?

Updating the station list could easily be accomplished via OTA updates of the SPIFFS/LittleFS partition. I have been very lazy and did not pursue this option. But if you design a solution I would be interested. 

Let me know if you are interested in the PCBs and where you are in the world.

minolmal, 2021-12-01:
>Using a WROVER gives you the Circular Buffer that I do find useful inasmuch that short (<8 seconds) interruptions of the data stream are hidden.

I think WROVER is a MUST for someone who is not just making it for the sake of making it. I've already seen several stations having problems with low buffer space. 

>If you use PlatformIO (VS Code) then the updates are much easier to do! Are you using the Arduino IDE?

I've migrated to PIO almost two years ago. workflow is much more forgiving and less tiring. I still use Arduino IDE for feature testing, as it need very little setting up, but I prefer PIO. 

>reset pin of the VS1053B must not be connected to the reset pin of the ESP32 but to another pin

yeah, I've seen this issue in previous little projects I've done. I used ESP RESET for now, just to get things going. Thanks for the heads-up, I'll keep that in mind when I design my PCB. 👍 

>Perhaps you wish to design your own PCBs?

I plan to add several other components(Rotary encoders, separate MCU for IO handling+drive a display). I will be designing my own PCB. going for 10cmx10cm stack-able design. I think I used some parts of your schematic too. Thanks for the offer. 😃 

>if you design a solution I would be interested.

I'll link you after. Might take some time. I'm doing these after my regular job, sadly. 😄 

Cheers.

RalphBacon, 2021-12-01:
It certainly sounds like you have all planned this out and I wish you luck. 

Yes, doing this sort of thing after your day-job can be tiring, even more so if you have a family but it will keep you occupied through the cold ❄️ winter months! ⛄

If you do get any issues, or have comments or suggestions etc feel free to post here. In the meantime I'll close this as I think we are done!

RalphBacon, 2021-12-01:
Closed #20

closed 2021-12-01 by RalphBacon