RalphBacon/205-Internet-Radio #21: Add EasyEDA Isolating Transformers PCB project to Github repository

rjshiffer, 2021-12-01:
Hi Ralph,

Would you consider adding your EasyEDA project (schematic & PCB) for the PCB for the 2 audio isolating transformers to this repository?

I used your previously provided two EasyEDA projects found in [210-ESP32-Web-Radio-And-More](https://github.com/RalphBacon/210-ESP32-Web-Radio-PCBs-And-More) for building the radio and would like to add the transformers.  It would be nice to have the full set of PCBs available.

Thanks for considering,
Richard

RalphBacon, 2021-12-02:
Hi Richard

 

I hope you are well and looking forward to the festivities!

 

Now, regarding the PCB: I have two options: I already have a new motherboard (i.e. without the screen, which has its own PCB) which already incorporates the audio transformers. They are not electrically connected to anything as the VS1053b is not soldered directly to the motherboard; the output from the 3.5mm socket has to be wired to the PCB pads (I use 3 header pins and floating wires) and the audio output then comes from another set of 3 pins. The audio transformers fit under the VS1053b as there is room if the suggested stand-offs are used. All pretty compact and neat.

 

Alternatively, I have a mini PCB just for the transformers which means you also have to wire them to the motherboard but can mount it anywhere. 

 

I do believe I have spare examples of both boards that I could send you, plus the EasyEDA project, of course. Your call!

 

I haven’t changed the design to route the VS1053b RESET to ESP32 GPIO pin 25 but the code assumes this.

If the hardware change hasn’t been made the code runs as it always did.

Hacking the current PCB (as I did) is pretty easy.

 

 

(very small, looks huge here)

 

Best Regards

Ralph S Bacon

 <https://www.youtube.com/ralphbacon> https://www.youtube.com/ralphbacon

rjshiffer, 2021-12-02:
Hi Ralph,

Your offer of both boards and matching EasyEDA projects sounds great.  I actually have two built radios and I'm now trying to figure out how to create enclosures for them.  (Not my greatest skill set as I've never made any kind of enclosure before.)  With both, I could experiment with different styles.

Would it be possible to have a picture of the new PCB which already incorporates the audio transformers assembled? This could give me clues both how to connect the transformers and also put it in an enclosure.

I am located in the United States, near Chicago.  Will that be a problem? Let me know what the expected postage would cost.

I hope that I'm not asking too much, but this radio is the coolest project I've made!

Thanks for all your efforts,

Richard

RalphBacon, 2021-12-03:
Email me your address (find it [here](https://www.youtube.com/RalphBacon/about)) and I'll get it organised. 

I don't think the postage will be much, about £3 - £4 but I'll let you know when I've actually sent them. As you live in the States I'll put it all into a nice padded envelope that the Post Office delight in charging me extra for ("Large letter"). Yeah, right.

I'll send you pics too.

I'm delighted you think this is a cool project; as I use it every single day (it's running now) I certainly find it one of my most useful projects. More importantly I achieved what I set out to do: to be able to listen to (foreign) streaming radio that other providers insisted on telling me I could not listen to in my part of the world (due to blah blah copyright blah blah more BS). Hah! I showed them!

BTW Have you got those little transformers? I bought a few so I might have some spare too.

Speak soon!

RalphBacon, 2021-12-30:
Leaving open until confirmation that PCBs received.