RalphBacon/205-Internet-Radio #22: Mistake. What's wrong?

samand587, 2022-11-06:
Arduino: 1.8.19 (Windows 10), Board:"ESP32 Wraver Module, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), QIO, 80MHz, 921600, None, Disabled"

C:\Users\Nestor\Documents\Arduino\libraries\LITTLEFS-master\src\LITTLEFS.cpp: In member function 'virtual bool LITTLEFSImpl::exists(const char*)':
C:\Users\Nestor\Documents\Arduino\libraries\LITTLEFS-master\src\LITTLEFS.cpp:44:28: error: no matching function for call to 'LITTLEFSImpl::open(const char*&, const char [2])'
File f = open(path, "r");
^
In file included from C:\Users\Nestor\Documents\Arduino\libraries\LITTLEFS-master\src\LITTLEFS.cpp:17:
C:\Users\Nestor\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5\libraries\FS\src/vfs_api.h:38:17: note: candidate: 'virtual fs::FileImplPtr VFSImpl::open(const char*, const char*, bool)'
FileImplPtr open(const char* path, const char* mode, const bool create) override;
^~~~
C:\Users\Nestor\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5\libraries\FS\src/vfs_api.h:38:17: note: candidate expects 3 arguments, 2 provided
Several libraries found for "Preferences.h"
Is used by: C:\Users\Nestor\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5\libraries\Preferences
Not used: C:\Users\Nestor\Documents\Arduino\libraries\Preferences-2.0.0
Several libraries found for "WiFi.h"
Is used by: C:\Users\Nestor\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5\libraries\WiFi
Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
Compilation error for the ESP32 Wrover Module board.

samand587, 2022-11-07:
I fixed it a little and I get this error. What can I see?

Arduino: 1.8.19 (Windows 10), Board:"ESP32 Wrover Module, No OTA (2MB APP/2MB SPIFFS), QIO, 80MHz, 921600, Verbose, Disabled"

C:\Users\Nestor\Documents\Arduino\libraries\LittleFS_esp32-1.0.6\src\LITTLEFS.cpp: In member function 'virtual bool LITTLEFSImpl::exists(const char*)':
C:\Users\Nestor\Documents\Arduino\libraries\LittleFS_esp32-1.0.6\src\LITTLEFS.cpp:44:28: error: no matching function for call to 'LITTLEFSImpl::open(const char*&, const char [2])'
File f = open(path, "r");
^
In file included from C:\Users\Nestor\Documents\Arduino\libraries\LittleFS_esp32-1.0.6\src\LITTLEFS.cpp:17:
C:\Users\Nestor\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5\libraries\FS\src/vfs_api.h:38:17: note: candidate: 'virtual fs::FileImplPtr VFSImpl::open(const char*, const char*, bool)'
FileImplPtr open(const char* path, const char* mode, const bool create) override;
^~~~
C:\Users\Nestor\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5\libraries\FS\src/vfs_api.h:38:17: note: candidate expects 3 arguments, 2 provided
Several libraries found for "WiFi.h"
Is used by: C:\Users\Nestor\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5\libraries\WiFi
Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
Compilation error for the ESP32 Wrover Module board.

RalphBacon, 2022-11-07:
Things have changed with LittleFS it would seem. I know that Espressif were planning to integrate it into the main ESP framework.

I will have to look at this in more detail but I cannot do this before Tuesday afternoon (8th Nov). Please standby!

samand587, 2022-11-10:
Hello, friend! I managed to fill in the sketch. I can see it in the port monitor. I have nothing but the ESP connected yet. Is this how it should be? And where can I get the connection diagram? Thanks!

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
[E][ESP32-WROVER_Web_Radio.ino:46] setup(): Version v1.210701_35 - 2021-07-01 14:37:42.854906
abort() was called at PC 0x4014fe03 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008d924:0x3ffb1dc0 0x4008db9d:0x3ffb1de0 0x4014fe03:0x3ffb1e00 0x4014fe4a:0x3ffb1e20 0x4014f523:0x3ffb1e40 0x4014f85a:0x3ffb1e60 0x4014f5c1:0x3ffb1e80 0x400e2122:0x3ffb1ea0 0x400d4499:0x3ffb1ec0 0x400e21b6:0x3ffb1fb0 0x4008eba2:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

samand587, 2022-11-10:
I tried to connect VS1053 according to the scheme shown here. Nothing changes, there is no sound, there is a constant reboot. What am I doing wrong? Here is the log

Rebooting...
ets Jun 8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
[E][ESP32-WROVER_Web_Radio.ino:46] setup(): Version v1.210701_35 - 2021-07-01 14:37:42.854906
abort() was called at PC 0x4014fe03 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008d924:0x3ffb1dc0 0x4008db9d:0x3ffb1de0 0x4014fe03:0x3ffb1e00 0x4014fe4a:0x3ffb1e20 0x4014f523:0x3ffb1e40 0x4014f85a:0x3ffb1e60 0x4014f5c1:0x3ffb1e80 0x400e2122:0x3ffb1ea0 0x400d4499:0x3ffb1ec0 0x400e21b6:0x3ffb1fb0 0x4008eba2:0x3ffb1fd0

Rebooting...
ets Jun 8 2016 00:22:57

dhcnet47, 2022-11-10:
Try

https://github.com/RalphBacon/205-Internet-Radio/tree/main/Arduino%20IDE%20Version/Circular%20Buffer

this fixed it for me!

dhcnet47, 2022-11-10:
> The constant reboot indicates that you are specifying a large circular buffer but have not allocated the PSRAM space for it so it is using all your SRAM. Reduce the circular buffer size to 10000 (that's 10K).
> #define CIRCULARBUFFERSIZE 10000
> Do this in BOTH places it is mentioned, replacing the 150K line (for now).

Sorry Ralph we crossed in or replies

Dave

samand587, 2022-11-10:
Dear author!
Please indicate where and in which file you need to make changes. I can't find the line you specified. Thanks!

samand587, 2022-11-10:
Changed in the main.h
#ifdef BOARD_HAS_PARAM file
#define CIRCULAR BUFFER SIZE 150000 <-- change this value to 10000
#else
#define CIRCULAR BUFFER SIZE 10000
#endif

The reboot has stopped, I get this, there is no sound.

ets Jun 8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun 8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
[E][ESP32-WROVER_Web_Radio.ino:46] setup(): Version v1.210701_35 - 2021-07-01 14:37:42.854906
C:\Users\Nestor\Documents\Arduino\libraries\LITTLEFS-master\src\lfs.c:1076:error: Corrupted dir pair at {0x0, 0x1}
E (865) esp_littlefs: mount failed, (-84)
E (869) esp_littlefs: Failed to initialize LittleFS

dhcnet47, 2022-11-10:
Have you created/uploaded the littlefs data?

RalphBacon, 2022-11-10:
You need to understand how to upload a LittleFS partition to your ESP32. 

That is where various files are held (including the station list). See video #203 https://youtu.be/4r6YZlLfKfw 

Luckily, the LittleFS system is now standard (you don't need to do anything to use it as far as I know) but you do need to upload a partition contained in your /data folder.

RalphBacon, 2022-11-10:
@dhcnet47 Don't worry about crossing replies! I'm currently under pressure to edit my next video so any help you can give is welcomed!

samand587, 2022-11-10:
Thank you, author! I'll try!
I did it
Changed in the main.h
#ifdef BOARD_HAS_PARAM file
#define CIRCULAR BUFFER SIZE 150000 <-- change this value to 10000
#else
#define CIRCULAR BUFFER SIZE 10000
#endif

dhcnet47, 2022-11-10:
It is exactly 12 months since I built my webradio!  Thanks for the help you gave me!

Dave

samand587, 2022-11-11:
Hi! Installed LittleFS What to see next? I get this error Thanks!
Arduino: 1.8.19 (Windows 10), Плата:"ESP32 Wrover Module, No OTA (2MB APP/2MB SPIFFS), QIO, 80MHz, 921600, None"

Chip : esp32
Using partition scheme from Arduino IDE.
Start: 0x210000
Size : 0x1f0000

LittleFS Error: mklittlefsnot found!

samand587, 2022-11-11:
Hi! I started the radio, but there are problems. The TOUCH does not work. When turned on, it writes: "Start audio player" and does not go further. The decoder and display are working properly, I checked the wiring. What can I see? And how is the playlist created? Thanks!

RalphBacon, 2022-11-11:
When you compile the code using the Arduino IDE, please set the Core Debugging Level option (Menu > Tools) to 5. This will give more output so I can see what is going on exactly. Then re-post the output which should have entries like@
[I][ESP32-WROVER_Web_Radio.ino:62] setup() LITTLEFS system mounted SUCCESSFUL

Note that the first character of these messages indicate the severity. V = verbose, D = debug, I=information, W = warning (something has gone wrong) and E = fatal error.

samand587, 2022-11-11:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
[E][ESP32-WROVER_Web_Radio.ino:46] setup(): Version v1.210701_35 - 2021-07-01 14:37:42.854906
[D][ESP32-WROVER_Web_Radio.ino:53] setup(): This board has PSRAM installed.
[D][ESP32-WROVER_Web_Radio.ino:60] setup(): Total heap: 329652
[D][ESP32-WROVER_Web_Radio.ino:61] setup(): Free heap: 303176
[D][ESP32-WROVER_Web_Radio.ino:62] setup(): Total PSRAM: 4194236
[D][ESP32-WROVER_Web_Radio.ino:63] setup(): Free PSRAM: 4184232
[D][ESP32-WROVER_Web_Radio.ino:64] setup(): Used PSRAM: 10004
[D][ESP32-WROVER_Web_Radio.ino:72] setup(): Starting SPI
[I][ESP32-WROVER_Web_Radio.ino:84] setup(): LITTLEFS system mounted SUCCESSFUL.
[D][tftHelpers.h:90] initDisplay(): 0179, 
[D][tftHelpers.h:90] initDisplay(): 0DD2, 
[D][tftHelpers.h:90] initDisplay(): 0104, 
[D][tftHelpers.h:90] initDisplay(): 0DD2, 
[D][tftHelpers.h:90] initDisplay(): 0007, 
[I][tftHelpers.h:94] initDisplay(): Touch Calibration completed
[E][esp32-hal-cpu.c:93] addApbChangeCallback(): duplicate func=400E2A78 arg=3FFBDDB4
[I][tftHelpers.h:219] setupDisplayModule(): TFT Initialised
[D][ESP32-WROVER_Web_Radio.ino:92] setup(): Starting Audio Player.
[D][VS1053.cpp:156] begin(): 

[D][VS1053.cpp:157] begin(): Reset VS1053...

[D][VS1053.cpp:161] begin(): End reset VS1053...

[D][VS1053.cpp:119] testComm(): VS1053 not properly installed!

RalphBacon, 2022-11-11:
So now you have your error. Correct the wiring for the VS1053 and it will progress beyond this stage.

samand587, 2022-11-12:
Hi! I checked the audio board on another project KaRadio 32. Everything works fine! I don't know why he doesn't want to run in your project. Can you tell me? Thanks!

RalphBacon, 2022-11-12:
I've looked at which pins are connected to the VS1053 from the ESP32. Can you confirm you have these:

VS1053_CS       32
VS1053_DCS    33
VS1053_DREQ  35

The VS1053 used to occasionally stop at this point so I updated the code in Visual Studio (but not Arduino) to specifically reset the VS1053 using pin D25.

// Separate reset for the VS1053
#define VS1053_RST 25

I then connected the VS1053 to pin 25 on the ESP32 and executed this code around line 88 of main.cpp, just before starting the Audio Player command: player.begin():

```
    // VS1053 MP3 decoder. If we don't get a response continue anyway.
    log_d("Starting Audio Player.");
    displayTrackArtist("Starting Audio Player.");

    pinMode(VS1053_RST, OUTPUT);
    digitalWrite(VS1053_RST, LOW);
    delay(100);
    digitalWrite(VS1053_RST, HIGH);
```

If you want to try this to see whether it fixes your issue let me know how you get on. But please do double check that wiring again!

samand587, 2022-11-12:
I checked the wiring, everything matches. I completely updated the wiring - nothing has changed! I am ready for any action, but I have little experience and need a hint. Thanks!

samand587, 2022-11-13:
Hi! When I insert
pinMode(VS1053_RST, OUTPUT) into the code;
digitalWrite(VS1053_RST, LOW);
delay(100);
digitalWrite(VS1053_RST, HIGH);
then I get this error
Arduino: 1.8.19 (Windows 10), Board:"ECP32 Rover Module, No OTO (2 MB APP/2MB SPIFFS), QIO, 80 MHz, 921600, None"

C:\Users\Nestor\Documents\Arduino\ESP32-WROVER_Web_Radio\ESP32-WROVER_Web_Radio.ino: In function 'void setup()':
ESP32-WROVER_Web_Radio:94:11: error: 'VS1053_RST' was not declared in this scope
pinMode(VS1053_RST, OUTPUT);
^
Several libraries found for "WiFi.h"
Used by: C:\Users\Nestor\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi
Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
'VS1053_RST' was not declared in this scope

RalphBacon, 2022-11-13:
Before the extra code you have added, insert this line:

int VS1053_RST = 25;

Recompile, upload and don't forget to connect the RESET of the VS1053 to pin 25 of y our ESP32.

samand587, 2022-11-13:
Great! The player began to load, I go to the main screen, but there is no connection to the station, there is no sound, the TOUCH does not work. The monitor writes this

[D][ESP32-WROVER_Web_Radio.ino:601] stationConnect(): -----------------------------------
[D][ESP32-WROVER_Web_Radio.ino:602] stationConnect():      Connecting to station 0
[D][ESP32-WROVER_Web_Radio.ino:603] stationConnect(): -----------------------------------
[D][ESP32-WROVER_Web_Radio.ino:604] stationConnect(): HOST: stream.antenne1.de
[D][ESP32-WROVER_Web_Radio.ino:605] stationConnect(): PATH: /a1stg/livestream1.aac
[D][ESP32-WROVER_Web_Radio.ino:606] stationConnect(): PORT: 80
[D][ESP32-WROVER_Web_Radio.ino:607] stationConnect(): NAME: Antenne1.de
[D][ESP32-WROVER_Web_Radio.ino:608] stationConnect(): DATA: Yes
[I][ESP32-WROVER_Web_Radio.ino:638] stationConnect(): Free memory: 238668 bytes
[D][ESP32-WROVER_Web_Radio.ino:673] stationConnect(): Host: stream.antenne1.de Port:80
[D][ESP32-WROVER_Web_Radio.ino:693] stationConnect(): Connected to stream.antenne1.de (Antenne1.de)
[I][ESP32-WROVER_Web_Radio.ino:703] stationConnect(): Getting data from /a1stg/livestream1.aac (WITHOUT Metadata)
[D][ESP32-WROVER_Web_Radio.ino:712] stationConnect(): Waiting for header data
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: HTTP/1.0 200 OK
[D][ESP32-WROVER_Web_Radio.ino:743] stationConnect(): 200 - OK response
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: icy-br: 64
[D][ESP32-WROVER_Web_Radio.ino:785] stationConnect(): Bit rate:64
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: icy-pub: 1
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: icy-description: Region Stuttgart
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: icy-audio-info: ice-samplerate=44100;ice-bitrate=64;ice-channels=2
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: icy-url: http://www.antenne1.de
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: Instance-id: d6d68036ca9c006af472aaba10b854ea
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: Cache-Control: no-cache
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: Server: Antenne 1 Streaming 8.4.3
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: icy-genre: antenne 1 Stuttgart
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: Expires: Mon, 26 Jul 1997 05:00:00 GMT
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: icy-metaint: 0
[D][ESP32-WROVER_Web_Radio.ino:769] stationConnect(): NEW Metadata Interval:0
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: Pragma: no-cache
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: icy-name: antenne 1 Stuttgart
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: Connection: close
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: Content-Type: audio/aacp
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: Set-Cookie: AISSessionId=634f369cbb1bcec0_2190739_4T3oxmDA__0000000h2Ef; Path=/; Domain=stream.antenne1.de; Max-Age=6000; Expires=Sun, 13 Nov 2022 09:41:10 GMT
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: 
[D][ESP32-WROVER_Web_Radio.ino:405] readHttpStream(): Low buffer: 0%, occurence 1
[D][ESP32-WROVER_Web_Radio.ino:405] readHttpStream(): Low buffer: 0%, occurence 2
[D][ESP32-WROVER_Web_Radio.ino:405] readHttpStream(): Low buffer: 0%, occurence 3
[D][ESP32-WROVER_Web_Radio.ino:405] readHttpStream(): Low buffer: 0%, occurence 4
[D][ESP32-WROVER_Web_Radio.ino:405] readHttpStream(): Low buffer: 0%, occurence 5
[D][ESP32-WROVER_Web_Radio.ino:405] readHttpStream(): Low buffer: 0%, occurence 6

RalphBacon, 2022-11-13:
OK, this line 
```
readHttpStream(): Low buffer: 0%, 
```
means that it cannot read the stream data. Try pressing RESET on the ESP32 to see if that changes anything.

Or perhaps the stream is blocked by your ISP. Which country do you live in? I can then find a suitable (local) station to try it on.

Also a (small) picture of the front screen would be useful.

samand587, 2022-11-13:
My country is Russia. I tried to change the playlist- I'm probably not doing it right, but it's playing. I attach the reboot log.

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
[E][ESP32-WROVER_Web_Radio.ino:46] setup(): Version v1.210701_35 - 2021-07-01 14:37:42.854906
[D][ESP32-WROVER_Web_Radio.ino:53] setup(): This board has PSRAM installed.
[D][ESP32-WROVER_Web_Radio.ino:60] setup(): Total heap: 329668
[D][ESP32-WROVER_Web_Radio.ino:61] setup(): Free heap: 303320
[D][ESP32-WROVER_Web_Radio.ino:62] setup(): Total PSRAM: 4194236
[D][ESP32-WROVER_Web_Radio.ino:63] setup(): Free PSRAM: 4184232
[D][ESP32-WROVER_Web_Radio.ino:64] setup(): Used PSRAM: 10004
[D][ESP32-WROVER_Web_Radio.ino:72] setup(): Starting SPI
[I][ESP32-WROVER_Web_Radio.ino:84] setup(): LITTLEFS system mounted SUCCESSFUL.
[D][tftHelpers.h:90] initDisplay(): 0179, 
[D][tftHelpers.h:90] initDisplay(): 0DD2, 
[D][tftHelpers.h:90] initDisplay(): 0104, 
[D][tftHelpers.h:90] initDisplay(): 0DD2, 
[D][tftHelpers.h:90] initDisplay(): 0007, 
[I][tftHelpers.h:94] initDisplay(): Touch Calibration completed
[E][esp32-hal-cpu.c:93] addApbChangeCallback(): duplicate func=400E2AA0 arg=3FFBDDB4
[I][tftHelpers.h:219] setupDisplayModule(): TFT Initialised
[D][ESP32-WROVER_Web_Radio.ino:92] setup(): Starting Audio Player.
[D][VS1053.cpp:156] begin(): 

[D][VS1053.cpp:157] begin(): Reset VS1053...

[D][VS1053.cpp:161] begin(): End reset VS1053...

[D][VS1053.cpp:132] testComm(): Slow SPI,Testing VS1053 read/write registers...

[D][VS1053.cpp:132] testComm(): Fast SPI, Testing VS1053 read/write registers again...

[D][VS1053.cpp:183] begin(): endFillByte is 0

[D][ESP32-WROVER_Web_Radio.ino:108] setup(): Switch player to MP3 mode
[D][VS1053.cpp:307] switchToMp3Mode(): Switched to mp3 mode

[D][VS1053.cpp:273] softReset(): Performing soft-reset

[D][ESP32-WROVER_Web_Radio.ino:122] setup(): Treble boost expected to be: 82, is: 82
[D][ESP32-WROVER_Web_Radio.ino:128] setup(): Bass boost expected to be: 138, is: 138
[I][stationSelectHelper.h:467] populateStationList(): Radio stations imported successfully.
[D][stationSelectHelper.h:478] populateStationList(): 0 - stream.antenne1.de /a1stg/livestream1.aac 80 (Antenne1.de) YES
[D][stationSelectHelper.h:478] populateStationList(): 1 - stream.live.vc.bbcmedia.co.uk /bbc_radio_two 80 (BBC Radio 2 (new)) NO
[D][stationSelectHelper.h:478] populateStationList(): 2 - stream.live.vc.bbcmedia.co.uk /bbc_radio_fourfm 80 (BBC Radio 4 (new)) NO
[D][stationSelectHelper.h:478] populateStationList(): 3 - stream.antenne1.de /a1stg/livestream2.mp3 80 (Antenne1 128k) YES
[D][stationSelectHelper.h:478] populateStationList(): 4 - listen.181fm.com /181-beatles_128k.mp3 80 (Beatles 128k) YES
[D][stationSelectHelper.h:478] populateStationList(): 5 - stream-mz.planetradio.co.uk /magicmellow.mp3 80 (Mellow Magic (Redirected)) YES
[D][stationSelectHelper.h:478] populateStationList(): 6 - edge-bauermz-03-gos2.sharp-stream.com /net2national.mp3 80 (Greatest Hits 112k (National)) YES
[D][stationSelectHelper.h:478] populateStationList(): 7 - airspectrum.cdnstream1.com /1302_192 8024 (Mowtown Magic Oldies) YES
[D][stationSelectHelper.h:478] populateStationList(): 8 - live-bauer-mz.sharp-stream.com /magicmellow.aac 80 (Mellow Magic (48k AAC)) YES
[D][stationSelectHelper.h:478] populateStationList(): 9 - stream.btsstream.com /seaac 8000 (South East Radio (AAC 64k)) YES
[D][stationSelectHelper.h:478] populateStationList(): 10 - uk2.internet-radio.com / 8024 (UK Dance Radio) YES
[D][stationSelectHelper.h:478] populateStationList(): 11 - us4.internet-radio.com /stream?type=http&nocache=72 8197 (EZ Hits South FLA) NO
[D][stationSelectHelper.h:478] populateStationList(): 12 - s1.slotex.pl / 7274 (Radio Kehlkopf) NO
[D][stationSelectHelper.h:478] populateStationList(): 13 - 204.187.100.18 / 1485 (De Natchegaal) YES
[D][stationSelectHelper.h:478] populateStationList(): 14 - live-bauer-mz.sharp-stream.com /net2lincoln.aac 80 (Greatest Hits (Lincoln)) YES
[D][stationSelectHelper.h:478] populateStationList(): 15 - ais-edge09-live365-dal02.cdnstream.com /a52107?listenerId=esAdblock0650048 80 (Heckington Community Radio) YES
[D][stationSelectHelper.h:478] populateStationList(): 16 - live-bauer-mz.sharp-stream.com /net1lincoln.aac 80 (Radio Lincs FM) YES
[D][stationSelectHelper.h:478] populateStationList(): 17 - uk3.internet-radio.com /live 8405 (Majestic Jukebox) YES
[D][stationSelectHelper.h:478] populateStationList(): 18 - media-ice.musicradio.com /ClassicFMMP3 80 (Classic FM) YES
[... 37 lines cut ...]
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: HTTP/1.0 200 OK
[D][ESP32-WROVER_Web_Radio.ino:743] stationConnect(): 200 - OK response
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: icy-br: 64
[D][ESP32-WROVER_Web_Radio.ino:785] stationConnect(): Bit rate:64
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: icy-pub: 1
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: icy-description: Region Stuttgart
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: icy-audio-info: ice-samplerate=44100;ice-bitrate=64;ice-channels=2
[... cut ...]
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: Set-Cookie: AISSessionId=634f369cbb1bcec0_2199692_rpgoQX4C__0000000h9mT; Path=/; Domain=stream.antenne1.de; Max-Age=6000; Expires=Sun, 13 Nov 2022 12:10:23 GMT
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: 
[D][ESP32-WROVER_Web_Radio.ino:238] setup(): Restored screen brightness to 255
[D][ESP32-WROVER_Web_Radio.ino:244] setup(): Free memory: 240288 bytes
[D][ESP32-WROVER_Web_Radio.ino:405] readHttpStream(): Low buffer: 0%, occurence 1
[D][ESP32-WROVER_Web_Radio.ino:428] readHttpStream(): Low buffer: recovered to 36%

samand587, 2022-11-13:
[image]

RalphBacon, 2022-11-13:
The image confirms that the stream is being read and buffered at 100%. You should hear the station. Can you?

To change the playlist you need to edit the file "stationList.txt" in the "/data" folder on your PC (the folder you previously uploaded to the ESP32). Then re-upload just that partition and the stations will be changed. I suggest you take a backup of that file before you edit it!

samand587, 2022-11-13:
My station is singing. But I can't go into the settings and try other stations - the TOUCH doesn't work for me. What can I check in the wheelbarrow? And I can't understand how and by what principle the playlist is edited. How do I edit a playlist?

RalphBacon, 2022-11-13:
### Editing the stations list
You edit the playlist of station by editing the "stationList.txt". This file is on your PC. It will be in the /data folder of your project. the format of the file is:
```
host-url, port, stream, friendly name, metadata flag 1=yes, 0=no
"stream.antenne1.de",80,"/a1stg/livestream2.mp3","Antenne1 128k",1
```
You will have to search the Internet to see whether you can find the station(s) you want to listen to and use this as a pattern to add (or replace) stations in that text file.

### Touch Screen
There are two versions of the screen: with and without touch. Are your **sure** you have the touch version? Did it come with a small white stylus?

samand587, 2022-11-13:
Yes, I ordered with a TOUCH and there was a stylus. And all the necessary components are installed on the display.

samand587, 2022-11-13:
I've sorted out the playlist, my stations are playing! Thanks!

It remains to deal with the TOUCH

samand587, 2022-11-14:
Hi! The radio worked successfully all evening, but very loudly. I decided to turn down the sound in the sketch. In the line
I put it like this: userVolume = preferences.getint("Volume", 75); and the radio stopped connecting to the station. There is no buffering. Returning the sketch to its original form userVolume = preferences.getint("Volume", 95); gives nothing. The radio is not working again. What could be the reason?

samand587, 2022-11-14:
Hi! Why on the display after connecting to the station and buffering 100, the entry on the display does not disappear:"Connected. Reading stream..." ?

RalphBacon, 2022-11-14:
That is because there is no meta data to display.

Either you have indicated you do not want the meta data (last digit on the stationList.txt entry =0) or the sender has indicated that no meta data is available. For the station you are showing there certainly is meta data so either you have edited the entry or perhaps have connected pin 36 to GNND (permanent "no meta data required"). Remove that connection if you have it.

Looking at the debug log you previously provided:
```
[I][ESP32-WROVER_Web_Radio.ino:703] stationConnect(): Getting data from /a1stg/livestream1.aac (WITHOUT Metadata)
```
This indicates that you have told the device you do not want meta data so you can correct this by following the instructions above.

samand587, 2022-11-14:
Here is the station playing "air spectrum.cdn stream 1.com",8024,"/1302_192"," Motown Magic Oldies",1
As seen on the Connected screen. Reading stream...
GPIO 36 is not connected anywhere. 
[image]

RalphBacon, 2022-11-14:
GPIO36 must be kept HIGH, via a 10K resistor to VCC. Try that and show the debug messages to me again please.
Also show me the entry in the stationList.txt for this station.

samand587, 2022-11-14:
"airspectrum.cdnstream1.com",8024,"/1302_192","Mowtown Magic Oldies",1
"online-1.gkvr.ru",8000,"/europa_eka_64.aac","Evropa PLUS","0"
"aacradio.stream.laut.fm",8080,"/aacradio","laut.fm", "0"
"online.sputnik107.ru",8000,"/sputnik107_aacplus","Radio Sputnik FM","0"
"online-1.gkvr.ru",8000,"/mc_eka_64.aac","Radio Monte Carlo ","0"
"pub0302.101.ru",8000,"/stream/air/aac/64/102","Radio Humor FM","0"
"pub0302.101.ru",8000,"/stream/air/aac/64/101","Radio Romantika","0"
"listen6.myradio24.com",9000,"/vitfmdance","Vitamin FM","0"
"choco.hostingradio.ru",10010,"/fm","Radio Chocolate","0"
"nashe1.hostingradio.ru",80,"/nashe-128.mp3","Radio Nache","0"
"online1.gkvr.ru",8000,"/radiola_eka_128.mp3","Radio Radiola","0"
"stream1.radiord.ru",8000,"/live96.aac","Radio RD ","0"
stream.live.vc.bbcmedia.co.uk,80,/bbc_radio_two,BBC Radio 2 (new),0
stream.live.vc.bbcmedia.co.uk,80,/bbc_radio_fourfm,BBC Radio 4 (new),0
"stream.antenne1.de",80,"/a1stg/livestream2.mp3","Antenne1 128k",1
"listen.181fm.com",80,"/181-beatles_128k.mp3","Beatles 128k",1
"stream-mz.planetradio.co.uk",80,"/magicmellow.mp3","Mellow Magic (Redirected)",1
"airspectrum.cdnstream1.com",8024,"/1302_192","Mowtown Magic Oldies",1
"stream.antenne1.de",80,"/a1stg/livestream1.aac","Antenne1.de","1"

samand587, 2022-11-14:
GPIO36 + 10 kOm +5 V  It helped! How can I deal with the TOUCH?

RalphBacon, 2022-11-14:
[image]
The T_IRQ and T_CS pins must be wired to GPIO 15 and 12 respectively.

samand587, 2022-11-14:
Connected correctly- checked

RalphBacon, 2022-11-14:
I would need a clean debug log. 

Start a new log session, reset the ESP32 device and attach the log file here. 

Also, I have seen it where the 'touch' screen is upside down, or even portrait mode - try tapping with the stylus over the screen from top to bottom to see if _anything_ responds.

samand587, 2022-11-14:
It does not respond to the wiring with a stylus. A new magazine here:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
[E][ESP32-WROVER_Web_Radio.ino:46] setup(): Version v1.210701_35 - 2021-07-01 14:37:42.854906
[D][ESP32-WROVER_Web_Radio.ino:55] setup(): This board has no PSRAM installed.
[D][ESP32-WROVER_Web_Radio.ino:60] setup(): Total heap: 352104
[D][ESP32-WROVER_Web_Radio.ino:61] setup(): Free heap: 315616
[D][ESP32-WROVER_Web_Radio.ino:62] setup(): Total PSRAM: 0
[D][ESP32-WROVER_Web_Radio.ino:63] setup(): Free PSRAM: 0
[D][ESP32-WROVER_Web_Radio.ino:64] setup(): Used PSRAM: 0
[D][ESP32-WROVER_Web_Radio.ino:72] setup(): Starting SPI
[I][ESP32-WROVER_Web_Radio.ino:84] setup(): LITTLEFS system mounted SUCCESSFUL.
[D][tftHelpers.h:90] initDisplay(): 0179, 
[D][tftHelpers.h:90] initDisplay(): 0DD2, 
[D][tftHelpers.h:90] initDisplay(): 0104, 
[D][tftHelpers.h:90] initDisplay(): 0DD2, 
[D][tftHelpers.h:90] initDisplay(): 0007, 
[I][tftHelpers.h:94] initDisplay(): Touch Calibration completed
[E][esp32-hal-cpu.c:93] addApbChangeCallback(): duplicate func=400E1DD4 arg=3FFBDDB4
[I][tftHelpers.h:219] setupDisplayModule(): TFT Initialised
[D][ESP32-WROVER_Web_Radio.ino:92] setup(): Starting Audio Player.
[D][VS1053.cpp:156] begin(): 

[D][VS1053.cpp:157] begin(): Reset VS1053...

[D][VS1053.cpp:161] begin(): End reset VS1053...

[D][VS1053.cpp:132] testComm(): Slow SPI,Testing VS1053 read/write registers...

[D][VS1053.cpp:132] testComm(): Fast SPI, Testing VS1053 read/write registers again...

[D][VS1053.cpp:183] begin(): endFillByte is 0

[D][ESP32-WROVER_Web_Radio.ino:108] setup(): Switch player to MP3 mode
[D][VS1053.cpp:307] switchToMp3Mode(): Switched to mp3 mode

[D][VS1053.cpp:273] softReset(): Performing soft-reset

[D][ESP32-WROVER_Web_Radio.ino:122] setup(): Treble boost expected to be: 82, is: 82
[D][ESP32-WROVER_Web_Radio.ino:128] setup(): Bass boost expected to be: 138, is: 138
[I][stationSelectHelper.h:467] populateStationList(): Radio stations imported successfully.
[D][stationSelectHelper.h:478] populateStationList(): 0 - online.sputnik107.ru /sputnik107_aacplus 8000 (Radio Sputnik FM) NO
[D][stationSelectHelper.h:478] populateStationList(): 1 - online-1.gkvr.ru /hitfm_eka_64.aac 8000 (HIT FM ) YES
[D][stationSelectHelper.h:478] populateStationList(): 2 - online-1.gkvr.ru /rusradio_eka_64.aac 8000 (Russkoe Radio ) NO
[D][stationSelectHelper.h:478] populateStationList(): 3 - online-1.gkvr.ru /record_eka_64.aac 8000 (Radio Record) NO
[D][stationSelectHelper.h:478] populateStationList(): 4 - online-1.gkvr.ru /europa_eka_64.aac 8000 (Evropa PLUS) NO
[D][stationSelectHelper.h:478] populateStationList(): 5 - aacradio.stream.laut.fm /aacradio 8080 (laut.fm) NO
[D][stationSelectHelper.h:478] populateStationList(): 6 - online-1.gkvr.ru /mc_eka_64.aac 8000 (Radio Monte Carlo ) NO
[D][stationSelectHelper.h:478] populateStationList(): 7 - pub0302.101.ru /stream/air/aac/64/102 8000 (Radio Humor FM) NO
[D][stationSelectHelper.h:478] populateStationList(): 8 - pub0302.101.ru /stream/air/aac/64/101 8000 (Radio Romantika) NO
[D][stationSelectHelper.h:478] populateStationList(): 9 - listen6.myradio24.com /vitfmdance 9000 (Vitamin FM) NO
[D][stationSelectHelper.h:478] populateStationList(): 10 - choco.hostingradio.ru /fm 10010 (Radio Chocolate) NO
[D][stationSelectHelper.h:478] populateStationList(): 11 - nashe1.hostingradio.ru /nashe-128.mp3 80 (Radio Nache) NO
[D][stationSelectHelper.h:478] populateStationList(): 12 - online1.gkvr.ru /radiola_eka_128.mp3 8000 (Radio Radiola) NO
[D][stationSelectHelper.h:478] populateStationList(): 13 - stream1.radiord.ru /live96.aac 8000 (Radio RD ) NO
[D][stationSelectHelper.h:478] populateStationList(): 14 - stream.live.vc.bbcmedia.co.uk /bbc_radio_two 80 (BBC Radio 2 (new)) NO
[D][stationSelectHelper.h:478] populateStationList(): 15 - stream.live.vc.bbcmedia.co.uk /bbc_radio_fourfm 80 (BBC Radio 4 (new)) NO
[D][stationSelectHelper.h:478] populateStationList(): 16 - stream.antenne1.de /a1stg/livestream2.mp3 80 (Antenne1 128k) YES
[D][stationSelectHelper.h:478] populateStationList(): 17 - listen.181fm.com /181-beatles_128k.mp3 80 (Beatles 128k) YES
[D][stationSelectHelper.h:478] populateStationList(): 18 - stream-mz.planetradio.co.uk /magicmellow.mp3 80 (Mellow Magic (Redirected)) YES
[... 41 lines cut ...]
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: HTTP/1.1 200 OK
[D][ESP32-WROVER_Web_Radio.ino:743] stationConnect(): 200 - OK response
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: Server: nginx
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: Date: Mon, 14 Nov 2022 13:20:49 GMT
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: Content-Type: audio/aacp
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: Connection: close
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: icy-br: 64
[D][ESP32-WROVER_Web_Radio.ino:785] stationConnect(): Bit rate:64
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: ice-audio-info: ice-samplerate=44100;ice-bitrate=64;ice-channels=2
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: icy-br: 64
[... cut ...]
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: Expires: Mon, 26 Jul 1997 05:00:00 GMT
[D][ESP32-WROVER_Web_Radio.ino:738] stationConnect(): HEADER: 
[D][ESP32-WROVER_Web_Radio.ino:238] setup(): Restored screen brightness to 255
[D][ESP32-WROVER_Web_Radio.ino:244] setup(): Free memory: 245800 bytes

samand587, 2022-11-14:
The log does not stop

samand587, 2022-11-14:
Starting...
x: 1116     y: 2236     z: 4099 
x: 1584     y: 2748     z: 4100 
x: 1583     y: 2430     z: 4103 
x: 1583     y: 1536     z: 4098 
x: 1568     y: 1784     z: 4097 
x: 1648     y: 1553     z: 3906 
x: 1599     y: 1727     z: 4042 
x: 1615     y: 2111     z: 4096 
x: 1679     y: 1607     z: 3936 
x: 1671     y: 2447     z: 4097 
x: 1632     y: 2768     z: 4099 
x: 1520     y: 2768     z: 4099 
x: 1567     y: 2494     z: 4101 
x: 1665     y: 2752     z: 4099 
x: 1616     y: 2368     z: 4095 
x: 1628     y: 2719     z: 4101 
x: 1648     y: 1571     z: 3939 
x: 1648     y: 2681     z: 4099 
x: 1599     y: 1614     z: 3911 
x: 1695     y: 1614     z: 3913 
x: 1665     y: 1927     z: 4103 
x: 1728     y: 2784     z: 4099 
x: 1635     y: 2656     z: 4100 
x: 1671     y: 2691     z: 4099 
x: 1632     y: 1440     z: 4097 
x: 1552     y: 2736     z: 4100 
x: 1599     y: 2687     z: 4099 
x: 1616     y: 2640     z: 4101 
x: 1596     y: 1551     z: 4099 
x: 1647     y: 1624     z: 3914 
x: 1647     y: 1615     z: 3899 
x: 1743     y: 2431     z: 4099 
x: 1504     y: 2672     z: 4097 
x: 1635     y: 2656     z: 4101 
x: 1632     y: 1712     z: 4096 
x: 1521     y: 2744     z: 4100 
x: 1680     y: 1632     z: 3935 
x: 1647     y: 1455     z: 4099 
x: 1615     y: 2752     z: 4099 
x: 1631     y: 2703     z: 4098 
x: 1592     y: 1888     z: 4096 
x: 1667     y: 2656     z: 4100 
x: 1615     y: 2460     z: 4102 
x: 1656     y: 1968     z: 4087 
x: 1567     y: 2703     z: 4095 
x: 1631     y: 2655     z: 4107 
x: 1541     y: 1488     z: 4097 
x: 1696     y: 1372     z: 4097 
x: 1568     y: 1744     z: 4097 
x: 1647     y: 2371     z: 4095 
x: 1591     y: 1727     z: 3947 
x: 1727     y: 2695     z: 4100 
x: 1600     y: 1504     z: 4097 
x: 1631     y: 1600     z: 4097 
x: 1632     y: 1984     z: 4097 
x: 1664     y: 1568     z: 4099 
x: 1599     y: 2703     z: 4099 
x: 1591     y: 1568     z: 4097 
x: 1664     y: 1376     z: 4097 
x: 1648     y: 1407     z: 4098 
x: 1599     y: 1423     z: 4097 
x: 1528     y: 2608     z: 4101 
x: 1599     y: 2703     z: 4100 
x: 1671     y: 2159     z: 4096 
x: 1567     y: 1471     z: 4097 
x: 1664     y: 1952     z: 4105 
x: 1664     y: 1711     z: 3912 
x: 1712     y: 2748     z: 4100 
x: 1664     y: 1392     z: 4097 
x: 1635     y: 1568     z: 4097 
x: 1583     y: 2687     z: 4095 
x: 1596     y: 1551     z: 4099 
x: 1617     y: 1799     z: 3921 
x: 1671     y: 1609     z: 4098 
x: 1648     y: 1920     z: 4106 
x: 1695     y: 1607     z: 3935 
x: 1711     y: 1472     z: 4097 
x: 1583     y: 2492     z: 4100 
x: 1583     y: 2751     z: 4101 
x: 1520     y: 2800     z: 4099 
x: 1664     y: 1799     z: 3921 
x: 1656     y: 1548     z: 4098 
x: 1619     y: 2695     z: 4099 
x: 1584     y: 1776     z: 4097

RalphBacon, 2022-11-14:
But did the values change as you moved the stylus on the screen? Or did the values change all by themselves without you touching the screen?

samand587, 2022-11-14:
Here I moved the stylus on the screen
Starting...
x: 1107     y: 2248     z: 4100 
x: 1537     y: 2736     z: 4099 
x: 1671     y: 1743     z: 3889 
x: 1599     y: 2671     z: 4097 
x: 1632     y: 1520     z: 4097 
x: 1671     y: 1575     z: 4099 
x: 1655     y: 1743     z: 3895 
x: 1679     y: 2735     z: 4099 
x: 1615     y: 1599     z: 4098 
x: 1680     y: 1872     z: 4044 
x: 1664     y: 1504     z: 4097 
x: 1568     y: 2718     z: 4101 
x: 1598     y: 2684     z: 4095 
x: 1560     y: 2703     z: 4100 
x: 1603     y: 1488     z: 4098 
x: 1592     y: 11     z: 4095 
x: 2529     y: 0     z: 4095 
x: 1537     y: 3     z: 4095 
x: 2563     y: 4     z: 4095 
x: 1567     y: 1     z: 4095 
x: 1616     y: 232     z: 4095 
x: 2159     y: 2383     z: 4104 
x: 2119     y: 1695     z: 3913 
x: 2072     y: 2943     z: 4114 
x: 1152     y: 8     z: 4094 
x: 2624     y: 13     z: 4095 
x: 1455     y: 27     z: 4116 
x: 2689     y: 1     z: 4078 
x: 2623     y: 22     z: 4109 
x: 2684     y: 58     z: 4171 
x: 2624     y: 46     z: 3984 
x: 2784     y: 0     z: 4095 
x: 2624     y: 21     z: 4095 
x: 1584     y: 0     z: 4095 
x: 2559     y: 1     z: 4095 
x: 1976     y: 1135     z: 4099 
x: 2000     y: 2684     z: 4097 
x: 2051     y: 2784     z: 4099 
x: 2032     y: 1408     z: 3923 
x: 1976     y: 2831     z: 4099 
x: 2063     y: 1631     z: 3918 
x: 2112     y: 1667     z: 3926 
x: 2016     y: 2112     z: 4117 
x: 2079     y: 3007     z: 4095 
x: 2056     y: 0     z: 4095 
x: 1744     y: 11     z: 4095 
x: 2111     y: 0     z: 4095 
x: 2492     y: 12     z: 4095 
x: 2511     y: 1     z: 4095 
x: 2656     y: 1     z: 4091 
x: 1676     y: 2776     z: 4125 
x: 2079     y: 1327     z: 3875 
x: 1991     y: 2656     z: 4100 
x: 2014     y: 0     z: 4095 
x: 2480     y: 0     z: 4095 
x: 1512     y: 26     z: 4095 
x: 1569     y: 2608     z: 4100 
x: 1679     y: 1607     z: 4097 
x: 1615     y: 1440     z: 4097 
x: 1599     y: 0     z: 4095 
x: 1564     y: 0     z: 4095 
x: 2527     y: 4     z: 4057 
x: 2033     y: 2800     z: 4103 
x: 1824     y: 0     z: 4095 
x: 1488     y: 0     z: 4039 
x: 2575     y: 0     z: 4095 
x: 1696     y: 1555     z: 3938 
x: 1662     y: 2492     z: 4103 
x: 1550     y: 2478     z: 4103 
x: 1667     y: 1927     z: 4105 
x: 1452     y: 0     z: 4095 
x: 2527     y: 10     z: 4099 
x: 2623     y: 20     z: 4111 
x: 2748     y: 8     z: 4118 
x: 2627     y: 0     z: 4095 
x: 2863     y: 10     z: 4084 
x: 2703     y: 3     z: 4031 
x: 2111     y: 1503     z: 4099 
x: 2000     y: 2943     z: 4103 
x: 2095     y: 2527     z: 4104 
x: 2050     y: 1631     z: 3958 
x: 2127     y: 2051     z: 4095 
x: 2096     y: 1640     z: 4097 
x: 1991     y: 2544     z: 4097 
x: 1502     y: 0     z: 4095 
x: 2416     y: 85     z: 4152 
x: 2591     y: 6     z: 4054 
x: 2735     y: 3     z: 4095 
x: 2672     y: 0     z: 4022 
x: 2831     y: 0     z: 4095 
x: 2624     y: 3     z: 4095 
x: 2816     y: 5     z: 4095 
x: 2616     y: 0     z: 4095 
x: 1532     y: 1     z: 4095 
x: 1423     y: 20     z: 4095 
x: 2051     y: 1171     z: 4097 
x: 2023     y: 2087     z: 4101 
x: 2112     y: 1312     z: 4097 
x: 1983     y: 10     z: 4095 
x: 2416     y: 0     z: 4095 
x: 2579     y: 16     z: 4152 
x: 2819     y: 2     z: 4095 
x: 2720     y: 0     z: 4076 
x: 2800     y: 0     z: 4095 
x: 2703     y: 0     z: 4095 
x: 2736     y: 2     z: 4095 
x: 1662     y: 0     z: 4095 
x: 2703     y: 0     z: 4095 
x: 2735     y: 5     z: 4101 
x: 2791     y: 11     z: 4099 
x: 2672     y: 0     z: 4095 
x: 2744     y: 0     z: 4095 
x: 2752     y: 6     z: 4039 
x: 1907     y: 2288     z: 4095 
x: 2081     y: 2736     z: 4099 
x: 2063     y: 1168     z: 4097

samand587, 2022-11-14:
And on the top the numbers themselves

samand587, 2022-11-14:
Here's a new one - I don't see the difference
Starting...
x: 1197     y: 2175     z: 4099 
x: 1532     y: 1632     z: 3928 
x: 1695     y: 1551     z: 4095 
x: 1599     y: 2444     z: 4103 
x: 1596     y: 2735     z: 4100 
x: 1580     y: 1998     z: 4097 
x: 1607     y: 1603     z: 4014 
x: 1616     y: 2544     z: 4101 
x: 1728     y: 1665     z: 3914 
x: 1744     y: 2748     z: 4100 
x: 1599     y: 2727     z: 4099 
x: 1583     y: 2703     z: 4095 
x: 1647     y: 2687     z: 4099 
x: 1599     y: 2655     z: 4100 
x: 1584     y: 1920     z: 4105 
x: 1632     y: 1532     z: 4099 
x: 1583     y: 2721     z: 4099 
x: 1627     y: 2703     z: 4095 
x: 1583     y: 95     z: 4103 
x: 2575     y: 2     z: 4095 
x: 2680     y: 0     z: 4095 
x: 2912     y: 4     z: 4053 
x: 1647     y: 1     z: 4095 
x: 2670     y: 0     z: 4095 
x: 2812     y: 0     z: 4095 
x: 1679     y: 0     z: 4095 
x: 1679     y: 2591     z: 4095 
x: 1711     y: 2307     z: 4096 
x: 1688     y: 0     z: 4095 
x: 1904     y: 1616     z: 3902 
x: 1855     y: 60     z: 4061 
x: 1607     y: 6     z: 4095 
x: 2655     y: 0     z: 4095 
x: 2896     y: 0     z: 4095 
x: 2736     y: 0     z: 4095 
x: 2819     y: 0     z: 4095 
x: 2752     y: 6     z: 4095 
x: 2912     y: 3     z: 4042 
x: 2807     y: 0     z: 4095 
x: 2104     y: 82     z: 4088 
x: 2672     y: 0     z: 4095 
x: 1887     y: 2287     z: 4103 
x: 2063     y: 1408     z: 4097 
x: 2131     y: 1552     z: 4097 
x: 2052     y: 2079     z: 4099 
x: 1536     y: 8     z: 4095 
x: 2625     y: 37     z: 4095 
x: 2864     y: 8     z: 4095 
x: 2800     y: 2     z: 4095 
x: 1840     y: 1806     z: 4097 
x: 2063     y: 2703     z: 4100 
x: 2111     y: 2879     z: 4101 
x: 2032     y: 3004     z: 4105 
x: 1639     y: 39     z: 4118 
x: 2544     y: 0     z: 4045 
x: 2831     y: 32     z: 4095 
x: 1503     y: 4     z: 4095 
x: 1624     y: 1454     z: 4099 
x: 1304     y: 0     z: 4095 
x: 1614     y: 2672     z: 4095 
x: 2048     y: 1440     z: 4097 
x: 2103     y: 1727     z: 3968 
x: 1712     y: 2744     z: 4100 
x: 1607     y: 2625     z: 4100 
x: 1543     y: 1695     z: 4035 
x: 1680     y: 2673     z: 4099 
x: 1551     y: 2752     z: 4099 
x: 1615     y: 1726     z: 4048 
x: 1548     y: 2512     z: 4095 
x: 1616     y: 2672     z: 4100

samand587, 2022-11-14:
I put the same display only version 1- it worked. But somehow it reacts very sluggishly to clicks. And it is unclear how to switch stations?

RalphBacon, 2022-11-15:
To change station you need to display the list of stations which you do by clicking the antenna symbol at the top.

[image]

samand587, 2022-11-15:
My TOUCH doesn't work! What can I do or do I have a faulty display? I put the button on GPIO13.

RalphBacon, 2022-11-15:
The "Next button is on GPIO 13, correct. That will change stations, forwards only. For testing.

Why the screen touch is not working, I do not know. I sent you a sketch yesterday but you did not answer the question(s). When you are NOT touching the screen, no serial monitor output should come out. Only when you touch the screen will the X and Y co-ordinates be shown.

Just to confirm that T_CS goes to GPIO pin 12 and T-IRQ goes to GPIO 15. GPIO15 will go LOW when you touch the screen. If you do NOT touch the screen it will be HIGH (3v3). If you have a multimeter you can check this is working.

RalphBacon, 2022-11-15:
OK, let's repeat this. Load up this sketch. Do not touch the screen once loaded.
```
#include <SPI.h>
#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();

void setup(void) {
  Serial.begin(115200);
  Serial.println("\n\nStarting...");
  tft.init();
}

void loop() {
      static int px, py;
      uint16_t x, y;
      tft.getTouchRaw(&x, &y);
    
    if (x != px || y != py) {
      px = x;
      py = y;
      Serial.printf("x: %i     ", x);
      Serial.printf("y: %i     ", y);
      Serial.printf("z: %i \n", tft.getTouchRawZ());
      delay(250);
    }
}
```
Does anything appear in the logs when you do NOT touch the screen. Send me what does get printed (if anything).

samand587, 2022-11-15:
That's what I have without touching

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
Starting...
x: 999     y: 2268     z: 4101 
x: 1500     y: 1551     z: 4098 
x: 1616     y: 1665     z: 3933 
x: 1655     y: 2400     z: 4095 
x: 1632     y: 2768     z: 4099 
x: 1584     y: 1407     z: 4098 
x: 1743     y: 2143     z: 4096 
x: 1567     y: 1743     z: 4099 
x: 1727     y: 2719     z: 4101 
x: 1679     y: 2383     z: 4103 
x: 1663     y: 1596     z: 3915 
x: 1583     y: 1919     z: 4095 
x: 1600     y: 1488     z: 4097 
x: 1551     y: 2695     z: 4099 
x: 1569     y: 1712     z: 4097 
x: 1584     y: 2744     z: 4100 
x: 1664     y: 2768     z: 4100 
x: 1571     y: 1696     z: 4097 
x: 1614     y: 2680     z: 4095 
x: 1631     y: 1539     z: 4098 
x: 1599     y: 2703     z: 4099 
x: 1544     y: 1518     z: 4099 
x: 1635     y: 1440     z: 4097 
x: 1566     y: 1551     z: 4097 
x: 1599     y: 2703     z: 4098 
x: 1584     y: 1920     z: 4105 
x: 1568     y: 2672     z: 4096 
x: 1648     y: 1532     z: 4099 
x: 1664     y: 2689     z: 4099 
x: 1552     y: 1840     z: 4097 
x: 1632     y: 2681     z: 4099 
x: 1599     y: 2703     z: 4100 
x: 1632     y: 1712     z: 4097 
x: 1488     y: 2608     z: 4101 
x: 1663     y: 1743     z: 4012 
x: 1663     y: 1664     z: 3932 
x: 1644     y: 2751     z: 4100 
x: 1619     y: 1943     z: 4104

RalphBacon, 2022-11-15:
Time to try something else that will help us work out whether your screen touch is working.

Please find this file "Keypad_240x320.ino" in the following folder:
```
ESP32 Better Internet Radio\lib\TFT_eSPI-2.2.23\examples\320 x 240\Keypad_240x320
```
and create a new sketch for it and load it.

This will show (we hope) whether your screen responds to touch.

samand587, 2022-11-15:
Returns an error

Arduino: 1.8.19 (Windows 10), Плата:"ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Debug"

C:\Users\Nestor\Desktop\205-Internet-Radio-main\205-Internet-Radio-main\Arduino IDE Version\ESP32-WROVER_Web_Radio\libraries\TFT_eSPI-2.2.23\examples\320 x 240\Keypad_240x320\Keypad_240x320.ino: In function 'void touch_calibrate()':
Keypad_240x320:211:8: error: 'SPIFFS' was not declared in this scope
   if (!SPIFFS.begin()) {
        ^
Keypad_240x320:218:7: error: 'SPIFFS' was not declared in this scope
   if (SPIFFS.exists(CALIBRATION_FILE)) {
       ^
Keypad_240x320:262:14: error: 'SPIFFS' was not declared in this scope
     File f = SPIFFS.open(CALIBRATION_FILE, "w");
              ^
exit status 1
'SPIFFS' was not declared in this scope

samand587, 2022-11-16:
I tried to upload a sketch. I get this error

Arduino: 1.8.19 (Windows 10), Board:"ESP 32 Dav Module, Disabled, Default 4MB with spiffs (1.2 MB APK/1.5 MB SPIFFS), 240 MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Debug"

C:\Users\Nestor\Desktop\205-Internet-Radio-main\205-Internet-Radio-main\Arduino IDE Version\ECP32-ROVER_Web_Radio\libraries\TFT_eSPI-2.2.23\examples\320 x 240\Keypad_240x320\Keypad_240x320.ino: In function 'void touch_calibrate()':
Keypad_240x320:211:8: error: 'SPIFFS' was not declared in this scope
if (!SPIFFS.begin()) {
^
Keypad_240x320:218:7: error: 'SPIFFS' was not declared in this scope
if (SPIFFS.exists(CALIBRATION_FILE)) {
^
Keypad_240x320:262:14: error: 'SPIFFS' was not declared in this scope
File f = SPIFFS.open(CALIBRATION_FILE, "w");
^
exit status 1
'SPIFFS' was not declared in this scope
This report will have more information with
the option File -> Settings ->
"Show detailed output at compile time" enabled

RalphBacon, 2022-11-16:
Edit the sketch and replace "SPIFFS." with "LITTLEFS." and see if that compiles.

samand587, 2022-11-16:
Here is the error

Arduino: 1.8.19 (Windows 10), Board:"ESP 32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Debug"
C:\Users\Nestor\Documents\Arduino\Keypad_240x320\Keypad_240x320.ino: In function 'void touch_calibrate()':
Keypad_240x320:211:8: error: 'LITTLEFS' was not declared in this scope
if (!LITTLEFS.begin()) {
^
Keypad_240x320:218:7: error: 'LITTLEFS' was not declared in this scope
if (LITTLEFS.exists(CALIBRATION_FILE)) {
^
Keypad_240x320:262:14: error: 'LITTLEFS' was not declared in this scope
File f = LITTLEFS.open(CALIBRATION_FILE, "w");
^
exit status 1
'LITTLEFS' was not declared in this scope

This report will have more information with
the option File -> Settings ->
"Show detailed output at compile time" enabled

[13 more replies]

closed 2022-11-28 by samand587